riscv-collab/riscv-gnu-toolchain #315: How or What solutions riscv tools support for making use of accelerator(rocc)?

allencho1222, 2018-01-03:
Finally, I am using master version of riscv-gnu-toolchain, and have followed that riscv tools no longer support its Xcustom instruction, which is used for interacting with **accelerator** (such as _dummy_rocc_).

If so, what should I use to enable _dummy_rocc_ to be activated?

p.s: I could compile _dummy_rocc_test.c_ with the help of #190. But they shows following outputs when running over spike:

`spike --extension=dummy_rocc pk a.rv`

```
z  0000000000000000 ra 00000000000100e2 sp 000000007f7e9b20 gp 0000000000015d70
tp 0000000000000000 t0 00000000000103f8 t1 0000000000000000 t2 0000000000000000
s0 000000007f7e9b50 s1 0000000000000000 a0 0000000000000001 a1 000000007f7e9b58
a2 0000000000000002 a3 000000000000007b a4 000000000000007b a5 0000000000000002
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 00000000000101c0 va 0000000000f7778b insn       00f7778b sr 8000000200046020
An illegal instruction was executed!
```
what can I do to use accelerator?

sorear, 2018-01-03:
Dup of #202, #190 

Current status riscv/riscv-binutils-gdb#104

noureddine-as, 2018-06-30:
Hello,
@sorear I guess it's not a duplicate issue. @allencho1222  was talking about Spike. All previous solutions were valid under the C++ hardware emulator. But in spike the problem occurs even if you set XS MSTATUS bits and add the dummy_rocc extension.

@allencho1222 did you find a solution?

Best regards,
Noureddine.

jim-wilson, 2018-06-30:
Upstream binutils now has .insn support which you can use for assembling non-standard instructions.

If spike is failing, that isn't a riscv-gnu-toolchain problem.  Try reporting it against spike.

closed 2018-07-31 by jim-wilson

EllenMiller25, 2019-08-07:
I am still having the same issue. I am not sure how did you fixed the problem.
I am under riscv-isa-sim/dummy_rocc and when I run:
riscv64-unknown-elf-gcc dummy_rocc_test.c

it gives me:
dummy_rocc_test.c: Assembler messages:
dummy_rocc_test.c:14: Error: unrecognized opcode `custom0 x0,a5,2,0'
dummy_rocc_test.c:17: Error: unrecognized opcode `custom0 a5,x0,2,1'
dummy_rocc_test.c:20: Error: unrecognized opcode `custom0 x0,a5,2,3'
dummy_rocc_test.c:22: Error: unrecognized opcode `custom0 a5,x0,2,1'
dummy_rocc_test.c:25: Error: unrecognized opcode `custom0 x0,a5,2,2'
dummy_rocc_test.c:26: Error: unrecognized opcode `custom0 x0,a5,2,3'
dummy_rocc_test.c:27: Error: unrecognized opcode `custom0 a5,x0,2,1'

I would appreciate if somebody can help! Thanks!

jim-wilson, 2019-08-15:
The rocc support was never upstreamed, and I've never seen it.  Rocc is not a part of the ISA spec, and binutils only supports the instructions in the standard ISA.  So either you find a toolchain with rocc support, somewhere, I don't know where.  Or you change the rocc code to use the supported .insn feature.  Or you stop trying to use the rocc extension.
https://sourceware.org/binutils/docs-2.32/as/RISC_002dV_002dFormats.html#RISC_002dV_002dFormats

aswaterman, 2019-08-15:
The only RoCC support that ever existed in the toolchain was basically a
restricted form of the .insn feature, so using .insn would be my
recommendation as well.