riscv-collab/riscv-gnu-toolchain #317: regarding cloning of riscv-gnu-toolchain

thotaraghava, 2018-01-08:
Hi All,

     I am cloning this database with GitHub Desktop software, I am seeing message "Receiving objects #%......".
   What does it mean.
   Is cloning is done ??
   I could see cloning dir in Windows target place.

Thanks,
Raghav

sorear, 2018-01-08:
No, it's in progress.

thotaraghava, 2018-01-08:
ok. Thanks alot.

thotaraghava, 2018-01-08:
Its taking long time like this.
So, I should wait untill I see message "transfer completed" it suppose

ilg-ul, 2018-01-08:
the submodules used by this project are very large repositories, with the source code of the compiler, libraries, tools, debugger, etc.

they **are** large, if you clone everything it is normal to take some time.

but apart from studying the source code, what do you plan to do with them on Windows?

thotaraghava, 2018-01-08:
My requirement is as below.
1. I have downloaded PULPino core from GitHub.
2. To simulate the core, I need to have toolchain of PULPino.

At the End of this cloning, I have got below message
"Cloning is Failed, Re-schedule is on process".

Should I again try to clone the same.
I am utter novice in this area.
Any small help is highly appreciated.

Thanks @ilg-ul

ilg-ul, 2018-01-08:
I don't have experience with PULPino, so I might be wrong, but are you sure the PULPino development environment runs without problems on Windows?

thotaraghava, 2018-01-09:
Thing is I am not working on Windows.
Just downloading there and transferring data to Linux environment.

thotaraghava, 2018-01-09:
I am seeing again cloning failed with many errors.
But database is being created in target directory in Windows.

Please suggest anything.

palmer-dabbelt, 2018-01-09:
This has nothing to do with PULPino, that's just another RTL RISC-V implementation.  This smells like he's at a hardware shop that segregates the real computers form the terminals, and doesn't allow internet access to a machine on which people can do work.  I've run into this a handful of times before.  I think the problem here is that your Windows environment is a mess (as Windows tends to be).  If for some reason you can't clone this repo on Windows, you have a few options:

* Buy a computer, attach it to the underside of your desk with double-sided sticky tape, and tell nobody it's there.  This requires you to live in a climate that's fairly temperature stable, as otherwise it'll fall off within a few weeks.  It's also a fine line to walk: if you go with <https://www.amazon.com/Scotch-Indoor-Mounting-1-Inch-125-Inches/> you run the risk of your computer falling down, but if you go with <https://www.amazon.com/3M-Heavy-Duty-Mounting-Black> then you run the risk of getting stuck to your own desk.  If someone is paying you to do this then it's probably the wrong answer, but if you're a student then this will quickly pay for even the expensive tape.
* Yell at your sysadmin until he enables you to do your job.  This may require either allowing your terminal to handle creating ~300k files totaling ~3GiB, or allowing your real computer to access the internet.  I assume this isn't an option :)
* Rent a real computer.  I've been using Alpharacks for years, and while I have many complaints I've never cancelled my service because I pay $0.50 per month.  It looks like they've raised their prices a bit (it's now $0.59/mo <https://lowendbox.com/blog/black-friday-deal-alpharacks-1gb-openvz-7-year/>) but it'l really still in way far down in the noise if you're considering building an ASIC.  There are hundreds of these providers, feel free to pick your favorite.  If you really can't afford it, I can try to figure out how to lend you one of mine.
* Download the relevant tarballs manually and copy them over.  There's nothing special about tagged releases in git land, so you can grab any hash as a tarball from github.  For example https://github.com/riscv/riscv-gnu-toolchain/archive/v20171231.tar.gz and https://github.com/riscv/riscv-gnu-toolchain/archive/d66c792dfe701c63a8df2028598174fbeee4e8f1.tar.gz should have the same contents.  You'll have to chase the submodule pointers yourself, but they're right there in the UI.
* File a bug report to github, informing them that they should include the submodule contents as part of their archives.  This will be tricky: they probably won't want to replace old archives with those containing submodules, so maybe they'll want some sort of "tag+give_me_the_submodules.tar.gz" URL scheme.

That's pretty much all I've got.  Good luck!

closed 2018-01-09 by palmer-dabbelt

jim-wilson, 2018-01-09:
You can do real work on a windows machine if you install cygwin or msys2.  Either one will give you a POSIX shell where you can run git commands directly instead of relying on git desktop, whatever that is.

The other issue is that if you are behind a firewall, then the firewall might be blocking some protocols.  Try a different URL to see if one of the others works, e.g. git, http, https.

If you continue to have problems, you probably need to talk to someone at your location, to get info on how to download stuff to there.

thotaraghava, 2018-01-09:
Thanks alot for your kind information.