riscv-collab/riscv-gnu-toolchain #318: Make file error while building the Newlib cross-compiler

sourav1394, 2018-01-11:
When I run the make command, I get the following error:

`../../riscv-gcc/gcc/doc/invoke.texi:1677: unknown command `DAC'
../../riscv-gcc/gcc/doc/invoke.texi:1677: @include: could not find /home/sourav/Desktop/Hack/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc/../.././riscv-gcc/gcc/../libiberty/at-file.texi
make[2]: *** [doc/gcc.info] Error 1
make[2]: Leaving directory `/home/sourav/Desktop/Hack@DAC/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/home/sourav/Desktop/Hack@DAC/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [stamps/build-gcc-newlib-stage1] Error 2`

How do I fix this?

ilg-ul, 2018-01-11:
to build the manuals, you need, among other things, texlive.

jim-wilson, 2018-01-12:
Don't put @ in the name of a parent directory of the riscv-gnu-toolchain dir.  @ starts a command for makeinfo, and it is interpreting the @ in the pathname as a command, which is not intended here.  This gives both the unknown command DAC error, and the broken pathname which is missing the "@DAC".  I don't see any obvious way to turn off this feature, so you just have to avoid putting @ in directory names.

jim-wilson, 2018-01-12:
Actually, using a relative path to configure instead of an absolute path might be enough to work around the problem without a directory name change.

sourav1394, 2018-01-12:
Thank you! It is working now.

closed 2018-01-12 by sourav1394