riscv-collab/riscv-gnu-toolchain #319: Can we make the GDB to be able to access the self-defined CSR registers?

SI-RISCV, 2018-01-17:
Dear Experts

In my implemented RISC-V Core, I have used some unused CSR address to define some my own CSR registers. For example, I used CSR address 0xBFF (not used address) to define a my own CSR register to control some features in my core.

Sometimes I want to debug my core via GDB, and I want to check the value of CSR registers, so I just use command "info reg csr3071" (3071 is the decimal value of 0xBFF), seems in the older version of GDB, this command worked fine, but in the latest version of GDB, looks like it is not supported, and it just return a message "invalid register csr3071"?

Do you know how can we access these non-standard self-defined CSR register via GDB? is there any way to make it?

Thanks
Bob

ilg-ul, 2018-01-17:
From the point of view of the debuggers and development tools, CSRs themselves are not standard, and there are no standard ways to access them.

Initially GDB assimilated the CSRs to General Registers. All 4096 of them. This was a mistake, since at each step GDB had to fetch and transfer all of them to the IDEs running on top of it. Even worse, for some tools passing arrays of 4000 strings at each call was not supported, and crashed.

In the recent version of GDB, a more elaborate mechanism of passing an XML with the names of the actual registers was enabled, but this requires cooperation of the JTAG tools (OpenOCD, J-Link, etc) to identify the device and pass the correct XML.

So, for the moment, if you made the mistake of using CSRs (non-standard by all means) instead of the classical MMIO registers, which are easily examined by any debugger, you may encounter some difficulties, at least for a while.

AntonKrug, 2018-01-17:
I think the XML is good step, in other platforms you can make you tdesc xml describing your platform so the GDB will not have to be considering all possible combinations, only the selected target. Which is good. It will mean for you that in future you will have to have custom xml for your custom implementation, but it's for better good.

timsifive, 2018-01-17:
Yes. There's an OpenOCD command that must be run before `init` to let it
expose additional registers. For instance, an OpenOCD config file might
contain:
```
riscv expose_csrs 3071
init
```
or
```
riscv expose_csrs 10-20,50,3071-3078
init
```
Once you've done that they'll show up in GDB just as before.

You're probably the first to use that feature. Let me know if it gives you
any trouble.

SI-RISCV, 2018-01-18:
@ilg-ul @AntonKrug @timsifive 

Dear Experts

Thanks very much for your quick response and suggestion. I hope we can add some mechanism to enable user to define their self-defined CSR and have GDB be able to access them easily.

The solution @timsifive proposed looks very good, that is, to just add some OpenOCD commands that must be run before `init`. It is easy to add and change it without need recompiling any openocd and GNU tools.

I have tried it at my end, it works indeed. I think it is already good enough with this way to access my own CSRs.

Thanks
Bob

timsifive, 2018-01-18:
The long-term solution is an ISA extension that enables a program to
determine without side effects whether a given CSR exists. Then the
debugger can simply figure out which ones exist when connecting to the
target.

ilg-ul, 2018-01-18:
The ISA solution sounds nice, but it might take a while.

For the moment we should use the XML definition, and this implies adding a command to OpenOCD to specify the XML file with the actual device register list. If the file is not explicitly defined, and the device has a unique JTAG ID, OpenOCD can try to match it from a list of known files, present in the OpenOCD distribution.

AntonKrug, 2018-01-18:
@timsifive  Would it be in the scope of the ISA extension to include the CSR access instructions so the CSRs could be accessed with registers instead of immediate, allowing less hackery way for an application to debug themselves? Like a minimalistic barebone OS which would be able to fetch the CSRs from it's embedded command line. Something very lightly in the spirit of "cat /proc/cpuinfo" "cat /proc/meminfo" but one able to access the CSRs dynamically?

timsifive, 2018-01-18:
@AntonKrug The ISA extension is completely hypothetical at this point. There isn't even a proposal for it. I'm not sure what you mean by accessing with registers instead of immediate. There already are instructions that transfer CSRs between GPRs (csrr and csrw, respectively). A running program can definitely read all the CSRs that it has access to.

ilg-ul, 2018-01-18:
> ... A running program can definitely read all the CSRs that it has access to.

@timsifive, imagine you have to write a monitor that runs on a RISC-V device, and you need to implement a command like:

```console
> csr read <N>
```

How would you do it? How do you get the value of a CSR whose number is not known at compile time?

jim-wilson, 2018-01-18:
Tim answered the original question.  Liviu's question has already been answered multiple times and is not worth further discussion.

closed 2018-01-18 by jim-wilson