riscv-collab/riscv-gnu-toolchain #322: Relaxation processing fails for vma and output offset of 0

red-mercury, 2018-02-07:
I was trying to compile some bare-metal code for RISC-V RV32IM with a base address of 0x00000000 and kept hitting the linker error "relocation truncated to fit: R_RISCV_PCREL_LO12_I against `.L0 '"
However linking worked if I changed the text base address to be non-zero (e.g. 0x00000004) or if I disabled relaxation in the linker with --no-relax (tested on 2.28, 2.29 and git master and riscv-next).
I narrowed it down to a test case with a subset of the linker script and start code:

Linker script (test.lds):
```
ENTRY( _start )

MEMORY
{
  imem (rxai!w) : ORIGIN = extorigin, LENGTH = 512K
  dmem (wxa!ri) : ORIGIN = 0x10000000, LENGTH = 16K
}

PHDRS
{
  imem PT_LOAD;
  dmem_init PT_LOAD;
  dmem PT_NULL;
}

SECTIONS
{
  .init :
  {
    KEEP (*(SORT_NONE(.init)))
  } >imem AT>imem :imem

  .text :
  {
    *(.text .text.*)
    *(.gnu.linkonce.t.*)
  } >imem AT>imem :imem

  .datainimemalign :
  {
    . = ALIGN(4);
    _data_imem_start = . ;
  } >imem AT>imem :imem

  .data          :
  {
    . = ALIGN(4);
    _data = . ;
    *(.data .data.*)
    *(.gnu.linkonce.d.*)

    __global_pointer$ = . + 0x800; 

    *(.srodata.cst16) *(.srodata.cst8) *(.srodata.cst4) *(.srodata.cst2) *(.srodata .srodata.*)
    *(.sdata .sdata.* .gnu.linkonce.s.*)

    . = ALIGN(4);
    _edata = . ;
  } >dmem AT>imem :dmem_init
}

```

Start code (start.S):
```
	.section .init
	.globl _start
	.type _start,@function
_start:

.option push
.option norelax
	la gp, __global_pointer$
.option pop
	la a0, _data_imem_start
1:
	j 1b
```

The following commands then show the problem:
```
# RISC-V relaxation test cases
riscv32-unknown-elf-gcc -march=rv32im -c start.S

echo "Testing with origin 0x00000004, relaxation enabled"
riscv32-unknown-elf-ld --defsym extorigin=0x00000004 -o testit1 -melf32lriscv -T test.lds start.o
# Works with 0 < extorigin < dmem

echo -e "\nTesting with origin 0x00000000, relaxation enabled"
riscv32-unknown-elf-ld --defsym extorigin=0x00000000 -o testit2 -melf32lriscv -T test.lds start.o
# Fails with:
# start.o: In function `.L0 ':
# (.init+0x8): relocation truncated to fit: R_RISCV_PCREL_LO12_I against `.L0 '

echo -e "\nTesting with origin 0x00000000, relaxation disabled"
riscv32-unknown-elf-ld --defsym extorigin=0x00000000 --no-relax -o testit3 -melf32lriscv -T test.lds start.o
# Works with 0 <= extorigin < dmem
```
Digging into the binutils code, in bfd/elfnn-riscv.c function _bfd_riscv_relax_section() there is a check:
```
          if (isym->st_shndx == SHN_UNDEF)
            sym_sec = sec, symval = sec_addr (sec) + rel->r_offset;
          else
            {
              BFD_ASSERT (isym->st_shndx < elf_numsections (abfd));
              sym_sec = elf_elfsections (abfd)[isym->st_shndx]->bfd_section;

              if (sec_addr (sym_sec) == 0)
                continue;

              symval = sec_addr (sym_sec) + isym->st_value;
            }
```
but sec_addr(sym_sec) in this case will be zero as the sym_sec->output_section->vma = 0 and sym_sec->output_offset = 0 for .init when extorigin = 0x00000000 and so the relax function won't be called for _data_imem_start.

Commenting out the check allows the link to correctly work at 0x00000000 (confirmed with objdump -dtr).

I'm not sure what the sec_addr(sym_sec) == 0 check is supposed to prevent against, but it looks like it is not a complete enough check.

If you can provide some help with the purpose of the check then I'm happy to create a patch to fix and add a testsuite test case.

PkmX, 2018-02-07:
This is also reported at riscv/riscv-gcc#120.

red-mercury, 2018-02-07:
Thought I'd checked through all the other issues, but missed that one and it looks like the same issue.  It probably makes sense to close this one.

jim-wilson, 2018-02-07:
Yes, a known problem, discussed in https://github.com/riscv/riscv-gcc/issues/120 , but unfortunately the solution is not yet known, other than just disabling the apparently unnecessary zero section address check.

closed 2018-02-07 by jim-wilson