riscv-collab/riscv-gnu-toolchain #323: qemu needs update for glibc 2.26

sorear, 2018-02-12:
`make build-qemu` fails on a Fedora 27 system with an error related to `struct ucontext`, which was renamed in glibc 2.26.  This was fixed in https://github.com/qemu/qemu/commit/04b33e21866412689f18b7ad6daf0a54d8f959a7 , but the submodule does not have the fix; moving my submodule pointer to qemu-upstream-v4 allowed it to build successfully.

filamoon, 2019-04-29:
Running into the same problem. Could you elaborate on how to make the change? Thanks!

jim-wilson, 2019-05-04:
riscv-gnu-toolchain contains qemu only for running the testsuite.  It is not part of riscv-gnu-toolchain; it is only included for convenience for toolchain developers.  If you need qemu for some other reason, you should not get it from here.  You should get it from qemu.org.

The 32-bit glibc sources are not upstream yet.  In order to continue testing binutils and gcc support for 32-bit linux we are deliberately using an old version of glibc that contains the 32-bit glibc support.  This in turn requires using an old qemu version.  If you upgrade qemu, the 32-bit linux support will break. 
 Eventually these problems will be fixed, but meanwhile, riscv-gnu-toolchain is a bit limited in what it can do.

If you don't care about 32-bit linux support, you can use git commands to upgrade qemu to whatever version you want.  Something like "cd riscv-qemu; git fetch; git checkout origin/master; git submodule update --init --recursive" for instance will give you the top of the riscv-qemu tree.  Just avoid doing a git submodule at the top of the tree after this, or riscv-qemu will revert to the previous state.  You can also use git remote to pull in a copy of the upstream qemu.org tree.

If you do care about the 32-bit linux support, then you can try extracting and applying the one patch you need for the struct ucontext_t change.  Or alternatively, use an older OS that this old version of qemu works with.  Ubuntu 16.04 is known to work for instance.

closed 2021-03-08 by lazyparser