riscv-collab/riscv-gnu-toolchain #324: SSL subject name and target host name mismatch for qemu directory while cloning

Sumaid, 2018-02-16:


Sumaid, 2018-02-16:
When I do `git submodule update --init --recursive`
I'm getting this error:
Cloning into 'dtc'...
fatal: unable to access 'https://git.qemu-project.org/dtc.git/': SSL: certificate subject name (qemu.org) does not match target host name 'git.qemu-project.org'
fatal: clone of 'git://git.qemu-project.org/dtc.git' into submodule path 'dtc' failed
Failed to recurse into submodule path 'riscv-gnu-toolchain/riscv-qemu'
Failed to recurse into submodule path 'riscv-gnu-toolchain'

I did this also but still it's not working 
[url "https://git.qemu.org/git"]
insteadOf = https://git.qemu-project.org

jim-wilson, 2018-02-16:
Did you place that into your ~/.gitconfig file?  It works for me when that url/insteadOf stanza is in my ~/.gitconfig file.

jim-wilson, 2018-02-16:
What exact error do you get after modifying your ~/.gitconfig file?

SunYongshuai, 2018-03-09:
I did this also but still it's not working
[url "https://git.qemu.org/git"]
insteadOf = https://git.qemu-project.org

but I change to 

[url "https://github.com/qemu/"]
insteadOf = https://git.qemu-project.org/

It works!

May be https://git.qemu.org/git also breaks down.

SunYongshuai, 2018-03-15:
Or 
[url "https://github.com/qemu/"]
insteadOf = git://git.qemu-project.org/

closed 2021-03-08 by lazyparser