riscv-collab/riscv-gnu-toolchain #326: Becommended gcc version? Build errors with gcc8

teamblubee, 2018-03-07:
I've installed all the prereqs, the config.status and config.log are below.
Is this an issue with my build tools or something else?

build error here:
```shell
rm -f plural.h
Live child 0x8006e44b0 (plural.c) PID 12403 
Reaping winning child 0x8006e44b0 PID 12403 
Removing child 0x8006e44b0 PID 12403 from chain.
        Successfully remade target file 'plural.c'.
        Pruning file 'config.h'.
        Pruning file 'libintl.h'.
        Pruning file '/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h'.
       Finished prerequisites of target file 'plural.o'.
      Must remake target 'plural.o'.
cc -c  -g -O2 -DHAVE_CONFIG_H  -I. -I/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl plural.c
Putting child 0x8006e44b0 (plural.o) PID 12404 on the chain.
Live child 0x8006e44b0 (plural.o) PID 12404 
plural.c:184:5: error: conflicting types for 'libintl_gettextparse'
int __gettextparse (void);
    ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural.y:40:25: note: expanded from macro '__gettextparse'
# define __gettextparse PLURAL_PARSE
                        ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:102:23: note: expanded from macro 'PLURAL_PARSE'
# define PLURAL_PARSE libintl_gettextparse
                      ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:114:12: note: previous declaration is here
extern int PLURAL_PARSE PARAMS ((void *arg));
           ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:102:23: note: expanded from macro 'PLURAL_PARSE'
# define PLURAL_PARSE libintl_gettextparse
                      ^
plural.c:1129:1: error: conflicting types for 'libintl_gettextparse'
yyparse (void)
^
plural.c:63:25: note: expanded from macro 'yyparse'
#define yyparse         __gettextparse
                        ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural.y:40:25: note: expanded from macro '__gettextparse'
# define __gettextparse PLURAL_PARSE
                        ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:102:23: note: expanded from macro 'PLURAL_PARSE'
# define PLURAL_PARSE libintl_gettextparse
                      ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:114:12: note: previous declaration is here
extern int PLURAL_PARSE PARAMS ((void *arg));
           ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural-exp.h:102:23: note: expanded from macro 'PLURAL_PARSE'
# define PLURAL_PARSE libintl_gettextparse
                      ^
plural.c:1298:30: error: too few arguments to function call, expected 2, have 1
      yychar = yylex (&yylval);
               ~~~~~         ^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural.y:69:1: note: '__gettextlex' declared here
static int yylex PARAMS ((YYSTYPE *lval, const char **pexp));
^
/usr/home/blubee/dev/riscv/_riscv-gnu-toolchain/riscv-binutils-gdb/intl/plural.y:178:29: error: use of undeclared identifier 'arg'
            ((struct parse_args *) arg)->res = (yyvsp[0].exp);
```

```shell
gcc8 --version
gcc8 (FreeBSD Ports Collection) 8.0.1 20180304 (experimental)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bison --version
bison (GNU Bison) 3.0.4
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
verbose gmake command output: https://pastebin.com/mwq0sQA6
config.status: https://pastebin.com/x7iVrdU1
config.log: https://pastebin.com/CZjGrf8A

jim-wilson, 2018-03-07:
GCC8 hasn't been released yet.  riscv-binutils-gdb is binutils-2.29 which was released well over a year before the potential target date for gcc-8, and hence could never have been tested with it.  It is a known problem that it doesn't  compile with gcc-8, primarily due to new warnings added between gcc 7 and gcc 8.

binutils-2.29 was obsoleted by binutils-2.30 which was released at end of January, but that one was not tested by gcc-8 either.

The binutils mainline sources have been fixed.  It required two large patches.  And since gcc-8 hasn't been released yet, it is possible that more patches may be necessary.  The info is here
    https://sourceware.org/bugzilla/show_bug.cgi?id=22823

For now, I would recommend using gcc-7 to compile binutils, and using the FSF binutils-2.30 release instead of the old riscv-binutils-gdb sources.

closed 2018-03-07 by jim-wilson

jim-wilson, 2018-03-07:
If you really want to compile binutils releases with gcc8, you can just modify the binutils Makefiles to disable warnings.  Just removing the -Werror should be enough.

Though looking at this again, I just noticed that there is a file bfd/development.sh which controls use of -Werror, and is supposed to be false in releases and true in development sources.  It is true in riscv-binutils-gdb which is odd, as I thought this was a release tree not a development tree.  Changing this file might be enough to solve the problem with gcc-8.  I haven't tested that.

teamblubee, 2018-03-09:
After doing some more research into this issue, it's related to bison and how it's parsing the "plurals.y" to create the .c file.

There seems to be some issues with the bison parsing. after running this command
```shell
bison -y --nae-prefix=__gettext --output=plural.c riscv-binutils-gdb/intl/plural.y
```
that produces the plurals.c file which causes the make command to fail with conflicting types for libintl_gettextparse.

You can see the output plural.c created by the bison command. Does anything seem out of place? make reports errors at
```shell
cc -c  -g -O2 -DHAVE_CONFIG_H  -I. -I/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/riscv-binutils-gdb/intl plural.c
plural.c:184:5: error: conflicting types for 'libintl_gettextparse'
int __gettextparse (void);

and

plural.c:1129:1: error: conflicting types for 'libintl_gettextparse'
yyparse (void)
```

```shell
/* A Bison parser, made from plural.y
   by GNU bison 1.35.  */

#define YYBISON 1  /* Identify Bison output.  */

#define yyparse __gettextparse
#define yylex __gettextlex
#define yyerror __gettexterror
#define yylval __gettextlval
#define yychar __gettextchar
#define yydebug __gettextdebug
#define yynerrs __gettextnerrs
# define	EQUOP2	257
# define	CMPOP2	258
# define	ADDOP2	259
# define	MULOP2	260
# define	NUMBER	261

#line 1 "plural.y"

/* Expression parsing for plural form selection.
   Copyright (C) 2000, 2001 Free Software Foundation, Inc.
   Written by Ulrich Drepper <drepper@cygnus.com>, 2000.

   This program is free software; you can redistribute it and/or modify it
   under the terms of the GNU Library General Public License as published
   by the Free Software Foundation; either version 2, or (at your option)
   any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
   Library General Public License for more details.

   You should have received a copy of the GNU Library General Public
   License along with this program; if not, write to the Free Software
   Foundation, Inc., 51 Franklin Street - Fifth Floor, Boston, MA 02110-1301,
   USA.  */

/* The bison generated parser uses alloca.  AIX 3 forces us to put this
   declaration at the beginning of the file.  The declaration in bison's
   skeleton file comes too late.  This must come before <config.h>
   because <config.h> may include arbitrary system headers.  */
#if defined _AIX && !defined __GNUC__
 #pragma alloca
#endif

#ifdef HAVE_CONFIG_H
# include <config.h>
[... 1421 lines cut ...]
      /* Be safe and let the user call this function again.  */
      --exp;
      result = YYEOF;
      break;

    default:
      result = YYERRCODE;
#if YYDEBUG != 0
      --exp;
#endif
      break;
    }

  *pexp = exp;

  return result;
}

static void
yyerror (str)
     const char *str;
{
  /* Do nothing.  We don't print error messages here.  */
}
```

jim-wilson, 2018-03-09:
libintl is used only if NLS is enabled.  This allows warnings/errors to be emitted in languages other than English.  If you don't need this feature, you can disable it with --disable-nls when configuring.

Normally, binutils will use the libintl that comes with the OS.  It will only build the self contained one if the OS one is missing.  So installing the OS package for libintl if you are on linux will solve the problem, as then you won't need to build libintl anymore.  The package name will depend on the OS, and could be libgettext or libintl or something else.

The binutils sources come with a plural.c file.  There is no need to run bison, and normally, bison will not be run when building libintl.  Bison will be run only if the plural.y file is newer than the provided plural.c file.  If bison is run, the resulting plural.c will be put in the build dir.  You are looking at the plural.c in the source dir which was not used by the compile, which is why the error messages don't make any sense.  Look at the plural.c in the build dir to see the problem.  And the problem is that the included libintl was built using an old bison version 1.35, and you probably have something newer like 3.0.4, which the included libintl does not work correctly with.  The easiest way to fix this problem is to delete the plural.c file in the build tree and touch the plural.c file in the source tree, so that bison won't be run during the build.

The upstream FSF Binutils sources still have this bug.  The FSF Binutils maintainers probably haven't seen this problem yet because a number of things have to go wrong to trigger it.  Timestamps in your source tree must be wrong, the OS libintl package must not be installed, and you must have NLS support enabled when configuring.  Since this is a not a riscv bug, but rather a FSF bug, it should be reported upstream.
  http://sourceware.org/binutils
Would you like to do that?  I can do that if you don't want to.

teamblubee, 2018-03-09:
That was a pretty interesting bug that I was able to dig up. I was able to resolve it by installing the local gettext and associated runtimes.

It might be worth while to report the problem upstream as well.

After making those changes the make command run for a lot longer but still fails looking for libexpat?

```shell
rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv64-unknown-elf-gcc /usr/home/blubee/dev/riscv/riscv-gnu-toolchain/riscv-binutils-gdb/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/home/blubee/dev/riscv/riscv-gnu-toolchain/_gnu_toolchain \
	 \
	--disable-werror \
	--with-expat=yes  \
	--enable-gdb
checking build system type... x86_64-unknown-freebsd12.0
checking host system type... x86_64-unknown-freebsd12.0
checking target system type... riscv64-unknown-elf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
...
checking whether to use expat... yes
checking for libexpat... no
gmake[2]: Leaving directory '/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/build-binutils-newlib'
gmake[1]: Leaving directory '/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/build-binutils-newlib'
```
full log here: https://pastebin.com/VHZengET

I have the software package installed and this error?

jim-wilson, 2018-03-09:
Gdb is actually checking for the libexpat development libraries, which is probably something like libexpat-dev or -devel.  Maybe you have the application installed, or the run-time libraries but not the compile-time libraries and header files. You can see what configure is doing by looking at the gdb/config.log file, just search for expat, and you should see a failing command, probably a command trying to link with -lexpat.

Gdb requires expat to use the OpenOCD XML support.  If you aren't using OpenOCD, you can modify the Makefile to disable expat support.

jim-wilson, 2018-03-09:
I found that there is already an upstream bug report for the intl problem.
    https://sourceware.org/bugzilla/show_bug.cgi?id=22941

teamblubee, 2018-03-10:
I found the definition for --with-expat in the riscv-gnu-toolchain/Makefile

```shell
# --with-expat is required to enable XML support used by OpenOCD.
# BINUTILS_GDB_TARGET_FLAGS := --with-expat=yes $(BINUTILS_GDB_TARGET_FLAGS_EXTRA)
BINUTILS_GDB_TARGET_FLAGS := $(BINUTILS_GDB_TARGET_FLAGS_EXTRA)
```

I commented out the --with-expat line and the build proceeds but it stalls much later in the process.

there's a file in the riscv-gcc/contrib/download_prerequisites, that tries to grab some files; it downloads gmp-6.1.0 but my system has gmp-6.1.2 installed already.

```shell
gmake[2]: Nothing to be done for 'install-target'.
gmake[2]: Leaving directory '/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/build-binutils-newlib'
gmake[1]: Leaving directory '/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/build-binutils-newlib'
mkdir -p stamps/ && touch stamps/build-binutils-newlib
if test -f /usr/home/blubee/dev/riscv/riscv-gnu-toolchain/riscv-gcc/contrib/download_prerequisites && test "true" == "true"; then cd /usr/home/blubee/dev/riscv/riscv-gnu-toolchain/riscv-gcc && ./contrib/download_prerequisites; fi
2018-03-10 10:19:00 URL: ftp://gcc.gnu.org/pub/gcc/infrastructure/gmp-6.1.0.tar.bz2 [2383840] -> "./gmp-6.1.0.tar.bz2" [1]
./contrib/download_prerequisites: sha512sum: not found
error: Cannot verify integrity of possibly corrupted file gmp-6.1.0.tar.bz2
gmake: *** [Makefile:349: stamps/build-gcc-newlib-stage1] Error 1
```

If gmp is not being found I bet it will be the same for mpfr mpc as well. They were all installed when GCC was installed.

jim-wilson, 2018-03-10:
Yes, this is another prerequisites issue.  You need the development packages that contain header files and link-time libraries in /usr/lib.  You probably only have the run-time libraries in /lib.  The compile/link time files and the run-time files are usually separate packages in most linux distros.  GCC requires the gmp, mpc, and mpfr development packages.  The riscv-gnu-toolchain README.md file has suggested commands to install missing packages for ubuntu and fedora.  The riscv-gnu-toolchain makefile has support to try to download sources for these packages if they are missing, and then build them as part of the gcc build, but this part failed for you because sha512sum is missing.  You can try installing the missing development library packages, or you could try installing the missing sha*sum programs.

teamblubee, 2018-03-10:
Hi

Thank you for your help, I made a few edits and was able to successfully compile the binutils.

After compilation finished I have these in the $PREFIX/bin directory

```shell
ls toolchain/bin/riscv64-unknown-elf-*
toolchain/bin/riscv64-unknown-elf-addr2line  toolchain/bin/riscv64-unknown-elf-gcc-ar     toolchain/bin/riscv64-unknown-elf-nm
toolchain/bin/riscv64-unknown-elf-ar         toolchain/bin/riscv64-unknown-elf-gcc-nm     toolchain/bin/riscv64-unknown-elf-objcopy
toolchain/bin/riscv64-unknown-elf-as         toolchain/bin/riscv64-unknown-elf-gcc-ranlib toolchain/bin/riscv64-unknown-elf-objdump
toolchain/bin/riscv64-unknown-elf-c++        toolchain/bin/riscv64-unknown-elf-gcov       toolchain/bin/riscv64-unknown-elf-ranlib
toolchain/bin/riscv64-unknown-elf-c++filt    toolchain/bin/riscv64-unknown-elf-gcov-dump  toolchain/bin/riscv64-unknown-elf-readelf
toolchain/bin/riscv64-unknown-elf-cpp        toolchain/bin/riscv64-unknown-elf-gcov-tool  toolchain/bin/riscv64-unknown-elf-run
toolchain/bin/riscv64-unknown-elf-elfedit    toolchain/bin/riscv64-unknown-elf-gdb        toolchain/bin/riscv64-unknown-elf-size
toolchain/bin/riscv64-unknown-elf-g++        toolchain/bin/riscv64-unknown-elf-gprof      toolchain/bin/riscv64-unknown-elf-strings
toolchain/bin/riscv64-unknown-elf-gcc        toolchain/bin/riscv64-unknown-elf-ld         toolchain/bin/riscv64-unknown-elf-strip
toolchain/bin/riscv64-unknown-elf-gcc-7.2.0  toolchain/bin/riscv64-unknown-elf-ld.bfd
```

I had to make these few changes;
This first one doesn't affect anything and expands the script to use FreeBSD shasum.

```shell
./riscv_gnu_toolchain/riscv-gcc/contrib/download_prerequisites

diff --git a/contrib/download_prerequisites b/contrib/download_prerequisites
index 3430ce1f19e..36118400ebe 100755
--- a/contrib/download_prerequisites
+++ b/contrib/download_prerequisites
@@ -50,6 +50,9 @@ case $OS in
   "Darwin")
     chksum='shasum -a 512 --check'
   ;;
+  "FreeBSD")
+    chksum='shasum -a 512'
+  ;;
   *)
     chksum='sha512sum --check'
   ;;
```

This other patch most likely breaks OpenOCD since that's what the comment says but I couldn't find a way to make it work.
```shell
./riscv_gnu_toolchain/makefile.in

diff --git a/Makefile.in b/Makefile.in
index f6f78f3..de35a00 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -57,7 +57,7 @@ NEWLIB_TUPLE ?= $(call make_tuple,$(XLEN),elf)
 CFLAGS_FOR_TARGET := $(CFLAGS_FOR_TARGET_EXTRA) @cmodel@
 ASFLAGS_FOR_TARGET := $(ASFLAGS_FOR_TARGET_EXTRA) @cmodel@
 # --with-expat is required to enable XML support used by OpenOCD.
-BINUTILS_GDB_TARGET_FLAGS := --with-expat=yes $(BINUTILS_GDB_TARGET_FLAGS_EXTRA)
+BINUTILS_GDB_TARGET_FLAGS := --with-expat=no $(BINUTILS_GDB_TARGET_FLAGS_EXTRA)
 BINUTILS_GDB_NATIVE_FLAGS := $(BINUTILS_GDB_NATIVE_FLAGS_EXTRA)
 GLIBC_TARGET_FLAGS := $(GLIBC_TARGET_FLAGS_EXTRA)
 GLIBC_CC_FOR_TARGET ?= $(LINUX_TUPLE)-gcc
```