riscv-collab/riscv-gnu-toolchain #328: riscv-gcc configuration not supported

teamblubee, 2018-03-11:
riscv-gnu-binutils builds successfully with this configuration:
```shell
stamps/build-binutils-freebsd: $(srcdir)/riscv-binutils-gdb
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
	cd $(notdir $@) && $</configure \
		--target=riscv$(XLEN)-unknown-freebsd12.0 \
		--prefix=$(INSTALL_DIR) \
		--with-sysroot=$(SYSROOT) \
		--disable-gdb \
		--disable-sim \
		--enable-tls \
		--disable-intl \
		--disable-werror
	$(MAKE) -C $(notdir $@)
	$(MAKE) -C $(notdir $@) install
	mkdir -p $(dir $@) && touch $@
```
I get these binaries from the above configure command
```shell
riscv64-unknown-freebsd12.0-addr2line riscv64-unknown-freebsd12.0-gprof     riscv64-unknown-freebsd12.0-ranlib
riscv64-unknown-freebsd12.0-ar        riscv64-unknown-freebsd12.0-ld        riscv64-unknown-freebsd12.0-readelf
riscv64-unknown-freebsd12.0-as        riscv64-unknown-freebsd12.0-ld.bfd    riscv64-unknown-freebsd12.0-run
riscv64-unknown-freebsd12.0-c++filt   riscv64-unknown-freebsd12.0-nm        riscv64-unknown-freebsd12.0-size
riscv64-unknown-freebsd12.0-elfedit   riscv64-unknown-freebsd12.0-objcopy   riscv64-unknown-freebsd12.0-strings
riscv64-unknown-freebsd12.0-gdb       riscv64-unknown-freebsd12.0-objdump   riscv64-unknown-freebsd12.0-strip
```
I'm trying to build riscv-gcc but that just keeps failing unexpectedly with this error:
```shell
checking whether setrlimit is declared... yes
checking whether getrusage is declared... yes
checking whether ldgetname is declared... no
checking whether times is declared... yes
checking whether sigaltstack is declared... yes
checking for struct tms... yes
checking for clock_t... yes
checking for F_SETLKW... yes
checking if mkdir takes one argument... no
*** Configuration riscv64-unknown-freebsd12.0 not supported
gmake[1]: *** [Makefile:4165: configure-gcc] Error 1
gmake[1]: Leaving directory '/usr/home/blubee/dev/riscv/riscv-gnu-toolchain/build-gcc-freebsd'
gmake: *** [Makefile:183: stamps/build-gcc-freebsd] Error 2
```
The error is cryptic and the logs doesn't really have anything that pops out.

configure full log: https://pastebin.com/kwW290QG
risc-gcc config.log: https://pastebin.com/udeUn0sy

Does this look like a sane configuration
```shell
stamps/build-gcc-freebsd: $(srcdir)/riscv-gcc stamps/build-binutils-freebsd \
                               stamps/build-freebsd-headers
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
	cd $(notdir $@) && $</configure \
		--target=riscv$(XLEN)-unknown-freebsd12.0 \
		--prefix=$(INSTALL_DIR) \
		--with-sysroot=$(SYSROOT) \
		--without-newlib \
		--without-headers \
		--disable-werror \
		--disable-shared \
		--enable-threads \
		--enable-tls \
		--enable-languages=c,c++ \
		--with-gmp=/usr/local \
		--with-mpfr=/usr/local \
		--with-mpc=/usr/local \
		--disable-libatomic \
		--disable-libmudflap \
		--disable-libssp \
		--disable-libquadmath \
		--disable-libgomp \
		--disable-nls \
		--disable-bootstrap \
		@with_system_zlib@ \
		--src=$(gccsrcdir) \
		$(GCC_CHECKING_FLAGS) \
		$(MULTILIB_FLAGS) \
		$(WITH_ABI) \
		$(WITH_ARCH)
	$(MAKE) -C $(notdir $@) inhibit-libc=true all-gcc
	$(MAKE) -C $(notdir $@) inhibit-libc=true install-gcc
	$(MAKE) -C $(notdir $@) inhibit-libc=true all-target-libgcc
	$(MAKE) -C $(notdir $@) inhibit-libc=true install-target-libgcc
	mkdir -p $(dir $@) && touch $@
```
Any advice on what could be going wrong?

jim-wilson, 2018-03-20:
There is no riscv*-freebsd support in the FSF binutils/gcc trees as yet.  Binutils doesn't actually require any patch to work on FreeBSD, so that one will build as is, though one will probably be added at some point.  GCC requires a patch.  There was one submitted but not accepted yet, we are waiting for copyright assignment paperwork.
    https://gcc.gnu.org/ml/gcc-patches/2018-02/msg01066.html
And for me to catch up on my work backlog.  This patch should get into gcc-8, but may not be added to gcc-7.x, since we usually don't add new targets to production releases.

The riscv-gnu-toolchain trees are out-of-date by the way.  They need to be updated to newer FSF binutils/gcc branches, but I haven't had time to do that yet.

bukinr, 2018-03-21:
Hi Jim

I sent signed documents twice (5 Feb and then again on 5 March) but didn't receive reply.
I wonder how to check that my documents received by FSF ?

jim-wilson, 2018-03-21:
An official FSF package maintainer has a log in account at the FSF where they can check the copyright list file.  I just did that, and still don't see your name in the list.  Otherwise, you just have to keep asking the copyright clerk about the status of your paperwork, but don't ask so often that you annoy them.  It has been two weeks. so it would be OK to ask them for status.  If the delay gets to be a problem I can try asking rms what is going on.

Is there a reason why you had to send in signed documents twice?  Sending them duplicate documents might confuse the issue a bit.

It is common for them to be slow to respond, but if this drags out into April then certain I can try applying some pressure from this side.  If this takes too long we might miss the gcc-8 branch which would be unfortunate, so we should be trying to get it in before the branch is made.

gvnn3, 2018-07-02:
Bump.  It's been nearly 6 months.

jim-wilson, 2018-07-02:
riscv freebsd support was added to the FSF gcc tree in April, and ended up in riscv-gcc when it was recently upgraded to gcc-8.1.0.

Keep in mind that the official trees are at the FSF, and these trees are just minimally maintained copies of the FSF tree.  If you want something new, you are better off using the FSF trees than these trees.

closed 2018-07-02 by jim-wilson