riscv-collab/riscv-gnu-toolchain #329: Software multiplication routines missing in libgcc

minexew, 2018-03-18:
This may be just me grossly misunderstanding the build process, but no matter what I do, I haven't been able to produce a correct runtime library for the RV32I target.
The problem is specifically `__mulsi3` missing in libgcc.a, but I assume the other multiplication/division routines are missing as well. Unsurprisingly, code that uses multiplication then fails to compile with `-march=rv32i -mabi=ilp32`

I will keep trying, but in the meantime, is this a known issue?

jim-wilson, 2018-03-18:
There is no known problem here.

We don't recommend building in the source tree.  You can do something like
  mkdir build
  cd build
  ../riscv-gnu-toolchain/configure ...
to build in a separate directory.

minexew, 2018-03-19:
As it turns out, `__mulsi3` was in libgcc.a all along, hidden in plain sight in `muldi3.o`.

The actual reason for the error is that libgcc wasn't being linked in *at all*, because I was using plain `ld` for linking instead of `gcc`. Using `ld` requires not only supplying `-lgcc`, but also its absolute path through `-L`, which is however a major PITA for writing clean makefiles.
I ended up just linking with `gcc` and prefixing all linker arguments with `-Wl,`

closed 2018-03-19 by minexew

jim-wilson, 2018-03-19:
The FSF toolchain developers generally recommend using gcc instead of ld to link, as there are a lot of special linker options necessary for correct output, and the knowledge of which linker options to use is encoded in gcc.

If you know what you are doing, and want to use ld to link, you can get the libgcc path by using "gcc --print-libgcc".  You can put that inside a $(shell ...) expression, or use backticks, or whatever, to embed it in a Makefile.  Keep in mind that there is often more than one copy of libgcc, and the right one to use may depend on other gcc and/or linker options, so you may need something a little more complicated than just gcc --print-libgcc to get the right result.

minexew, 2018-03-19:
Thanks for the tips. `--print-libgcc` is new to me.

https://wiki.osdev.org/Libgcc also has some useful notes. `gcc -nostdlib -lgcc` is probably what I'll want to use, because I will be rolling my own libc functions.