riscv-collab/riscv-gnu-toolchain #330: ld problem: Symbol needs debug section which does not exist

serdarog, 2018-03-29:
We want to run a baremetal code on FPGA which does not support compressed instructions.
So I have built riscv-tools for march rv64g and mabi lp64. 

To test this, I want to compile a baremetal code, so I use freedom-e-sdk. I modified the coreplexip-e51-arty setting to 
RISCV_ARCH := rv64imafd
RISCV_ABI  := lp64

and I use the compiler I built for march rv64g and mabi lp64.

When I compile the hello project under software/hello, I get the following error. 

riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status

I see that there is a similar issue #[here](https://github.com/riscv/riscv-gnu-toolchain/issues/284) but could not find any fix for this issue.

jim-wilson, 2018-03-29:
This is a cascading error that should always be ignored.  The real error is the one that occurred immediately before this one, which you unfortunately did not report.

A patch was applied to upstream FSF Binutils to eliminate this cascading error, which will end up in riscv-gnu-toolchain when it eventually gets updated.

serdarog, 2018-03-29:
The part before this one is:

/riscv/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text+0x18): relocation truncated to fit: R_RISCV_HI20 against symbol `_global_impure_ptr' defined in .sdata section in /riscv/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)

The entire output is in the file:
[riscv-build-error.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1860640/riscv-build-error.txt)

jim-wilson, 2018-03-29:
The R_RISCV_HI20 relocation truncated error is a real problem, unfortunately there are many ways that you can hit this problem, so it doesn't narrow the problem down much.  Palmer has a blog that talks about how relocations work.
  https://www.sifive.com/blog/2017/08/07/all-aboard-part-0-introduction/
Part 2 talks about relocations and how to debug relocation truncation failures.  I'd suggest using the FSF binutils tree which has the fix for the cascading error, and then adding -Wl,--noinhibit-exec which tells the linker to produce output even though there was an error.  This allows you to get a better look at what the linker did, and maybe figure out why the relocation error occurred.

Meanwhile, I tried reproducing your problem.  I added a Makefile.local file containing
BOARD=coreplexip-e51-arty
PROGRAM=hello
I modified the bsp/env/coreplexip-e51-arty/settings.mk file use the RISCV_ARCH and RISCV_ABI options you mentioned.  I used "make tools" to build a toolchain and "make software" to build the hello project.  I get errors complaining that hard-float code can't be linked with soft-float code.  The problem here is that the toolchain is built multilib, but none of the default multilibs match your ARCH/ABI choice.  To fix this problem, I had to modify the riscv-gnu-toolchain configure rule in the toplevel Makefile.
diff --git a/Makefile b/Makefile
index 6b8e1c6..fd76de9 100644
--- a/Makefile
+++ b/Makefile
@@ -145,7 +145,8 @@ $(builddir)/riscv-gnu-toolchain/%-elf/configure.stamp:
        cd $(dir $@); $(abspath $(toolchain_srcdir)/configure) \
                --prefix=$(abspath $(dir $@)/prefix) \
                --disable-linux \
-               --enable-multilib \
+               --with-arch=$(RISCV_ARCH) \
+               --with-abi=$(RISCV_ABI) \
                --with-cmodel=medany \
                --with-libgcc-cmodel
        date > $@
gamma02:2062$ 
I did an rm -rf work to start over from scratch, ran "make tools" and "make software" as before, and this time I got a correctly linked hello program that is rv64imafd/lp64.  I can't reproduce the relocation truncated error that you saw.  So you must have done something different.  WIthout knowing exactly what you did, I can't figure out what might have gone wrong.

closed 2018-06-05 by serdarog

LUKEVINZENT, 2019-03-14:
Hi, I have the same problem.
my code is:

_**riscv32-unknown-elf-gcc -T /home/caelab/riscv-tools/riscv-tests/benchmarks/common/test.ld /home/caelab/riscv-tools/riscv-tests/benchmarks/common/syscalls.c -lc counter.o -o counter.run
/tmp/ccRFFjuF.o: In function `_init':
syscalls.c:(.text+0x62e): multiple definition of `_init'
/home/caelab/riscv-tools/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/lib/crt0.o:(.text+0x52): first defined here
/home/caelab/riscv-tools/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x8): undefined reference to `_edata'
/home/caelab/riscv-tools/lib/gcc/riscv32-unknown-elf/7.2.0/crtbegin.o: In function `deregister_tm_clones':
crtstuff.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against `__TMC_LIST__'
/home/caelab/riscv-tools/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status**_

can you help me understand what I'm doing wrong?

LUKEVINZENT, 2019-03-14:
_caelab@caelab-virtual-machine:~/file_c$ riscv32-unknown-elf-gcc -T /home/caelab/riscv-tools/riscv-tests/benchmarks/common/test.ld /home/caelab/riscv-tools/riscv-tests/benchmarks/common/syscalls.c -lc counter.o -o counter.run_

This in my command

ilia-diachkov, 2019-03-14:
It looks like the same  issue as described in https://github.com/riscv/riscv-gnu-toolchain/issues/415