riscv-collab/riscv-gnu-toolchain #333: Build Native C Compiler for RISC-V Linux

akashgj, 2018-04-17:
Please anyone share how to configure and  build native C compiler for RISC-V Linux ?

akashgj, 2018-04-18:
I want a canadian cross build. I tried with _make linux-native_. It generate the cross compiler _riscv64-unknown-linux-gnu-gcc_, but native compiler is not generated.

kito-cheng, 2018-04-18:
native compiler will installed in `<prefix>/native`, however it's lack of headers and libc, you must copy sysroot (in `<prefix>`/sysroot) into your RISC-V linux too.

kito-cheng, 2018-04-18:
BTW, it's not fully verified it's functionality, if you want to used in production level instead of just fun, I will strongly suggest you get from Fedora or Debian as Jim suggested.

jim-wilson, 2018-04-18:
If you have a target linux system that already has a glibc build, then it is fairly easy to do a toolchain build using a sysroot.  Just create a sysroot that contains /lib*, /usr/lib*, and /usr/include from the target root filesystem.  Or alternatively, if your target root file system is mounted on your build system, you can use that directly.

If you don't have a target linux system with a glibc build, then a toolchain build gets very complicated because of interdependencies between kernel/glibc/binugile/gcc which requires multiple partial builds before you can do final full builds.  It is best to use a script that someone has written.  Every linux distro has their own scripts for this.  OpenEmbedded/Yocto are popular choices for embedded linux.  There is also crosstools-ng that I have used but I'm not sure if it supports a canadian cross build.  We have the linux-native rule in our makefiles, but we don't use it, and offer no guarantees that it will work.

Here are the scripts I used last time I canadian cross built a native compiler using fedora header files.  You need to build the cross (x) binutils and gcc first, then the host (h) binutils and gcc.

```
gamma02:2007$ more x-binutils.script 
../binutils-2.30/configure --target=riscv64-unknown-linux-gnu --prefix=/scratch/
jimw/fedora-cross/install --with-sysroot=/scratch/jimw/fedora-cross/sysroot
gamma02:2008$ more x-gcc.script 
../gcc-7.3.0/configure --target=riscv64-unknown-linux-gnu --prefix=/scratch/jimw
/fedora-cross/install --with-sysroot=/scratch/jimw/fedora-cross/sysroot --disabl
e-multilib
gamma02:2009$ more h-binutils.script 
../binutils-2.30/configure --host=riscv64-unknown-linux-gnu --target=riscv64-unk
nown-linux-gnu --prefix=/scratch/jimw/fedora-cross/h-install --with-sysroot=/scr
atch/jimw/fedora-cross/sysroot
gamma02:2010$ more h-gcc.script 
../gcc-7.3.0/configure --host=riscv64-unknown-linux-gnu --target=riscv64-unknown
-linux-gnu --prefix=/scratch/jimw/fedora-cross/h-install --with-sysroot=/scratch
/jimw/fedora-cross/sysroot --disable-multilib
gamma02:2011$ 
```