riscv-collab/riscv-gnu-toolchain #334: Configure error

yosukezith, 2018-04-25:
When I do ./configure, I found "checking whether we are cross compiling... no" .
Is this problem?

$ ./configure --prefix=/opt/riscv_linux

```
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /home/cybersecuritylab/.pyenv/versions/anaconda2-4.3.1/bin//curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
```

Please help anyone.

jim-wilson, 2018-04-25:
This is not a problem.

This is a generic configure script, and it only applies to the top level directory, which is not being cross compiled.  The cross compiler itself is not cross compiled either.  But the target libraries will be cross compiled by the cross compiler you just built.  So some directories will be compiled native and some will be compiled cross.  This is normal.

closed 2018-04-25 by jim-wilson

yosukezith, 2018-04-25:
Thank you for answering the question.