riscv-collab/riscv-gnu-toolchain #340: Bump binutils to 2.30

arunthomas, 2018-06-20:
Any plans to bump binutils to 2.30 in the near future? Thanks a lot.

arunthomas, 2018-06-20:
Or plans to bump to gdb upstream --  with @timsifive's recent commits cherry-picked?

jim-wilson, 2018-06-20:
For about 6 months, we have been planning to update riscv-gnu-toolchain any day now.  Maybe it will happen this time.  Maybe it won't.

Gdb is an additional problem, as the upstream port is different than the port in riscv-gnu-toolchain, and the upstream port is not working for us as well as the current one, so we would rather not upgrade gdb until that is fixed.  Tim's patches are another issue.  I've told him that he needs to submit his patches upstream or they will be lost.  But it looks like he hasn't done that for all of his patches.

Gcc is also a problem, as there is an unresolved question as to whether we should upgrade to gcc-7.3 or gcc-8.1.

You can of course do your own upgrades locally.  You can use FSF sources directly.  If you like the riscv-gnu-toolchain build scripts, you can use FSF sources inside this tree.  For instance, in the riscv-binutils-gdb tree after a git submodule update, you can do something like
  git add remote upstream git://sourceware.org/git/binutils-gdb.git
  git fetch upstream
  git checkout -b upstream-binutils-2_30 --track upstream/binutils-2_30-branch

This may break gdb though, as I think the binutils folks only tag the binutils directories, and you may end up with incompatible sources for binutils and gdb.  If we want to track FSF releases, at some point, we really need to have two different source trees for binutils and gdb in the riscv-gnu-toolchain project.  They share the same development tree, but they do not share release trees, so to track FSF release trees, we need separate binutils and gdb source trees.  This is yet another complication.  Personally, I just use upstream/master to track the FSF top of tree.  Binutils is rarely broken, and gdb is so broken that it doesn't matter which source tree you use.

You can also do something similar with the other packages, except that the gcc top-of-tree is called upstream/trunk because gcc is still using svn as the primary repo.

timsifive, 2018-06-20:
I've been avoiding upstreaming my gdb patches without a good reason. It's
something I mean to do, but there's always something else that feels more
helpful. (Usually tweaking OpenOCD.)

arunthomas, 2018-06-21:
Thanks for the explanation, @jim-wilson. Do you happen to have a list of current known issues in upstream gdb? Perhaps we could coordinate a community effort to fix up upstream gdb?

I'll give upstream binutils-gdb a go and see what breaks for me.

jim-wilson, 2018-06-21:
My embedded cross compiler stuff has all been with simulators, and gdb doesn't really work in that case.  I've also done linux native work.  There is no native linux gdb support as yet, though I have a start on a patch for this.  

The interesting case is embedded gdb + OpenOCD connected to a target board, which I've never used, but I've heard is working poorly.  There are a number of people looking at this, including SiFive and Embecosm.  Since I've never used this, I don't know what the problems are.  This will get fixed, because we need it fixed, it is just a question of when.

There are a few common complaints I've seen.  2-byte breakpoints don't work.  I've seen this with my linux native work, and suspect a linux kernel bug.  I don't know if it is a problem for embedded work.  There is a curious error from gdb "Register 3921 is not available" which I've also seen on linux now.  I don't know if this affects embedded work.  This is an attempt to read the priv-1.9 misa after the priv 1.9.1+ misa read failed.  I believe this is a gdb bug.  The first misa read attempt is inside a TRY, but the second one isn't and ends up cancelling the command when it shouldn't.  Also the priv-1.9 misa regnum upstream is wrong, as it is missing the +65 required as the CSRs come after the first 65 registers.  On the linux side, there is no ptrace support for FP registers or CSRs or anything other than the integer registers and pc.  I've already found and fixed one linux kernel riscv ptrace bug.  The upstream command to disable 2-byte breakpoints doesn't work because it only disables one of the two hooks that control breakpoint size.

I suspect the embedded gdb+OpenOCD issues are very different from the linux native ones.

jim-wilson, 2018-07-06:
We now have separate binutils and gdb source and build trees, and riscv-binutils has been upgraded to FSF binutils-2.30.

closed 2018-07-06 by jim-wilson

ilg-ul, 2018-07-06:
> We now have separate binutils and gdb source ...

Am I missing something or the two are still sharing the same git repo, and the 'separation' is in fact a 'duplication'?

jim-wilson, 2018-07-06:
There is one github repo, riscv-binutils-gdb.  In the checked out riscv-gnu-toolchain tree, riscv-binutils and riscv-gdb are two different source trees checked out from the same repo, from two different branches.  This is necessary because binutils and gdb are different projects, with different release schedules and different release branches.  You can not (easily) have a single source tree that works for both binutils and gdb if you want to track official FSF binutils and gdb releases.  You can have a single source tree if you want to track the development trees, because binutils and gdb share the same development tree.  But if we track the development tree, then we run the risk of occasional breakage.  Also, the FSF tends to discourage people from making official releases off of development trees.  Now that we are mostly upstream, we should be trying to follow upstream rules for releases.

ilg-ul, 2018-07-06:
Thank you Jim for the explanation. 

Indeed, if the upstream maintains both binutils and gdb in the same git repo, there is not much to do to avoid the duplication.

I was confused because in the arm-none-eabi build script, ARM packs binutils and gdb as separate archives, and I thought that they come from separate repos.