riscv-collab/riscv-gnu-toolchain #341: Searching for explainations to define a new instruction in the "struct riscv-opcode "

arkhan91, 2018-06-20:
Hi, 

I am trying to interpret the statements mentioned in "const struct riscv_opcode riscv_opcodes[]" in riscv-opc.h.  
for example  
````
const struct riscv_opcode riscv_opcodes[] =

{
/* name,      isa,   operands, match, mask, match_func, pinfo.  */
....
{"and",       "I",   "d,s,j",  MATCH_ANDI, MASK_ANDI, match_opcode, INSN_ALIAS },
....
{"add",       "I",   "d,s,t",  MATCH_ADD, MASK_ADD, match_opcode, 0 }
....
} ````

Is there any document or source where i can find how to pass write statement specially operands.
I trying to append the custom instruction .

aswaterman, 2018-06-20:
Not really, as this is an internal interface.  I suggest finding an instruction that has identical operand formats and copying it.

Adding new operand formats requires modifying several places in tc-riscv.c and riscv-dis.c.  I wouldn't recommend it unless truly necessary.

arkhan91, 2018-06-20:
Can you please elaborate meaning of general operands like "d,s,t" ,"d,s,j" etc,
thanks

aswaterman, 2018-06-20:
I can’t document them all, but offhand I know that:

d = integer rd
s = integer rs1
t = integer rs2
j = 12-bit I-type immediate

closed 2018-06-20 by arkhan91

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2