riscv-collab/riscv-gnu-toolchain #349: build error with both gcc 7.3 and gcc 8.1

allenlorenz, 2018-07-29:
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverlo
aded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/genmdd
eps \
    build/genmddeps.o build/read-md.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/read-md.o: In function `md_reader::~md_reader()':
/home/alorenz/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc/../.././riscv-gcc/gcc/read-md.c:1049: undefined reference to `operator delete(void*, unsigned long)'
build/read-md.o: In function `noop_reader::~noop_reader()':
/home/alorenz/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc/../.././riscv-gcc/gcc/read-md.h:264: undefined reference to `operator delete(void*, unsigned long)'
collect2: error: ld returned 1 exit status

jim-wilson, 2018-07-29:
There appears to be something wrong on your side.  The missing operator delete function should come from libstdc++.  Libstdc++ is included automatically in the link when using g++, and an option forces use of the static version instead of the dynamic version of the library.

You could try dropping this -static-libstdc++ option from the link line to see if it helps.  This option is useful for native bootstraps, but should not be needed for cross compiler builds.  If dropping the option helps, then that suggests that there is a problem with your static copy of libstdc++.

The reference to 7.3 and 8.1 is ambiguous.  Is this the compiler sources you are trying to build, or the compiler you are building with?  I'll assume it is compiler sources, which suggests that it isn't the compiler sources that are broken, if two different compiler trees fail to build the same way.

What OS is your build host?  And what does "g++ --version" say?

allenlorenz, 2018-07-29:
Yes most problems I have are my own ;)
Thank you for quick response.  I am running fedora 28, upgraded from 27, and have verified up to date.

g++ --version 
g++ (GCC) 8.1.1 20180712 (Red Hat 8.1.1-5)

I did following .. clone, prerequisite,  configure, make.
1) $ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
2) sudo yum install autoconf automake libmpc-devel mpfr-devel gmp-devel gawk  bison flex texinfo patchutils gcc gcc-c++ zlib-devel expat-devel
3) ./configure --prefix=/opt/riscv
4) make

given I didn't see how I could have screwed up these instruction and 20+ min of googling did NOT help prompted posting the issue.

 I will see about searching in the makefile or configure file for dropping the -static-libstdc++ option.

allenlorenz, 2018-07-29:
dough .. the simple way.... 
but first find the right directory..
going to riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc
and cutting and pasting the command  failed....( good )
removing -static-libstdc++  give success... 

so I have NOT modified the build scripts, so why would static library be used over dynamic ? 
(As a rule I get in trouble when not using the defaults! )

 will do a little yum searching, and see bout reinstall libstdc++

allenlorenz, 2018-07-29:
sudo dnf list installed | grep libstdc++
libstdc++.i686                         8.1.1-5.fc28                @updates     
libstdc++.x86_64                       8.1.1-5.fc28                @updates     
libstdc++-devel.x86_64                 8.1.1-5.fc28                @updates 
looks normal.

jim-wilson, 2018-07-29:
libstdc++ is part of gcc.  If we link dynamically, and then build a new shared libstdc++, or vice versa, there might be issues with using the wrong one at run-time.

allenlorenz, 2018-07-30:
Thanks for the help.  i will access to a different machine in a couple of days, and see about doing some other tests, then report back.

allenlorenz, 2018-07-31:
Dough .. the answer was "obvious"... and Jim did almost point it out.    For Fedora you also need to   add    libstdc++-static  to the yum / dnf  Prerequisites.

closed 2021-03-04 by lazyparser