riscv-collab/riscv-gnu-toolchain #350: /bin/sh: 1: /home/*/riscv/riscv-gnu-toolchain/riscv-binutils/configure: not found

Pinpiew, 2018-07-31:
rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv64-unknown-elf-gcc /home/*/riscv/riscv-gnu-toolchain/riscv-binutils/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/opt/r1 \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-libreadline
**/bin/sh: 1: /home/*/riscv/riscv-gnu-toolchain/riscv-binutils/configure: not found**
Makefile:375: recipe for target 'stamps/build-binutils-newlib' failed
make: *** [stamps/build-binutils-newlib] Error 127

jim-wilson, 2018-07-31:
You haven't provided enough info to understand what the problem is.  I need enough info to reproduce.  How exactly did you configure?  What should the /home//riscv/riscv-gnu-toolchain/riscv-biutils/configure path be instead?  Why did you use /home/*/riscv in the title?  Where did the * come from?

The Makefile will be trying to run $(srcdir)/riscv-binutils/configure where srcdir is defined on the first line of the Makefile.  Is srcdir wrong?

if riscv-binutils/configure is missing, then the problem is that you did a git pull or git clone, and need to follow it with a "git submodule update --init --recursive" command.

Pinpiew, 2018-08-01:
execute the following, and run OK
git submodule update --init --recursive

closed 2018-08-01 by jim-wilson