riscv-collab/riscv-gnu-toolchain #352: Build error with Linux multilib

zhuanhao-wu, 2018-08-09:
I used these commands to build with Linux multilib
```
./configure --prefix=/opt/riscv --enable-multilib
make linux
```

The source code is located in `/opt/riscv-gnu-toolchain`.

But the build is interrupted by these error messages:
```
rm -rf stamps/build-glibc-linux-rv32imac-ilp32 build-glibc-linux-rv32imac-ilp32
mkdir build-glibc-linux-rv32imac-ilp32
cd build-glibc-linux-rv32imac-ilp32 && \
	CC="riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32" \
	CFLAGS=" -mcmodel=medlow -g -O2 -march=rv32imac -mabi=ilp32" \
	CXXFLAGS=" -mcmodel=medlow -g -O2 -march=rv32imac -mabi=ilp32" \
	ASFLAGS=" -mcmodel=medlow -march=rv32imac -mabi=ilp32" \
	/opt/riscv-gnu-toolchain/riscv-glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/opt/riscv-gnu-toolchain/linux-headers/include \
	--enable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib32/ilp32 libc_cv_slibdir=/lib32/ilp32 libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32 accepts -g... yes
checking for gcc... gcc
checking for riscv32-unknown-linux-gnu-readelf... riscv32-unknown-linux-gnu-readelf
checking for riscv32-unknown-linux-gnu-g++... no
checking for riscv32-unknown-linux-gnu-c++... no
checking for riscv32-unknown-linux-gnu-gpp... no
checking for riscv32-unknown-linux-gnu-aCC... no
checking for riscv32-unknown-linux-gnu-CC... no
checking for riscv32-unknown-linux-gnu-cxx... no
checking for riscv32-unknown-linux-gnu-cc++... no
checking for riscv32-unknown-linux-gnu-cl.exe... no
checking for riscv32-unknown-linux-gnu-FCC... no
checking for riscv32-unknown-linux-gnu-KCC... no
checking for riscv32-unknown-linux-gnu-RCC... no
checking for riscv32-unknown-linux-gnu-xlC_r... no
checking for riscv32-unknown-linux-gnu-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... no
checking whether g++ accepts -g... yes
checking whether g++ can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm hppa i386 m68k microblaze mips nios2 powerpc riscv s390 sh sparc tile x86_64 
configure: running configure fragment for add-on libidn
checking for -fstack-protector... yes
checking for -fstack-protector-strong... yes
checking for -fstack-protector-all... yes
checking for assembler and linker STT_GNU_IFUNC support... no
checking for gcc attribute ifunc support... no
checking sysdep dirs... sysdeps/unix/sysv/linux/riscv/rv32/ sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic/wordsize-32 sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/riscv/rv32 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv32/ sysdeps/riscv/rv32 sysdeps/riscv sysdeps/init_array sysdeps/ieee754/ldbl-128 sysdeps/ieee754/dbl-64 sysdeps/ieee754/flt-32 sysdeps/wordsize-32 sysdeps/ieee754 sysdeps/generic
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln -s works... yes
checking whether /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/as is GNU as... yes
checking whether /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld is GNU ld... yes
checking for /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/as... /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/as
checking version of /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/as... 2.30, ok
checking for /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld... /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld
checking version of /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld... 2.30, ok
checking for gnumake... no
checking for gmake... no
checking for make... make
checking version of make... 4.1, ok
checking for gnumsgfmt... no
checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.7, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 6.1, ok
checking for sed... /bin/sed
checking version of /bin/sed... 4.2.2, ok
checking for gawk... /usr/bin/gawk
checking version of /usr/bin/gawk... 4.1.3, ok
checking if riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32 is sufficient to build libc... yes
checking for riscv32-unknown-linux-gnu-nm... riscv32-unknown-linux-gnu-nm
checking for python3... python3
checking LD_LIBRARY_PATH variable... ok
checking for bash... /bin/bash
checking for perl... /usr/bin/perl
checking for install-info... /usr/bin/install-info
checking for bison... bison
checking version of bison... 3.0.4, ok
checking for .set assembler directive... no
checking linker support for protected data symbol... no
checking for broken __attribute__((alias()))... no
checking whether to put _rtld_local into .sdata section... no
checking whether to use .ctors/.dtors header and trailer... configure: error: missing __attribute__ ((constructor)) support??
Makefile:226: recipe for target 'stamps/build-glibc-linux-rv32imac-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imac-ilp32] Error 1
```

My gcc version:
```
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I am building with Ubuntu 16.04.

Thanks

zhuanhao-wu, 2018-08-09:
I also tried 
```
./configure --prefix=/opt/riscv --enable-multilib libc_cv_ctors_header=yes
make linux
```

But it doesn't work.

zhuanhao-wu, 2018-08-09:
It seems that the linker is only providing `elf32lriscv` and `elf64lriscv`, but not `elf32lriscv_ilp32` etc.

jim-wilson, 2018-08-09:
​Most of the FSF GCC riscv related patches have been backported to the riscv-gnu-toolchain tree, but the FSF Binutils riscv related patches have not.  Unfortunately, one of the gcc patches requires a binutils patch.  That is triggering the failure you see.  I think it only shows up with a multilib linux target, which probably hasn't been tested recently.​  The binutils patch you need is
    https://sourceware.org/ml/binutils/2018-05/msg00043.html
    https://sourceware.org/ml/binutils-cvs/2018-05/msg00042.html
The already backported GCC patch that relies on this is
    https://gcc.gnu.org/ml/gcc-patches/2018-05/msg00249.html

You can fix this by either adding the missing binutils patch or removing the gcc patch.  A more proper fix is to backport all of the FSF binutils patches, just like we already did for FSF GCC.

zhuanhao-wu, 2018-08-10:
Thank you for pointing out the patch

nuaaysh, 2018-08-13:
Could any one give me some suggestions about how to use the binutils patch?
I am not familiar with gcc. I try to understand the command 'patch' but failed.

palmer-dabbelt, 2018-08-13:
It's been merged, but for future reference something like `curl https://github.com/riscv/riscv-gnu-toolchain/pull/354/commits/08380d8f32719b33ed603d8ba0c5bc8ebc6e6ac4.patch | git am -C riscv-binutils -3s` is what I usually do.

jim-wilson, 2018-08-13:
Fixed by #354

closed 2018-08-13 by jim-wilson