riscv-collab/riscv-gnu-toolchain #356: can't link hard-float modules with soft-float modules

billasurendra, 2018-08-14:
I am trying to cross compiling OpenCV application for riscv architecture.I build a tool chain with ### **build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv64ima
build_project riscv-gnu-toolchain --prefix=$RISCV --with-arch=rv64ima **

Building a tool chain process going well but when I am going to cross compiling OpenCV application, I am getting errors.I want a floating point operation  enable..

Error :-1: 

/home/billa/Pictures/riscv/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: test.o: can't link hard-float modules with soft-float modules
/home/billa/Pictures/riscv/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: failed to merge target specific data of file test.o

Can any one please give me solution to my problem .

Thanks in Advance.

billasurendra, 2018-08-16:
> The commands you mentioned are for riscv-tools, but I don't recommend
> using riscv-tools, as it is poorly maintained and difficult to build. I
> would recommend a direct git clone of riscv-gnu-toolchain to get the
> current version of the compiler, and better build scripts. gcc-7.2.0 is
> obsolete by the way, and known to have some bugs. You should use either
> gcc-7.3 or gcc-8, which you can get from riscv-gnu-toolchain.
>
> If you configure for rv64ima then you get a soft-float toolchain, with
> libraries compiled for soft-float. If you then use -march=rv64gc you get
> hard float code, which can't be linked with libraries compiled for
> soft-float, as they use different ABIs.
>
> One solution is to build a multilib toolchain, i.e. build multiple copies
> of every library, compiled with for different architectures and ABIs. You
> can do this by adding --enable-multilib. This will build a default
> selection of library arch/abi choices, one of which may be OK for what you
> are trying to do.
>
> Another solution is to build a compiler for the arch/abi choices that you
> want to use when compiling.

As you mentioned above,I am following this procedure to install riscv-tool
chain

1. mkdir riscv-tool
2. cd riscv-tool/
3. git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
4. .cd riscv-gnu-toolchain
5. ./configure --prefix=/opt/riscv --enable-multilib
6. make newlib -j8
7. make linux -j8

Here my doubt is if I am following, where I want use this "rv64ima" option
to make use of floating point support for cross compiling opencv library
and cross compiling opencv application..Please give me reply

Thanks in advance

jim-wilson, 2018-08-16:
If you are using rv64ima then you aren't using floating point instructions, you are using soft-float libraries.  You need to decide if you are using an FPU or not.  This is a little confusing.

Since you are using --enable-multilib, you are getting multiple library builds.  You can see what the library choices are by using the gcc --print-mulitlib option.  Not every possible combination is built.  If you try to use a combination that isn't supported by default, then you might run into trouble.  Note that the newlib target builds lots of combinations, but the linux target builds only a few.

When you run gcc, you can use the -march=$ARCH and -mabi=$ABI options to choose what architecture and ABI you want.  If you are using one of the supported combinations, then this should work fine.

If you want a specific architecture and ABI, then you can use --with-arch=$ARCH and --with-abi=$ABI when configuring to get a toolchain that matches your specific requirements.  This may work better if you are using a combination that isn't supported by default.

billasurendra, 2018-08-17:
I am literally confused, what command and which flags I want to use for my
tool chain working with  FPU.As you said I have used following commands in
tool chain build script.

build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV
--with-isa=rv64ima
build_project riscv-gnu-toolchain --prefix=$RISCV --with-arch=$ARCH
--with-abi=$ABI

Tool chain build successfully but when I run Opencv c++ Program and I
generated  objdump file but here there is no FPU operations.

Please give me solution to my problem

Thanks in Advance

jim-wilson, 2018-08-17:
If you have a recent linux distro, like Ubuntu 18.04 or Debian testing (I think), you can just install a package to get a RISC-V compiler.  E.g.
```
sudo apt install gcc-7-riscv64-linux-gnu
```
That would be easier than trying to build your own.

You might want to read up a little on the RISC-V architecture.  There is a book called "The RISC-V Reader" which is a good introduction to the architecture.  It explains what the architecture and ABI choices are.

jim-wilson, 2018-08-17:
Palmer has a blog series that talks about RISC-V toolchains.  That is another place to get info.
https://www.sifive.com/blog/2017/08/07/all-aboard-part-0-introduction/

leeehui, 2018-09-28:
Hi, jim-wilson
I got confused when compiling riscv-gnu-tools.

I am using riscv core with rv64imc,  I compiled the tools with "./configure --enable-multilib" (default --with-arch=rv64gc --with-arch=lp64d)according to one of your reply.  there are still errors about "soft-float hard-float".

It turns out that I must explicitly use --with-arch=rv64imc and --wit-abi=lp64. 

So, can I say --enable-multilib has nothing to do with "fd" related issues?

jim-wilson, 2018-09-28:
There are a limited number of default multilibs built. it is impractical to build hundreds of different libraries for hundreds of different arch/abi combinations.  If you try to compile for a target which isn't one of the default multilibs, then this could fail.  You can see the default list of multilibs with
gcc --print-multi-lib
If you don't see your target listed, then you will need to change the multilib list to include it, or the easier solution is to just configure explicitly for that target.

leeehui, 2018-09-29:
got it! 
I prefer to explicitly configure what I want, 
Compiling the tools is a little time-consuming on my virtual machine.
thx!

closed 2021-03-04 by lazyparser