riscv-collab/riscv-gnu-toolchain #357: make linux error (when trying to build the 32-bit RV32GC toolchain)

nuaaysh, 2018-08-20:
This issue looks like #300 and #257 
root@flexpretnuaa:/home/flexpretnuaa/riscv-gnu-toolchain# ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
root@flexpretnuaa:/home/flexpretnuaa/riscv-gnu-toolchain# make linux
rm -rf stamps/build-glibc-linux-rv32gc-ilp32d build-glibc-linux-rv32gc-ilp32d
mkdir build-glibc-linux-rv32gc-ilp32d
cd build-glibc-linux-rv32gc-ilp32d && \
		CC="riscv32-unknown-linux-gnu-gcc " \
		CFLAGS=" -mcmodel=medlow -g -O2 " \
		CXXFLAGS=" -mcmodel=medlow -g -O2 " \
		ASFLAGS=" -mcmodel=medlow " \
		/home/flexpretnuaa/riscv-gnu-toolchain/riscv-glibc/configure \
		--host=riscv32-unknown-linux-gnu \
		--prefix=/usr \
		--disable-werror \
		--enable-shared \
		--enable-obsolete-rpc \
		--with-headers=/home/flexpretnuaa/riscv-gnu-toolchain/linux-headers/include \
		--disable-multilib \
		--enable-kernel=3.0.0 \
		 \
		--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... configure: error: in `/home/flexpretnuaa/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [stamps/build-glibc-linux-rv32gc-ilp32d] error 1
——————————————————————————————————————————————
The config.log in `/home/flexpretnuaa/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Library configure (see version.h), which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /home/flexpretnuaa/riscv-gnu-toolchain/riscv-glibc/configure --host=riscv32-unknown-linux-gnu --prefix=/usr --disable-werror --enable-shared --enable-obsolete-rpc --with-headers=/home/flexpretnuaa/riscv-gnu-toolchain/linux-headers/include --disable-multilib --enable-kernel=3.0.0 --libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib

## --------- ##
## Platform. ##
## --------- ##

hostname = flexpretnuaa
uname -m = x86_64
uname -r = 3.13.0-24-generic
uname -s = Linux
uname -v = #46-Ubuntu SMP Thu Apr 10 19:11:08 UTC 2014
[... 277 lines cut ...]

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "GNU C Library"
#define PACKAGE_TARNAME "glibc"
#define PACKAGE_VERSION "(see version.h)"
#define PACKAGE_STRING "GNU C Library (see version.h)"
#define PACKAGE_BUGREPORT "http://sourceware.org/bugzilla/"
#define PACKAGE_URL "http://www.gnu.org/software/glibc/"
#define PKGVERSION "(GNU libc) "
#define REPORT_BUGS_TO "<http://www.gnu.org/software/libc/bugs.html>"

configure: exit 1
——————————————————————————————————————————————
It says 'riscv32-unknown-linux-gnu-gcc: command not found'.
I find I only have the folders (in /opt/riscv) 'riscv64-unknown-elf' and 'riscv64-unknown-linux-gnu', there are no any informations about riscv32.

(*I have used the commands './configure --prefix=/opt/riscv; make' and ./configure --prefix=/opt/riscv --enable-multilib make linux', it can work*)

What should I do?  Could you give some suggestions about how to use appropriate commands to build the the 32-bit RV32GC toolchain.

Thanks a lot

nuaaysh, 2018-08-20:
I find this question when I try to use other project (https://github.com/pretis/flexpret), I use the command 'make run', but it tells me 
'riscv64-unknown-elf-gcc -m32 -Itests/include -c tests/isa/add.S -o tests/isa/build/emulator/add.o
riscv64-unknown-elf-gcc: error: unrecognized command line option '-m32' ', 
So I try to build the 32-bit RV32GC toolchain, I hope this information can help you.

jim-wilson, 2018-08-20:
-m32 isn't a valid riscv compiler option.  The flexpret makefile would have to be changed to use -march=rv32gc -mabi=ilp32d for riscv instead.

There was a bug fixed a week ago that prevented 32-bit linux builds from working.  That broke both a default 64-bit configure with --enable-linux --enable-multilib and an explicit 32-bit configure with --enable-linux.  See for instance issue #352.  If this is the same problem, and you have a tree checked out more than a week ago, then all you need is a git pull, and maybe also a git submodule update --init --recursive.

The glibc config.log messages are curious.  I'd expect a message about a missing linker emulation.  It appears that riscv32-unknown-linux-gnu-gcc does exist, but gives a non-existent file message when you try to run it.  That suggests a problem with a missing dynamic linker, ld.so.  This might be a different problem.  Try running the riscv32-unknown-linux-gnu-gcc program manually to see what happens.  Also try running ldd on the binary in your install tree to see if it complains about a missing file.

nuaaysh, 2018-09-11:
Thanks! 
I must apologize this message is so late. 

I am trying to build the RISCV compiler.
The author of flexpret told me "RISC-V toolchain version this branch is developed against: https://github.com/riscv/riscv-gnu-toolchain/commit/9a8a0aa98571c97291702e2e283fc1056f3ce2e2"
I download from this link "https://github.com/riscv/riscv-gnu-toolchain/tree/9a8a0aa98571c97291702e2e283fc1056f3ce2e2" 
However when I run the command "make linux" (to build Installation (Linux)), it returns:
*************************************************************************************
....
checking build system type... x86_64-unknown-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc
checking for suffix of object files... configure: error: in `/home/flexpretnuaa/riscv-gnu-toolchain/build-glibc-linux-headers':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [stamps/build-glibc-linux-headers] error 1

the file config.log (the path:~/riscv-gnu-toolchain/build-glibc-linux-headers/):
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Library configure (see version.h), which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /home/flexpretnuaa/riscv-gnu-toolchain/src/glibc/configure --host=riscv64-unknown-linux-gnu --prefix=/usr libc_cv_forced_unwind=yes libc_cv_c_cleanup=yes --enable-shared --enable-__thread --disable-multilib --enable-kernel=2.6.32

## --------- ##
## Platform. ##
## --------- ##

hostname = flexpretnuaa
uname -m = x86_64
uname -r = 3.13.0-24-generic
uname -s = Linux
uname -v = #46-Ubuntu SMP Thu Apr 10 19:11:08 UTC 2014

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/riscv/bin
PATH: /home/flexpretnuaa/riscv-gnu-toolchain/scripts/wrapper
PATH: /opt/riscv/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2365: checking build system type
configure:2379: result: x86_64-unknown-linux-gnu
configure:2399: checking host system type
configure:2412: result: riscv64-unknown-linux-gnu
configure:2441: checking for riscv64-unknown-linux-gnu-gcc
configure:2457: found /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc
configure:2468: result: riscv64-unknown-linux-gnu-gcc
[... 515 lines cut ...]
pdfdir='${docdir}'
prefix='/opt/riscv'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define HAVE_LIBGMP 1
#define HAVE_LIBMPFR 1
#define HAVE_LIBMPC 1

configure: exit 0

jim-wilson, 2018-09-17:
The compiler error is here
configure:2784: riscv64-unknown-linux-gnu-gcc -c conftest.c >&5
Assembler messages:
Fatal error: I' must be the first ISA subset name specified (got r) 

So either there is something wrong with the options the compiler is passing to the assembler, or there is something wrong with the assembler parsing options.  You can see what options gcc passes to the assembler by adding a -v option.

You mentioned using 
https://github.com/riscv/riscv-gnu-toolchain/commit/9a8a0aa98571c97291702e2e283fc1056f3ce2e2
Looking at this, I see it is from March 2015.  That is a very long time ago.  RISC-V development is moving so quickly that nothing older than gcc-7.3 from January of this year is likely to be useful.

But in the config log, I see references to gcc-8.2.  If you are trying to mix tools from 2018 and 2015 that is certainly not going to work.  There have been a lot of changes, mostly due to upstreaming the tools, so  lots of things work slightly differently than they used to, and as a result mixing versions won't work.

lazyparser, 2021-03-04:
@nuaaysh Hi, can we close this issue, if you have fixed the error?

closed 2021-03-08 by lazyparser