riscv-collab/riscv-gnu-toolchain #358: Repositories not available

shubhankar039, 2018-08-20:
riscv-binutils @ 0d2fb1b
riscv-dejagnu @ 2e99dc0
riscv-gcc @ 489c9e7
riscv-gdb @ 635c14e
riscv-glibc @ 2f626de
riscv-newlib @ 320b28e
These repositories are not available.

jim-wilson, 2018-08-20:
These repositories are most certainly available.  Why do you think that they are not available?  What exact command did you type, and what exact error did you get?  See for instance
    https://github.com/riscv/riscv-gcc/commit/489c9e74775b580b9cb57adc80ef3277af5fee11
as proof that the gcc one exists.

Maybe you forgot to run this command?
  git submodule update --init --recursive

shubhankar039, 2018-08-20:
Hi,  
I ran the git submodule update --init --recursive command 
after cloning the repo on https://github.com/riscv/riscv-tools
After the whole process, I found that the riscv-binutils, riscv-dejagnu, riscv-gcc
riscv-gdb, riscv-glibc, and riscv-newlib folders are empty. 
What should I do?

I ran the following commands: 
git clone https://github.com/riscv/riscv-tools.git
git submodule update --init --recursive

Please check this page: https://github.com/riscv/riscv-gnu-toolchain
I am not able to view the codes of riscv-gcc from here.

jim-wilson, 2018-08-20:
I just checked out riscv-gnu-toolchain using both git and https and had no problem.  I see you are checking out riscv-tools which I don't recommend, but I tried that too and that also worked.

I do see that links are broken on the riscv-gnu-toolchain github page.  That looks like a github problem.  The linked to files are actually there.

If something went wrong, then you should have received an error when you typed one of your commands.  I can't help you if I don't know what error you are getting.  If you aren't getting explicit errors, you may need to use git debugging commands to figure out what is wrong.

Or maybe this was a temporary github problem and you just need to try it again?

Did you remember to cd into riscv-tools before running the git submodule command?

shubhankar039, 2018-08-20:
Thanks, solved

closed 2018-08-20 by shubhankar039