riscv-collab/riscv-gnu-toolchain #362: lib/gcc/riscv32-unknown-elf/8.2.0 is currently compiled using double-float even though the 8.1.0 version compiles to soft-float

novakboskov, 2018-08-23:
After building the toolchain using:
```shell
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32
make
```
the binaries within `lib/gcc/riscv32-unknown-elf/8.2.0` are compiled using `double-float` even though the `8.1.0` counterparts are compiled for `soft-float`:
```shell
$ readelf -h lib/gcc/riscv32-unknown-elf/8.2.0/crtbegin.o | grep Flags
Flags:                             0x5, RVC, double-float ABI
$ readelf -h lib/gcc/riscv32-unknown-elf/8.1.0/crtbegin.o | grep Flags
Flags:                             0x1, RVC, soft-float ABI
```

I've noticed this when tried to compile a trivial program using the latest LLVM backend (https://github.com/lowRISC/riscv-llvm) which internally invokes GNU's linker from this toolchain.

# Reproduce steps
- build LLVM backend as described (https://github.com/lowRISC/riscv-llvm#how-can-i-build-upstream-llvmclang-and-use-it-to-cross-compile-for-a-riscv32-target)
- try compilation using `clang`
- error is:
```shell
lib/gcc/riscv32-unknown-elf/8.2.0/../../../../bin/riscv32-unknown-elf-ld: lib/gcc/riscv32-unknown-elf/8.2.0/crtend.o: can't link hard-float modules with soft-float modules
``` 

# Hot fix
Replace the content of `8.2.0` directory with the content of `8.1.0`.

jim-wilson, 2018-08-23:
I checked out a copy of riscv-gnu-toolchain, configured it using --with-arch=rv32gc --with-abi=ilp32 , and built it.  Looking in the install tree, in lib/gcc/riscv32-unknown-elf/8.2.0, I see that crtbegin.o was compiled soft-float as expected.

I would suspect that you did something wrong.  If you want me to look at this, you will have to demonstrate a set of commands starting from an empty directory that produce a bad crtbegin.o.  Otherwise, I don't believe that there is a problem here.

If you kept a build log, you might try looking at the commands that were used to build the crtbegin.o file, and try to figure out what went wrong.

You mentioned that you have 8.1.0 and 8.2.0 files in the same install tree.  It is possible that something went wrong from installing into an already populated install tree.  Maybe something was used from the install tree at build time instead of from the build tree, and something was configured differently in the install tree.  I'd suggest trying to use a new --prefix option to see if that makes a difference.

novakboskov, 2018-08-24:
Yes, you're absolutely right. Probably the problem happens because I run `make` multiple times with the same non-empty directory. I tried to do that as a way to update.
If I compile to a clean directory I get only `8.2.0` and it uses `soft-float`.
I'm closing the issue.

BTW, what is the reason why make does not flush the `--prefix` directory as the first step here?

closed 2018-08-24 by novakboskov

jim-wilson, 2018-08-24:
It is possible to build multiple versions of gcc and install them into the same tree and have them all work.  Of course, that is the reason the FSF makefile doesn't clean the install tree.

As for riscv-gnu-toolchain, I doubt that anyone ever thought about this.  There is a potential problem here with what to remove from the install tree.  If someone is building multiple packages with the same prefix, then it would be wrong to remove those other packages from the install tree.  So we can't rm -rf the whole install tree.  If you are using different versions of the toolchain, then we can't safely identify all files that might be from a different toolchain version, so we can't remove that easily.  We could perhaps try to remove the current version, but again there is the problem of identifying the files to remove.  I think we can only tell what files to remove after we have built the toolchain, and by then it is too late.  There might also be problems with different configurations of the same version building different files which there is no easy way to handle.  I don't think that there is any easy solution to this.

novakboskov, 2018-08-24:
Thank you for the clarification.