riscv-collab/riscv-gnu-toolchain #363: riscv32-unknown-elf-gcc compile dhrystone.c

chlizheng, 2018-08-24:
Hi,
I use the command  "riscv32-unknown-elf-gcc dhry_1.c -march=rv32im -mabi=ilp32 -o- -S -O3" to compile the c procedure. And then  I get the Pseudoinstruction, but I want to get the Base Instrucion.Do you what command should I use to get the Base Instruction?
best wishes!

jim-wilson, 2018-08-24:
Compile with -c to get an object file, and then use objdump -dr to see the instructions plus relocations after assembler macros are expanded.

palmer-dabbelt, 2018-08-24:
There's also something like `-Mno-aliases`, which when passed to objdump will only report instructions that are part of the RISC-V ISA (as opposed to instruction aliases).  For example, with `-Mno-aliases` you'll get `jr x1` instead of `ret`.

jim-wilson, 2018-08-25:
You should define what you mean by "pseudoinstruction".  Give an example.  objdump does produce actual instructions not psuedo instructions, so you are probably misinterpreting the objdump output.

chlizheng, 2018-08-29:
Thank you,I have resolve it.
And then I want to compile rv32EM instructions.But the riscv-gcc only support -march=rv32im and not support -march=rv32em, so there are 32 registers in the result instructions at least.Do you know how to compile rv32EM instructions?
best wishes

chlizheng, 2018-08-29:
@palmer-dabbelt 
You may use objdump -dr --dessembler-options=no_aliases

jim-wilson, 2018-08-29:
rv32e support was submitted to the FSF in May, for both binutils and gcc.  The gcc support has since been backported.  The binutils support is in the process of being backported via issue #367.  With that checked in, you can use for instance "-march=rv32em -mabi=ilp32e" to get rve code.  We don't have multilibs for rve, so you will have to configure a toolchain for rve in order to be able to link.

kito-cheng, 2020-10-12:
Close this issue, seems issue resolved and long time no further discussion/response.

closed 2020-10-12 by kito-cheng