riscv-collab/riscv-gnu-toolchain #365: encounter Syntax Error in Makefile when do make report for newlib

zidanemarks, 2018-08-26:
Ubuntu 18.04:
toolchain commit id is  : 
commit 5dd463359192d07b19fd9cbac87edca37dd9fc9f (HEAD -> master)
my command steps is as below :
  $export PATH= $PATH:/opt/riscv/bin
  $eport RISCV=/opt/riscv
  $mkdir build-rv32ima-newlib
  $cd build-rv32ima-newlib
  $../riscv-gnu-toolchain/configure --prefix=$RISCV --disable-linux --with-arch=rv32imac
  $make
before we do make report , i have do make distclean in 
/riscv-binutils-gdb/
/riscv-dejagnu/
/riscv-gcc/
/riscv-glibc/
/riscv-newlib/
/riscv-qemu/
and then , i exectute make report at my work directory:
and i encouter a syntax error as below:

********************************************************************************************************
# of expected passes		89657
# of expected failures		318
# of unsupported tests		4635
/home/project/build_rv32imac/build-gcc-newlib-stage2/gcc/xg++  version 7.2.0 (GCC) 

make[3]: Leaving directory '/home/project/build_rv32imac/build-gcc-newlib-stage2/gcc'
make[2]: Leaving directory '/home/project/build_rv32imac/build-gcc-newlib-stage2/gcc'
make[1]: Leaving directory '/home/project/build_rv32imac/build-gcc-newlib-stage2'
mkdir -p stamps/
date > stamps/check-gcc-newlib
stat /home/project/build_rv32imac/../riscv-gnu-toolchain/test/gcc-newlib/rv32imac-ilp32.log || exit 1
  File: /home/project/build_rv32imac/../riscv-gnu-toolchain/test/gcc-newlib/rv32imac-ilp32.log
  Size: 1006      	Blocks: 8          IO Block: 4096   regular file
Device: 818h/2072d	Inode: 530184      Links: 1
Access: (0644/-rw-r--r--)  Uid: ( 1000/shawntan)   Gid: ( 1000/shawntan)
Access: 2018-08-23 15:07:54.313550330 +0800
Modify: 2018-06-12 10:44:53.086896309 +0800
Change: 2018-06-12 10:44:53.086896309 +0800
 Birth: -
if find build-gcc-newlib-stage2/gcc/testsuite/ -iname '*.sum' | xargs grep ^FAIL | sort | grep -F -v --file=/home/project/build_rv32imac/../riscv-gnu-toolchain/test/gcc-newlib/rv32imac-ilp32.log; then false; else true; fi
if [[ "$(find build-gcc-newlib-stage2/gcc/testsuite/ -iname '*.sum' | xargs cat | wc -l)" -lt 100 ]]; then false; else true; fi
/bin/sh: 1: [[: not found
***********************************************************************************************************************
my qemu version is:
 commit 18a398f6a39df4b08ff86ac0d38384193ca5f4cc (HEAD -> master, tag: v3.0.0-rc2, origin/master)

jim-wilson, 2018-08-28:
Fixed.

closed 2018-08-28 by jim-wilson

zidanemarks, 2018-08-30:
thank you very much