riscv-collab/riscv-gnu-toolchain #371: can't do make for no configure files at /riscv-gnu-toolchain/riscv-binutils/

zidanemarks, 2018-08-30:
when pull the master code to:
commit 2ec34088477ae87f422f01e7cf23c444ebbf38da (HEAD -> master, origin/master, origin/HEAD)
i configure my makefile as this commnad:
../riscv-gnu-toolchain/configure --prefix=$RISCV --disable-linux --with-arch=rv32imac
and i encounter the issue as below:

**********************************************************************************************************************************cd build-binutils-newlib && CC_FOR_TARGET=riscv32-unknown-elf-gcc /home/project/build_rv32imac/../riscv-gnu-toolchain/riscv-binutils/configure \
	--target=riscv32-unknown-elf \
	 \
	--prefix=/opt/riscv \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-libreadline
/bin/sh: 1: /home/project/build_rv32imac/../riscv-gnu-toolchain/riscv-binutils/configure: not found
Makefile:375: recipe for target 'stamps/build-binutils-newlib' failed
make: *** [stamps/build-binutils-newlib] Error 127
root@shawntan-ThinkPad-T450s:/home/project/build_rv32imac# ../riscv-gnu-toolchain/configure --prefix=$RISCV --disable-linux --with-arch=rv32imac
**********************************************************************************************************************************
i found there was no repository about  /riscv-binutils/ whether my local or remote github 
could you please help me solve this issue

jim-wilson, 2018-08-30:
In the riscv-gnu-toolchain dir, do "git submodule update --init --recursive" after the git clone to get the submodule sources.

zidanemarks, 2018-08-30:
when i do update:
the info is:
root@shawntan-ThinkPad-T450s:/home/project/riscv-gnu-toolchain# git submodule update --init --recursive
***********************************************************************************************************************8
Submodule 'riscv-binutils' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Submodule 'riscv-gdb' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-gdb'
fatal: destination path '/home/project/riscv-gnu-toolchain/riscv-binutils' already exists and is not an empty directory.
fatal: clone of 'https://github.com/riscv/riscv-binutils-gdb.git' into submodule path '/home/project/riscv-gnu-toolchain/riscv-binutils' failed
Failed to clone 'riscv-binutils'. Retry scheduled
Cloning into '/home/project/riscv-gnu-toolchain/riscv-gdb'...

zidanemarks, 2018-08-30:
i'm very sorry. it's my fault to search the repository by the dir's name
both the riscv-binutils and riscv-gdb are share the same URL of https://github.com/riscv/riscv-binutils-gdb.git 
thank you very much for your help

closed 2018-08-30 by zidanemarks