riscv-collab/riscv-gnu-toolchain #375: CentOS 7.5 Compile Error

dragonware-com, 2018-09-05:
make[3]: Entering directory `/home/wsun/work/github/riscv-gnu-toolchain/build-binutils-newlib/sim'
/home/wsun/riscv
make[4]: Entering directory `/home/wsun/riscv'
make[4]: *** No targets specified and no makefile found.  Stop.
make[4]: Leaving directory `/home/wsun/riscv'
make[3]: *** [all] Error 1
make[3]: Leaving directory `/home/wsun/work/github/riscv-gnu-toolchain/build-binutils-newlib/sim'
make[2]: *** [all-sim] Error 2
make[2]: Leaving directory `/home/wsun/work/github/riscv-gnu-toolchain/build-binutils-newlib'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/wsun/work/github/riscv-gnu-toolchain/build-binutils-newlib'
make: *** [stamps/build-binutils-newlib] Error 2

dragonware-com, 2018-09-05:
It turns out that there are much less Makefile generated on CentOS comparing to Ubuntu.

$ diff failed.make.lst makefile.lst 
0a1,463
> ./Makefile
> ./build-binutils-newlib/Makefile
> ./build-binutils-newlib/bfd/Makefile
> ./build-binutils-newlib/bfd/doc/Makefile
> ./build-binutils-newlib/bfd/po/Makefile
> ./build-binutils-newlib/binutils/Makefile
> ./build-binutils-newlib/binutils/doc/Makefile
> ./build-binutils-newlib/binutils/po/Makefile
> ./build-binutils-newlib/etc/Makefile
> ./build-binutils-newlib/gas/Makefile
> ./build-binutils-newlib/gas/doc/Makefile
> ./build-binutils-newlib/gas/po/Makefile
> ./build-binutils-newlib/gdb/Makefile
> ./build-binutils-newlib/gdb/build-gnulib/Makefile
> ./build-binutils-newlib/gdb/build-gnulib/import/Makefile
> ./build-binutils-newlib/gdb/data-directory/Makefile
> ./build-binutils-newlib/gdb/doc/Makefile
> ./build-binutils-newlib/gdb/testsuite/Makefile
> ./build-binutils-newlib/gprof/Makefile
> ./build-binutils-newlib/gprof/po/Makefile
> ./build-binutils-newlib/intl/Makefile
> ./build-binutils-newlib/ld/Makefile
> ./build-binutils-newlib/ld/po/Makefile
> ./build-binutils-newlib/libdecnumber/Makefile
> ./build-binutils-newlib/libiberty/Makefile
> ./build-binutils-newlib/libiberty/testsuite/Makefile
> ./build-binutils-newlib/opcodes/Makefile
> ./build-binutils-newlib/opcodes/po/Makefile
> ./build-binutils-newlib/readline/Makefile
> ./build-binutils-newlib/readline/doc/Makefile
> ./build-binutils-newlib/readline/examples/Makefile
> ./build-binutils-newlib/readline/shlib/Makefile
> ./build-binutils-newlib/sim/Makefile
> ./build-binutils-newlib/sim/riscv/Makefile
> ./build-binutils-newlib/sim/testsuite/Makefile
> ./build-binutils-newlib/zlib/Makefile
> ./build-gcc-newlib-stage1/Makefile
> ./build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/fixincludes/Makefile
> ./build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/libcpp/Makefile
> ./build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/libiberty/Makefile
> ./build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/libiberty/testsuite/Makefile
> ./build-gcc-newlib-stage1/fixincludes/Makefile
> ./build-gcc-newlib-stage1/gcc/Makefile
> ./build-gcc-newlib-stage1/gcc/ada/Makefile
> ./build-gcc-newlib-stage1/gcc/ada/gcc-interface/Makefile
> ./build-gcc-newlib-stage1/intl/Makefile
> ./build-gcc-newlib-stage1/libbacktrace/Makefile
> ./build-gcc-newlib-stage1/libcpp/Makefile
> ./build-gcc-newlib-stage1/libdecnumber/Makefile
> ./build-gcc-newlib-stage1/libiberty/Makefile
> ./build-gcc-newlib-stage1/libiberty/testsuite/Makefile
> ./build-gcc-newlib-stage1/lto-plugin/Makefile
> ./build-gcc-newlib-stage2/Makefile
> ./build-gcc-newlib-stage2/build-x86_64-pc-linux-gnu/fixincludes/Makefile
> ./build-gcc-newlib-stage2/build-x86_64-pc-linux-gnu/libcpp/Makefile
> ./build-gcc-newlib-stage2/build-x86_64-pc-linux-gnu/libiberty/Makefile
> ./build-gcc-newlib-stage2/build-x86_64-pc-linux-gnu/libiberty/testsuite/Makefile
> ./build-gcc-newlib-stage2/fixincludes/Makefile
> ./build-gcc-newlib-stage2/gcc/Makefile
> ./build-gcc-newlib-stage2/gcc/ada/Makefile
> ./build-gcc-newlib-stage2/gcc/ada/gcc-interface/Makefile
> ./build-gcc-newlib-stage2/intl/Makefile
> ./build-gcc-newlib-stage2/libbacktrace/Makefile
> ./build-gcc-newlib-stage2/libcc1/Makefile
> ./build-gcc-newlib-stage2/libcpp/Makefile
> ./build-gcc-newlib-stage2/libdecnumber/Makefile
[... 415 lines cut ...]
< ./build-binutils-newlib/gdb/testsuite/Makefile
< ./build-binutils-newlib/gprof/Makefile
< ./build-binutils-newlib/gprof/po/Makefile
< ./build-binutils-newlib/intl/Makefile
< ./build-binutils-newlib/ld/Makefile
< ./build-binutils-newlib/ld/po/Makefile
< ./build-binutils-newlib/libdecnumber/Makefile
< ./build-binutils-newlib/libiberty/Makefile
< ./build-binutils-newlib/libiberty/testsuite/Makefile
< ./build-binutils-newlib/opcodes/Makefile
< ./build-binutils-newlib/opcodes/po/Makefile
< ./build-binutils-newlib/readline/Makefile
< ./build-binutils-newlib/readline/doc/Makefile
< ./build-binutils-newlib/readline/examples/Makefile
< ./build-binutils-newlib/readline/shlib/Makefile
< ./build-binutils-newlib/sim/Makefile
< ./build-binutils-newlib/sim/riscv/Makefile
< ./build-binutils-newlib/sim/testsuite/Makefile
< ./build-binutils-newlib/zlib/Makefile
< ./riscv-binutils-gdb/gdb/features/Makefile
< ./riscv-binutils-gdb/zlib/contrib/blast/Makefile
< ./riscv-binutils-gdb/zlib/contrib/minizip/Makefile
< ./riscv-binutils-gdb/zlib/contrib/puff/Makefile
< ./riscv-binutils-gdb/zlib/contrib/untgz/Makefile
< ./riscv-binutils-gdb/zlib/nintendods/Makefile

jim-wilson, 2018-09-06:
The initial configure only creates the top level makefile.  The other makefiles are created when individual sub-packages are configured and built.  But since your build failed on the first sub-package, only a small number of Makefiles were created.  This does not appear relevant to the problem.

It looks like the problem might be that you have a build-binutils-newlib/sim/Makefile file.  Current riscv-gnu-toolchain configures binutils using --without-sim, so the sim directory should not have been created.
If you updated your riscv-gnu-toolchain sources recently, and already had a build tree, then you need to rm -rf the old build tree and create a new one from scratch because of changes to how binutils and gdb are built.

Unless maybe you have an old version of riscv-gnu-toolchain in which case the the sim directory would be created.

I need more info about how to reproduce the failure.  And you may need to do a clean build from scratch.

dragonware-com, 2018-09-06:
My current guess is that this is a CentOS/REL only issue, since I can build the riscv-toolchain on Ubuntu 16.04 successfully. My CentOS version is 7.5. To debug the problem, it should be done on a clean CentOS (virtual)machine, 7.5, everything up-to-date.

My current workaround is to build the tool-chain on Ubuntu 16.04, and copy the installation folder to the CentOS machine. It is not ideal, but it works.

jim-wilson, 2018-09-19:
I tried riscv-gnu-toolchain builds on Centos 7.5 last night and had no problem doing both newlib and linux builds.

kito-cheng, 2020-10-12:
Close this issue due to can't reproduce and long time no further discussion/response .

closed 2020-10-12 by kito-cheng