riscv-collab/riscv-gnu-toolchain #377: _GLOBAL_OFFSET_TABLE_  in .data sections

Pinpiew, 2018-09-13:
the result of compile, _GLOBAL_OFFSET_TABLE_  in .data sections

but at startup, the ins "la t0, trap_vector",  may be use the  _GLOBAL_OFFSET_TABLE_, but this time .data sections have not load to VMA.

then register t0 get value '0'

disassemble as following: (_ftext is the entry)
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         0000316a  0000000020000000  0000000020000000  00001000  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .rodata       0000085c  0000000020003170  0000000020003170  00004170  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .data         000000f1  0000000020020000  00000000200039cc  00005000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  3 .bss          000022c0  0000000020020100  0000000020003abd  00005100  2**4

0000000020000000 <_ftext>:
    20000000:   34001073            csrw    mscratch,zero
    20000004:   00020297            auipc   t0,0x20
    20000008:   0742b283            ld  t0,116(t0) # 20020078 <_GLOBAL_OFFSET_TABLE_+0x8>
    2000000c:   30529073            csrw    mtvec,t0
    20000010:   30502373            csrr    t1,mtvec
    20000014:   00629063            bne t0,t1,20000014 <_ftext+0x14>
    20000018:   00020117            auipc   sp,0x20
    2000001c:   06813103            ld  sp,104(sp) # 20020080 <_GLOBAL_OFFSET_TABLE_+0x10>
    20000020:   f14026f3            csrr    a3,mhartid
    20000024:   00168693            addi    a3,a3,1
    20000028:   00169693            slli    a3,a3,0x1
    2000002c:   00c69613            slli    a2,a3,0xc
    20000030:   00c10133            add sp,sp,a2
    20000034:   00020617            auipc   a2,0x20
    20000038:   06463603            ld  a2,100(a2) # 20020098 <_GLOBAL_OFFSET_TABLE_+0x28>
    2000003c:   00020697            auipc   a3,0x20
    20000040:   04c6b683            ld  a3,76(a3) # 20020088 <_GLOBAL_OFFSET_TABLE_+0x18>
    20000044:   00020717            auipc   a4,0x20
    20000048:   04c73703            ld  a4,76(a4) # 20020090 <_GLOBAL_OFFSET_TABLE_+0x20>

jim-wilson, 2018-09-13:
This doesn't look like a toolchain problem.  It looks like you are writing your own OS, and your OS code is buggy.  You need to load .data into memory before trying to read values from .data.  You might try looking at other OS source code trees to see how they handle OS startup.

Pinpiew, 2018-09-17:
it is a startup code, without OS.

if VMA == LMA， there is no problem, if not, problem may be happened.

if VMA != LMA, first must copy the .data section to VMA, this operation need linker srcipt symbal, but get the symbal address rely on the GLOBAL_OFFSET_TABLE, and then GLOBAL_OFFSET_TABLE is at the .data section. so get linker srcipt symbal address error.

why GLOBAL_OFFSET_TABLE data place at .data section?  if move to .rodata section, is it better?

jim-wilson, 2018-09-17:
So you are trying to load the address of the data section so you can copy it into memory, but this address was itself placed in the data section?  That isn't going to work.  You may need to write your code differently, or compile it differently.  Though I would expect the data address to be in rodata not data, and rodata should be next to text and hence should already be in memory here.  GLOBAL_OFFSET_TABLE is a pointer into the data section, we can't move that.  Or maybe you just need a better linker script to make this work.

You can find an example of similar code in sifive/freedom-e-sdk in the hifive1 support where binaries get loaded into rom, and then data is copied to flash at run-time to make it writable.

lazyparser, 2021-03-04:
@Pinpiew Hi, would you like to update the status of this issue? Is it safe to close now?

TommyMurphyTM1234, 2022-09-27:
> This doesn't look like a toolchain problem. It looks like you are writing your own OS, and your OS code is buggy. You need to load .data into memory before trying to read values from .data. You might try looking at other OS source code trees to see how they handle OS startup.

closed 2022-09-27 by TommyMurphyTM1234