riscv-collab/riscv-gnu-toolchain #380: Vector extension encoding conflicts

hawajkm, 2018-09-18:
I am working on implementing vector extension draft on a GNU toolchain. So far, I was able to successfully implement the complete draft; however, I had to improvise in couple of issues that I discover throughout the process.

The first issue was `op_code` encoding. The `op_code` chosen for most vector operation conflict with `JAL`. Although we can tell most instructions apart through bit 14, `VMADD`, `VMSUB`, `VNMADD`, and `VNMSUB` all have definite conflict with `JAL`. Also, `op_code` for all vector memory operations just mirror that of the float-point memory operations.

It seems that the opcode assignment were made as fillers? For now, I have used three of the reserved standard opcodes for vector-op, vector-load, and vector-store. I think that is not what the intent going forward; sacrificing three standard op-code extensions seems too much. But, for now, until the draft becomes more concrete, this would unblock my work.

The second issue is just a mere question: Would the immediate fields used in the draft be the same going forward? I am curious because the immediate field for load/store seems too small to encode anything useful and most likely the compiler will end-up filling it with zeros. Maybe it makes sense to make the immediate some multiple of a constant and/or it would be an upper-immediate?

aswaterman, 2018-09-18:
The encoding does not conflict, because MADD etc. require that m={1,2,3}.  JALR corresponds to the case m=0.  However, this encoding is temporary, and a revamped one will be released in the next month or so.  Sorry for the confusion.

aswaterman, 2018-09-18:
Closed, since this is a cross-post of https://github.com/riscv/riscv-isa-manual/issues/227

closed 2018-09-18 by aswaterman

hawajkm, 2018-09-18:
Sorry! I think the topic belongs to ISA manual, maybe? :)

Ohhh I see.