riscv-collab/riscv-gnu-toolchain #384: Error while building riscv32i toolchain in windows

nanakeys, 2018-10-14:
I am trying to build a pure rv32i tool-chain by following the instructions from [here](https://github.com/cliffordwolf/picorv32/blob/master/README.md). After the `make -j$(nproc)` command, I got multiple errors including a permission denied. So I checked the ownership of the file, and I am still the owner. I found another page [here](https://www.emdalo.com/posts/risc-v-gnu-compiler-toolchain-howto-compile-on-windows/) and inserted the line ` --src=../src/newlib-gcc \` as instructed, but still the same error.  I have attached my error file to this post as make.err.log. I have also attached the log file as make.log in case it would be needed. I am new to this so please go easy on me :D. I even deleted the repository created and started the whole process two more times to make sure I didn't miss anything. The errors are found on lines 1067-1070 of the make file. How do I solve these errors? I appreciate any efforts and thanks in advance

[make.err.log](https://github.com/riscv/riscv-gnu-toolchain/files/2476371/make.err.log)
[make.log](https://github.com/riscv/riscv-gnu-toolchain/files/2476372/make.log)

jim-wilson, 2018-10-14:
I see that you are trying to build a MS Windows toolchain.  The easiest way to do that is to actually do a canadian cross build from a linux machine.  Building on windows should work, but is rarely tested, and is more likely to fail from problems with the window tools like msys/cygwin than it is from problems with our tools.

You will have to try to debug this on your side.  Use "ls -l" to verify that the program is there and executable.  Use "file" to verify that it is a proper windows binary.  Use "ldd" to check that all of the shared libraries are available.  Try running the failing command by hand, maybe it was just a spurious non-repeatable failure, in which case just trying the build again might work.

Liviu builds windows binaries for his toolchains.  You could try downloading and using his tools.  You could also try using his build scripts if you want to build your own.
https://github.com/gnu-mcu-eclipse

ilg-ul, 2018-10-14:
> build a MS Windows toolchain. The easiest way to do that is to actually do a canadian cross build from a linux machine.

I confirm. 
using mingw-w64 on linux or mac is probably the easiest way to create windows binaries for the gnu tools.

nanakeys, 2018-10-15:
Hello Jim. Thanks for the reply. So what you are asking me to do is do a Canadian build. Can you point me to some sources you trust? 
Also, I did the "file" command and, yes, the file is executable. However, "ldd" threw a 'permission denied' error. Perhaps it requires higher privileges, so I tried to use sudo or su but both are not recognised commands. I even started MYSYS2 32-bit with administrator priviledges but to no avail.  The actual command is 
`/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib-stage1/./gcc/xgcc -B/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib-stage1/./gcc/ -xc -nostdinc nul -S -o nul -fself-test=../../../riscv-gcc/gcc/testsuite/selftests` 
One more thing, If I do the build in linux as you suggested, will I be able to test my designs in windows by means of simulation and packaging?

jim-wilson, 2018-10-15:
The riscv-gnu-toolchain sources are good.  If you want to use FSF sources, you can use the FSF binutils 2.30 or 2.31 releases, or the FSF GCC 7.3 or 8.1 releases.  Most any newlib should work.  There are many good choices.  The toolchain sources should not be a problem.  The problem is with the windows tools like cygwin and msys which are complicated to use.

I have no idea about simulation and packaging.  That doesn't sound like a riscv-gnu-toolchain question.  I'm not a hardware guy, and can't answer any hardware questions.

jim-wilson, 2018-10-15:
I rarely do builds on windows.  I would suggest doing a web search to look for info and/or try reading msys/cygwin FAQs.  The riscv-unknown-elf-gcc.exe binary will only appear in the install tree after the build completes, and the build did not complete, so this is expected.

nanakeys, 2018-10-15:
Ok. Thanks a lot. If I fixed it, I will update this thread. Thanks Jim and ilg-ul

topchenhui, 2019-03-21:
hello nanakeys.I also encountered similar problems with you on Windows, how do you solve the problem after you?

closed 2021-03-08 by lazyparser