riscv-collab/riscv-gnu-toolchain #385: build failure for rv32imc ("error ATOMIC_EXCHANGE_USES_CAS has to be defined")

gsomlo, 2018-10-15:
After configuring with
`./configure --prefix=$HOME/RiscV --with-arch=rv32imc --with-abi=ilp32`

and running `make linux`, I get the error below. Is that a bad arch/abi combo, or is this a bona-fide bug?

```
[...]
if test -r /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/csu/abi-tag.h.new; then mv -f /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/csu/abi-tag.h.new /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/csu/abi-tag.h; \
else echo >&2 'This configuration not matched in ../abi-tags'; exit 1; fi
/usr/bin/gawk -f ../scripts/gen-as-const.awk ../sysdeps/riscv/nptl/tcb-offsets.sym \
| riscv32-unknown-linux-gnu-gcc  -S -o /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.hT3 -std=gnu11 -fgnu89-inline  -O2 -Wall -Wundef -Wwrite-strings -fmerge-all-constants -fno-stack-protector -frounding-math -g -mcmodel=medlow -Wstrict-prototypes -Wold-style-definition     -ftls-model=initial-exec      -I../include -I/home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/csu  -I/home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32  -I../sysdeps/unix/sysv/linux/riscv/rv32/  -I../sysdeps/unix/sysv/linux/riscv  -I../sysdeps/riscv/nptl  -I../sysdeps/unix/sysv/linux/generic/wordsize-32  -I../sysdeps/unix/sysv/linux/generic  -I../sysdeps/unix/sysv/linux/riscv/rv32  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/riscv/rv32/  -I../sysdeps/riscv/rv32  -I../sysdeps/riscv  -I../sysdeps/init_array  -I../sysdeps/ieee754/ldbl-128  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754/flt-32  -I../sysdeps/wordsize-32  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem /home/somlo/RiscV/lib/gcc/riscv32-unknown-linux-gnu/8.2.0/include -isystem /home/somlo/RiscV/lib/gcc/riscv32-unknown-linux-gnu/8.2.0/include-fixed -isystem /home/somlo/riscv-gnu-toolchain/linux-headers/include  -D_LIBC_REENTRANT -include /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/libc-modules.h -DMODULE_NAME=libc -include ../include/libc-symbols.h       -DTOP_NAMESPACE=glibc -x c - \
	-MD -MP -MF /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h.dT -MT '/home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h.d /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h'
In file included from ../sysdeps/nptl/lowlevellock.h:22,
                 from ../nptl/descr.h:29,
                 from ../sysdeps/riscv/nptl/tls.h:41,
                 from ../sysdeps/unix/sysv/linux/riscv/sysdep.h:23,
                 from <stdin>:1:
../include/atomic.h:826:3: error: #error ATOMIC_EXCHANGE_USES_CAS has to be defined.
 # error ATOMIC_EXCHANGE_USES_CAS has to be defined.
   ^~~~~
make[3]: *** [../Makerules:281: /home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h] Error 1
make[3]: Leaving directory '/home/somlo/riscv-gnu-toolchain/riscv-glibc/csu'
make[2]: *** [Makefile:215: csu/subdir_lib] Error 2
make[2]: Leaving directory '/home/somlo/riscv-gnu-toolchain/riscv-glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/home/somlo/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32'
make: *** [Makefile:238: stamps/build-glibc-linux-rv32imc-ilp32] Error 2
```

kito-cheng, 2018-10-15:
Atomic extension required for Linux/glibc. So you need rv32imac at least.

ZvonimirBandic, 2019-04-24:
I am getting something quite similar - have you ever figured out what the problem is:

	-MD -MP -MF /home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h.dT -MT '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h.d /home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h'
In file included from ../sysdeps/nptl/lowlevellock.h:22,
                 from ../nptl/descr.h:29,
                 from ../sysdeps/riscv/nptl/tls.h:41,
                 from ../sysdeps/unix/sysv/linux/riscv/sysdep.h:23,
                 from <stdin>:1:
../include/atomic.h:826:3: error: #error ATOMIC_EXCHANGE_USES_CAS has to be defined.
 # error ATOMIC_EXCHANGE_USES_CAS has to be defined.
   ^~~~~
../Makerules:281: recipe for target '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h' failed
make[3]: *** [/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32/tcb-offsets.h] Error 1
make[3]: Leaving directory '/home/bandic/riscv-gnu-toolchain/riscv-glibc/csu'
Makefile:215: recipe for target 'csu/subdir_lib' failed
make[2]: *** [csu/subdir_lib] Error 2
make[2]: Leaving directory '/home/bandic/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imc-ilp32'
Makefile:244: recipe for target 'stamps/build-glibc-linux-rv32imc-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imc-ilp32] Error 2

jim-wilson, 2019-04-24:
The RISC-V glibc port requires the A extension as previously mentioned.  You can't build a linux toolchain for rv32imc because that doesn't have the A extension.  See the file sysdeps/unix/sysv/linux/riscv/atomic-machine.h.

ZvonimirBandic, 2019-04-25:
I also tried with --with-arch=rv32imac, but got the same problem. I also tried newlib - that worked.
I am not sure why glibc would not work?

	-MD -MP -MF /home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h.dT -MT '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h.d /home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h'
In file included from ../sysdeps/nptl/lowlevellock.h:22,
                 from ../nptl/descr.h:29,
                 from ../sysdeps/riscv/nptl/tls.h:41,
                 from ../sysdeps/unix/sysv/linux/riscv/sysdep.h:23,
                 from <stdin>:1:
../include/atomic.h:826:3: error: #error ATOMIC_EXCHANGE_USES_CAS has to be defined.
 # error ATOMIC_EXCHANGE_USES_CAS has to be defined.
   ^~~~~
../Makerules:281: recipe for target '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h' failed
make[3]: *** [/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h] Error 1
make[3]: Leaving directory '/home/bandic/riscv-gnu-toolchain/riscv-glibc/csu'
Makefile:215: recipe for target 'csu/subdir_lib' failed
make[2]: *** [csu/subdir_lib] Error 2
make[2]: Leaving directory '/home/bandic/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/bandic/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32'
Makefile:244: recipe for target 'stamps/build-glibc-linux-rv32imac-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imac-ilp32] Error 2
bandic@bandic-VirtualBox:~/riscv-gnu-toolchain$

jim-wilson, 2019-04-25:
The error indicates that the compiler does not have atomic instructions enabled.  More specifically, that it isn't defining __riscv_atomic.

rohan:2009$ touch tmp.c
rohan:2010$ riscv32-unknown-linux-gnu-gcc -dM -E tmp.c | grep riscv_atomic
#define __riscv_atomic 1
rohan:2011$ 

The riscv-gnu-toolchain rv32imac-ilp32 linux support is tested occasionally and is expected to work.  If it didn't work for you, then you probably made a mistake at some point.  But since I don't know exactly what you did, I can't say what you might have done wrong.  I can't give any meaningful help unless you explain all steps required from the beginning to reproduce the result you got.

ZvonimirBandic, 2019-04-25:
Thanks Jim - I am sure I have made a mistake along the way. In my defense, I did follow the steps on the GitHub pretty religiously for the prerequisites (i.e. literally copy/pasted the cmd lines). I will start hunting why compiler does not have A enabled...I am probably the only person on the planet for whom glibc for RISC-V did not work. I will ask my people as well - I wanted to make sure to fairly report here first. Super thanks - Zvonimir

closed 2019-05-16 by jim-wilson

wxiaoyan, 2020-12-04:
I am getting something quite similar - have you ever figured out what the problem is？

jim-wilson, 2020-12-04:
You can't build a linux toolchain without the A (atomic) extension.  Either build an embedded elf toolchain, or add the A extension to arch.

If you do have the A extension in arch, then the problem might be that you have another (partial?) toolchain in yout path, or another (partial?) toolchain in the --prefix install dir, and the build is failing because of the other toolchain.  You should make sure that there is no other riscv linux toolchain on your path when you start the build.  And that there is no riscv linux toolchain in the --prefix install dir when you start the build.