riscv-collab/riscv-gnu-toolchain #388: A problem with the compiler option -march=rv32i

hyf6661669, 2018-10-30:
I'm trying to compile a simple test program including multiplication, division and modulo operation, which is supposed to run on a very simple risc-v core that only supports rv32i. So I use the option ```-march=rv32i``` for compiling. Here is the code of ```hello.c```:
```
void main(void)
{
  int a, b, c, d, e, f;
  a = 5;
  b = 10;
  c = 10 +  a * 2 * b;
  b = c / a;
  a = c % b;
  
  d = c;
  e = b;
  f = a;
  while(1);
}
```

The commands for compiling and disassembling are:
```
riscv32-unknown-elf-gcc -Wall -O0 -march=rv32i hello.c -o hello
riscv32-unknown-elf-objdump -d hello > hello.dump
```
But the disassembly file still contains many compressed instructions, for example:
```
hello:     file format elf32-littleriscv

Disassembly of section .text:

00010074 <_start>:
   10074:	00002197          	auipc	gp,0x2
   10078:	0e418193          	addi	gp,gp,228 # 12158 <__global_pointer$>
   1007c:	80c18513          	addi	a0,gp,-2036 # 11964 <_edata>
   10080:	82818613          	addi	a2,gp,-2008 # 11980 <_end>
   10084:	8e09                	sub	a2,a2,a0
   10086:	4581                	li	a1,0
   10088:	2c5d                	jal	1033e <memset>
   1008a:	00000517          	auipc	a0,0x0
   1008e:	20e50513          	addi	a0,a0,526 # 10298 <__libc_fini_array>
   10092:	22c5                	jal	10272 <atexit>
   10094:	24a9                	jal	102de <__libc_init_array>
   10096:	4502                	lw	a0,0(sp)
   10098:	004c                	addi	a1,sp,4
   1009a:	4601                	li	a2,0
   1009c:	2865                	jal	10154 <main>
   1009e:	aaf9                	j	1027c <exit>
```

Can u tell me how to generate an output file that only contains instructions from rv32i? Thanks.

jim-wilson, 2018-10-30:
Your libraries were built with default compiler options.  You apparently configured for a target that supports compresssed instructions, so your library files contain compressed instructions.  To fix this, you either need to build multiple copies of the libraries, which you can do by adding --enable-multilib when configuring.  Or alternatively, configure and build a toolchain for rv32i so that you get rv32i libraries.

hyf6661669, 2018-10-31:
> Your libraries were built with default compiler options. You apparently configured for a target that supports compresssed instructions, so your library files contain compressed instructions. To fix [...]

Thanks for your reply, now I can generate a target file that only contains instructions in rv32i.

closed 2018-10-31 by hyf6661669