riscv-collab/riscv-gnu-toolchain #395: How to riscv32-unknown-elf-run --watch-cycles-ACTION [+]COUNT?

NoErrorYu, 2018-11-26:
How to riscv32-unknown-elf-run --watch-cycles-ACTION [+]COUNT?
$riscv32-unknown-elf-gcc -mabi=ilp32 -march=rv32g -time -pipe -Wall -g -w fo_test.c -o test
$riscv32-unknown-elf-run --watch-cycles-ACTION [+]COUNT test

riscv32-unknown-elf-run --watch-cycles-ACTION [+]COUNT test
riscv32-unknown-elf-run: unrecognized option '--watch-cycles-ACTION'
Use --help for a complete list of options.
makefile:31: recipe for target 'run' failed
make: *** [run] Error 1

I have trｉｅｄ mant options:
$riscv32-unknown-elf-run --watch-cycles-ACTION  test
$riscv32-unknown-elf-run --watch-cycles-ACTION COUNT test
$riscv32-unknown-elf-run --watch-cycles-ACTION [+]COUNT test
riscv32-unknown-elf-run --watch-cycles-COUNT test

all fail:
riscv32-unknown-elf-run: unrecognized option '--watch-cycles-COUNT'
And when I used --help,It just display as follow,and I still dont konw how to make it work.

Use --help for a complete list oUsage: riscv32-unknown-elf-run [options] program [program args]
Options:
  --watch-cycles-ACTION [+]COUNT
                              Watch the simulator, take ACTION in COUNT cycles
                              (`+' for every COUNT cycles), ACTION is int
                              breakpoint
  --watch-pc-ACTION [!]ADDRESS
                              Watch the PC, take ACTION when matches ADDRESS
                              (in range ADDRESS,ADDRESS), `!' negates test
  --watch-clock-ACTION [+]MILLISECONDS
                              Watch the clock, take ACTION after MILLISECONDS
                              (`+' for every MILLISECONDS)
  --watch-delete IDENT|all|pc|cycles|clock
                              Delete a watchpoint
  --watch-info                List scheduled watchpoints
  --memory-delete ADDRESS|all, --delete-memory ADDRESS
                              Delete memory at ADDRESS (all addresses)
f options.

why it does't work ?
how to make it work ?
What additional options ｛ＧＣＣ |RUN｝are needed?

ＯＳ：Linux work ４.15.0-39-generic #42~16.04.1-Ubuntu SMP Wed Oct 24 17:09:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
RV-GCC:Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/dl/RISC-V/riscv/libexec/gcc/riscv32-unknown-elf/7.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/dl/RISC-V/riscv-tools/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv32-unknown-elf --prefix=/home/dl/RISC-V/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/dl/RISC-V/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=../../riscv-gcc --enable-checking=yes --disable-multilib --with-abi=ilp32 --with-arch=rv32ima 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 7.2.0 (GCC) 
RV-RUN:riscv-gnu-toolchain

riscv32-unknown-elf-run --version
GNU simulator (SIM) 8.0.50.20170724-git

ps:I have been looking for a detailed riscv-gnu-toolchain manual(what is this and how to use)  but have not find it.

THANKS A LOT.

NoErrorYu, 2018-11-26:
> ACTION is a variable that can have the value "int" or "breakpoint". So there is no --watch-cycles-ACTION option, but there is a --watch-cycles-int option.
> 
> Many simulator options require target dependent support, and will not work if that target dependent support is missing. The list of options printed with --help is a generic list, and not all of them will work, because not all of them have been implemented yet. --watch-cycles-ACTION does seem to work, but the result doesn't look very useful as the simulator just exits on the signal.
> 
> I don't think there is a general doc on the riscv toolchain. It works the same as all other GNU toolchains though. Individual tools have some RISC-V specific documentation, e.g. gcc, gas, ld, etc.

THANKS!you are right!,I  run ""riscv32-unknown-elf-run --watch-cycles-breakpoint +COUNT test" and it passed.
display:program stopped with signal 2 (Interrupt).
THANKS AGAIN.

After listening to what you said, I found that this is not what I want. I want to test the efficiency of a program running on RV: the number of cycles, and a report that takes a long time.
I should re-edit my question:
How to use RV－toolchain to measure program operation efficiency？ which tools is needed?
SPIKE or   riscv32-unknown-elf-run ？　Do I need to change the software?　Do I need to add some option when gcc compile?

jim-wilson, 2018-11-26:
The gdb simulator isn't a cycle accurate simulator, it is just an ISA simulator.  It can count number of instructions, but it can't give you cycle counts.  Similarly qemu isn't cycle accurate.  I don't know about spike, but that one is more likely to be cycle accurate than the others.

Regardless of which simulator you use, you can read the cycle and/or instret CSR registers to get performance info.  The instret register should give the same result with every simulator.  The cycle register will vary depending on how accurate the simulator is, for gdb sim and qemu the cycle count is the same as the instret count.

NoErrorYu, 2018-11-27:
> The gdb simulator isn't a cycle accurate simulator, it is just an ISA simulator. It can count number of instructions, but it can't give you cycle counts. Similarly qemu isn't cycle accurate. I don't [...]
－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
Thank you for solving the problem that has been bothering me for a long time.

As a software tool for beginners，how can I find the manual for SPIKE OR Riscv{xlen}-unknown-elf-run? If there is no RV manual, which manual should I refer to?

E.g, -log-cache-miss, need to set L1 L2 cache, the default is not work.
Spike pk -s test , need add -S to display the number of cycles,How do Ｉ know that Ｉ need to add -s and　What else can I add options？ 
Where can I get this information?

jim-wilson, 2018-11-27:
I don't know if there is any RISC-V specific manual for this.  The gdb sim doesn't have any documentation that I know of.  You can use "run --help" to see the command line arguments it takes.  Inside gdb, you can use
```
target sim
load
run
```
to run a program on the gdb simulator.   This is mentioned in the gdb docs under the target command.  You can also set breakpoints and use ^C to interrupt as usual with gdb.

I don't know if spike has documentation.  I haven't used spike much.

The -s option you mentioned is a pk option.  I didn't know that pk had support for options, I don't think this is documented anywhere either.

spike and pk aren't part of the toolchain.  You could try asking somewhere more appropriate for help with them.

NoErrorYu, 2018-11-28:
I am clear, thank you again for your reply and help.

Maybe hope ,I can find or sort out some useful things in the process of learning,  and contribute to the community.

closed 2018-11-28 by NoErrorYu