riscv-collab/riscv-gnu-toolchain #398: Error building on macOS 10.14.2

thelonious, 2018-12-29:
I'm seeing the following errors when running `./configure --prefix=/opt/risc; make`:

```
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:83:7: error: use of undeclared identifier 'isl_id_free'
      isl_id_free (it->first);
      ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:262:3: error: use of undeclared identifier 'isl_id_free'
  isl_id_free (tmp_isl_id);
  ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:598:5: error: use of undeclared identifier 'isl_id_free'
    isl_id_free (res->first);
    ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:638:53: error: use of undeclared identifier 'isl_id_get_user'
      ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
                                                    ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:641:7: error: use of undeclared identifier 'isl_id_free'
      isl_id_free (id);
      ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:802:31: error: use of undeclared identifier 'isl_id_get_user'
  poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
                              ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:808:3: error: use of undeclared identifier 'isl_id_free'
  isl_id_free (name_id);
  ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:1367:19: error: use of undeclared identifier 'isl_space_dim'
  int dimension = isl_space_dim (schedule_space, isl_dim_out);
                  ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:1371:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (schedule_space);
  ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:1372:16: error: use of undeclared identifier 'isl_id_alloc'; did you
      mean 'isl_mat_alloc'?
  isl_id *id = isl_id_alloc (isl_ast_build_get_ctx (build), "", for_info);
               ^~~~~~~~~~~~
               isl_mat_alloc
/usr/local/include/isl/mat.h:28:21: note: 'isl_mat_alloc' declared here
__isl_give isl_mat *isl_mat_alloc(isl_ctx *ctx,
                    ^
../.././riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:1372:61: error: cannot initialize a parameter of type 'unsigned int'
      with an lvalue of type 'const char [1]'
  isl_id *id = isl_id_alloc (isl_ast_build_get_ctx (build), "", for_info);
                                                            ^~
/usr/local/include/isl/mat.h:29:11: note: passing argument to parameter 'n_row' here
        unsigned n_row, unsigned n_col);
                 ^
40 warnings and 11 errors generated.
make[2]: *** [graphite-isl-ast-to-gimple.o] Error 1
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
```

It looks like an issue with isl. I've confirmed that isl has been installed with brew:

```
$ brew info isl
isl: stable 0.20 (bottled), HEAD
Integer Set Library for the polyhedral model
http://isl.gforge.inria.fr/
/usr/local/Cellar/isl/0.20 (71 files, 3.9MB) *
  Poured from bottle on 2018-12-01 at 18:53:47
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/isl.rb
==> Dependencies
Required: gmp ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 52,616 (30 days), 167,512 (90 days), 566,611 (365 days)
install_on_request: 487 (30 days), 1,852 (90 days), 9,633 (365 days)
build_error: 0 (30 days)
```

Any suggestions on how to resolve this?

Cheers,
Kevin

jim-wilson, 2018-12-30:
This is
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=86724
and has a trivial fix.  The patch was too late for the gcc-8.2 release which is what we have in riscv-gnu-toolchain, but it is already fixed upstream on the gcc 8 branch and on gcc mainline.

thelonious, 2018-12-30:
That did the trick! Thanks so much.

closed 2018-12-30 by thelonious