riscv-collab/riscv-gnu-toolchain #403: openocd riscv expose_csrs causes companion gdb to assert fail

RobertHenry6bev, 2019-02-05:
I'm  trying to use `riscv expose_csrs` command to openocd. openocd seems happy (I think) but the companion gdb takes an assertion: riscv-gnu-toolchain/riscv-gdb/gdb/target-descriptions.c:823: internal-error: int tdesc_register_bitsize(const tdesc_feature*, const char*): Assertion `reg != NULL' failed.

Tim said 13 months ago in https://github.com/riscv/riscv-gnu-toolchain/issues/319 that the OP was probably the first to use that feature.

What is the state of the art/practice to make this feature play?  Do I need additional registration steps in openocd to attribute my new CSRs in order for gdb to be happy when it gets config information from openocd?

jim-wilson, 2019-02-06:
I don't know anything about openocd other than the fact that I've run it via riscv-tests to help test gdb support.  I don't know what expose_csrs might be doing.

The gdb xml support is rather new.  If you do something someone hasn't tried before it might fail.

I'd need a reproducible testcase in order to do anything.

RobertHenry6bev, 2019-02-06:
I'll try to cook up a test case.  FWIW I'm using a software stack talking to a hifive1 board, which has no fp registers, and uses debug spec 0.11. gdb is probing the target asking about register ft0, somehow thinking the cpu supports the 'F' extension.

jim-wilson, 2019-02-06:
I don't have a hifive1, so I never learned how any of that stuff works.  But now that we have xml register set support, I would expect openocd to provide a list of target registers to gdb, rather than gdb probing for registers.  I think that should happen if you use an appropriate config file for openocd, but I don't know how openocd works.  freedom-e-sdk has an openocd config file for the hifive1, but this is pre-xml support, so I don't know if that matters or helps.

There is also an openocd config file for the hifive1 in riscv-tests/debug, but I don't know if that one is meant to work with hardware.  It is used with spike (or qemu) I think when running debug tests in riscv-tests.  That one happens to use the expose_csrs command, for testing purposes.  I wonder if the problem is reproducible there.  I've run some of the generic tests, but haven't tried running the target specific tests.

timsifive, 2019-02-06:
riscv-tests/debug contains a test case for `expose_csrs`. You can run it as follows:
```
riscv-tools/riscv-tests/debug$ ./gdbserver.py targets/RISC-V/spike32.py Custom
```
You can use --gdb to point it at a specific gdb binary. I'm still using the one in riscv-tools, which I know is getting dated.

OpenOCD does give gdb an XML list of registers, and `expose_csrs` adds to the list of registers. This list should only include the FPR registers if the F or D extensions are implemented.

lazyparser, 2021-03-04:
Hi @RobertHenry6bev would you like to update the status of this issue?

TommyMurphyTM1234, 2022-09-29:
@timsifive - do you know offhand if this issue is moot/not an issue at this stage? Thanks.

timsifive, 2022-09-29:
I don't off-hand know anything about this.

cmuellner, 2022-09-29:
Closing, since this issue is stale for almost 4 years and Tim is not aware of any issues.

closed 2022-09-29 by cmuellner

TommyMurphyTM1234, 2022-09-29:
Thanks @timsifive and @cmuellner.