riscv-collab/riscv-gnu-toolchain #405: unrecognized option '--64'

skyyong, 2019-02-16:
Hi,
I have followed the install guide to clone the lasted database and set path as:

      export PATH=$PATH:/opt/riscv/bin

But when I install the software with command 

     ./configure --prefix=/opt/riscv
      make

I still got the error as these:

    TARGET_CPU_DEFAULT="" \
    HEADERS="auto-host.h ansidecl.h" DEFINES="" \
    /bin/sh ../.././riscv-gcc/gcc/mkconfig.sh bconfig.h
    g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -   fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
    /opt/riscv/riscv64-unknown-elf/bin/as: unrecognized option '--64'
    make[2]: *** [build/genmodes.o] Error 1
    make[2]: Leaving directory `/tools/RISCV_toolchain/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
    make[1]: *** [all-gcc] Error 2
    make[1]: Leaving directory `/tools/RISCV_toolchain/riscv-gnu-toolchain/build-gcc-newlib-stage1'
    make: *** [stamps/build-gcc-newlib-stage1] Error 2

I searched the web to find this may be caused the path setting -- But I think I have set the right path. Is there any thing I missed? thanks

jim-wilson, 2019-02-16:
It looks like you have confused your native x86_64 and cross riscv toolchains.  It looks like "g++" is an x86_64 compiler, which will pass --64 to the assembler, but the assembler is a riscv assembler that does not accept the --64 option, and will not be able to assemble x86_64 assembly code.

"which g++" will tell you what g++ you are running.  And "which as" will tell you what as you are running.  They should both be native x86_64 tools, or cross riscv tools.  You can also do "g++ -v" and "as -v" to check what kinds of tools they are.

You said that you put /opt/riscv/bin on your path.  But I think that you accidentally put /opt/riscv/riscv64-unknown-elf/bin on your path which is a mistake.  These are not tools that you normally want to use directly.  They are primarily intended to be used internally by the compiler.  What does "echo $PATH" say?

skyyong, 2019-02-17:
Hi,

thank you very much for your kind reply. Here is my snapshot from screen:

    which g++         ---->        /bin/g++

    which as          ---->        /bin/as

    echo $PATH        ---->      /:/bin:/usr/bin:/sbin:/bin:/usr/local/bin:/opt/riscv/bin

thanks.

jim-wilson, 2019-02-17:
That doesn't show me what the problem is unfortunately, so something else is wrong.  Try compiling a hello world program with g++.  Does that work?  If not, run with -v to see what g++ is doing.  Maybe there is an environment variable set wrong.

Or maybe the problem is only in the build tree.  The build tree sets up an as link for the cross compiler, but this should only be run by the just built compiler.  This can be run by accident if you have "." on your path, or an empty string in your path.  If your path starts with ":" then that is equivalent to having dot on your path.  You must not have dot on your path while building a cross toolchain as then the wrong binaries may be run at build time.  The info you gave shows your path starting with / which is odd, maybe that is a typo, and it actually started with dot or an empty string?  If you cd into your build tree, into the build-gcc-newlib-stage1 dir and run "which as" from there, and it gives the as in the current dir instead of /bin/as, then you have dot on your path and the build will fail.

skyyong, 2019-02-17:
thank your for your kind guidance. The problem is caused by the wrong path settings, as you pointed out, the start "/:" made the problem. After correct this problem, I see the make process for Newlib finishes. Now I'm starting for installing Linux multilib now. Thanks.

closed 2019-02-17 by jim-wilson

xinkai132, 2019-02-18:
Hi,I have met this problem too,I failed whatever I modify my PATH.
Could you show your bash file  or how to set the right path in detail?
Thank you

skyyong, 2019-02-18:
Hi,
try this:  
         PATH=/opt/riscv/bin:/bin:/usr/bin:/sbin:/bin:/usr/local/bin

xinkai132, 2019-02-19:
Thank you ,It worked