riscv-collab/riscv-gnu-toolchain #408: Static linking of Tool chain

cghalibkhan, 2019-02-18:
Hello everyone,
Actually am building Riscv gnu Toolchain for cross-native compiler, After building it, it's creating a dynamic linking, so i want to make it statically Linking instead of Dynamically linking, for statically linking i change  the option of --enable-shared to --disable-shared and Added --enable-static in the related Targets of linux-native: of makefile.in and configured it as the step's i followed below.

$cd /riscv-gnu-toolchain
$ ./configure  --prefix=$PATH
$ make linux-native 
Here am getting an Error .

WARNING: liblto_plugin.la is static, not copying to ../gcc/liblto_plugin.la
WARNING: liblto_plugin.la is static, not copying to ../gcc/liblto_plugin.la
configure: WARNING: decimal float is not supported for this target, ignored
configure: WARNING: libidn add-on ignored with --disable-shared
configure: WARNING: libidn add-on ignored with --disable-shared
cc1: fatal error: /home/exaleapserver/khan-ws/cross-native/riscv-gnu-toolchain/bulid/build-glibc-linux-rv64imafdc-lp64d/libc-modules.h: No such file or directory
compilation terminated.
make[3]: *** [/home/exaleapserver/khan-ws/cross-native/riscv-gnu-toolchain/bulid/build-glibc-linux-rv64imafdc-lp64d/tcb-offsets.h] Error 1
make[2]: *** [csu/subdir_lib] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2
make: *** Waiting for unfinished jobs....

cghalibkhan, 2019-02-18:
Here is the Log of build Target linux-native: of riscv-gnu-toolchian 
--> [linux-native.log](https://github.com/riscv/riscv-gnu-toolchain/files/2875132/linux-native.log)
Please given yours valuable suggestions on statically Linking.
Dynamically linked one is shown below:
$ file riscv64-unknown-linux-gnu-gcc
$ riscv64-unknown-linux-gnu-gcc: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 3.0.0, not stripped

jim-wilson, 2019-02-18:
On Mon, Feb 18, 2019 at 2:36 AM cghalibkhan <notifications@github.com>
wrote:

> Here is the Log of build Target linux-native: of riscv-gnu-toolchian
> --> linux-native.log
> <https://github.com/riscv/riscv-gnu-toolchain/files/2875132/linux-native.log>
> Please given yours valuable [...]

It is possible to use the riscv-gnu-toolchain linux-native rules to cross
build a native compiler, but it isn't recommended.  This was useful at the
beginning when trying to bootstrap riscv-linux software, but we are way
past that point.  There are many better ways to solve this problem now.
Any reasonable riscv linux system should already have a compiler, or have
support for building one.  You should use that instead of
riscv-gnu-toolchain.  If you have a target OS and it doesn't have a
compiler, then the best way to build a toolchain is to create a sysroot
from the target OS, and then use that to cross build binutils and gcc.  You
don't need to build glibc as part of this.

I think the problem here is that you are trying to use --disable-shared
with the glibc build.  This should not be necessary.  You need to disable
shared for the gcc build to ensure that you don't get binaries dynamically
linked against the wrong glibc version (which is why you shouldn't build a
compiler this way in the first place), but there is no need to disable
shared support in glibc itself.  It may be the case that --disable-shared
no longer works for glibc.  Glibc dropped full support for static linking
about 20 years ago, and has required shared linking ever since, so you
shouldn't be trying to disable it.

You can probably get a better answer if you define what problem you are
trying to solve, rather than ask for help on how to do something the wrong
way.

Jim

closed 2019-05-16 by jim-wilson