riscv-collab/riscv-gnu-toolchain #409: How to generate RV32i  assemly code of c code

Mohsannaeem, 2019-02-19:
I had installed riscv-gnu-toolchain.I had also compile c code using 
1)     riscv64-unknown-elf-gcc hello.c
and also get assembly code and data and text section of this code using 
2)     riscv64-unknown-elf-objdump -Ds a.out > hello.s
 but the problem is this this is for 64 bit  but i want assembly and text and code section for  32I  bit
How Can i compile and get assembly  code and text and data section for my c code......??????? 
My 64 hello.s file 
a.out:     file format elf64-littleriscv

Contents of section .text:
 100b0 97210000 938181be 138501f6 138681f9  .!..............
 100c0 098e8145 ef00e01c 17050000 13058512  ...E............
 100d0 ef00600f ef004015 02452c00 0146ef00  ..`...@..E,..F..
 100e0 400b6f00 000f8280 138781f4 938781f4  @.o.............
 100f0 638be700 37030000 13030300 63050300  c...7.......c...
 10100 138581f4 02838280 938581f4 938781f4  ................
 10110 9d8d8d85 8947b3c5 f50291c9 37030000  .....G......7...
 10120 13030300 63050300 138581f4 02838280  ....c...........
  ........
Contents of section .eh_frame:
 11480 00000000                             ....            
Contents of section .init_array:
 11488 6a010100 00000000                    j.......        
Contents of section .fini_array:
 11490 30010100 00000000                    0.......        
Contents of section .data:
 11498 00000000 00000000 d0190100 00000000  ................
 114a8 801a0100 00000000 301b0100 00000000  ........0.......
 114b8 00000000 00000000 00000000 00000000  ................
 114c8 00000000 00000000 00000000 00000000  ................
 114d8 00000000 00000000 00000000 00000000  ................
 114e8 00000000 00000000 00000000 00000000  ................
 114f8 00000000 00000000 00000000 00000000  ................
 11508 00000000 00000000 00000000 00000000  ................
  ......... 
Contents of section .sdata:
 11be0 98140100 00000000 00000000 00000000  ................
 11bf0 98140100 00000000                    ........        
Contents of section .comment:
 0000 4743433a 2028474e 55292038 2e322e30  GCC: (GNU) 8.2.0
 0010 00                                   .               

Disassembly of section .text:

00000000000100b0 <_start>:
   100b0:	00002197          	auipc	gp,0x2
   100b4:	be818193          	addi	gp,gp,-1048 # 11c98 <__global_pointer$>
   100b8:	f6018513          	addi	a0,gp,-160 # 11bf8 <_edata>
   100bc:	f9818613          	addi	a2,gp,-104 # 11c30 <__BSS_END__>
   100c0:	8e09                	sub	a2,a2,a0
   100c2:	4581                	li	a1,0
   100c4:	1ce000ef          	jal	ra,10292 <memset>
   100c8:	00000517          	auipc	a0,0x0
   100cc:	12850513          	addi	a0,a0,296 # 101f0 <__libc_fini_array>
   100d0:	0f6000ef          	jal	ra,101c6 <atexit>
   100d4:	154000ef          	jal	ra,10228 <__libc_init_array>
   100d8:	4502                	lw	a0,0(sp)
   100da:	002c                	addi	a1,sp,8
   100dc:	4601                	li	a2,0
   100de:	0b4000ef          	jal	ra,10192 <main>
   100e2:	0f00006f          	j	101d2 <exit>

00000000000100e6 <_fini>:
   100e6:	8082                	ret
....................................................
THANKS

jim-wilson, 2019-02-19:
Did you build your own riscv-gnu-toolchain, or install one that you found on the net?  If you found one on the net, which one was it?

You can get rv32i code by using -march=rv32i -mabi=ilp32.  But if you want to compile programs, then you also need libraries for rv32i/ilp32.  If you have a toolchain from elsewhere, and it is multilib, then this should just work.  Try adding --print-multi-lib to see the available multilibs.

If you built it yourself, you can get a multilib toolchain by adding the --enable-multilib configure option.  Or alternatively, you can use --with-arch=rv32i --with-abi=ilp32 to get a toolchain that defaults to rv32i/ilp32 code.

Mohsannaeem, 2019-02-20:
I had build my own toolchain by following these commands
   https://github.com/riscv/riscv-gnu-toolchain
I am just trying your option  by running these commands
 ./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32 
 make linux
For multi lib installation:
  ./configure --prefix=/opt/riscv --enable-multilib
   make linux
But i have another problem:
    by this commands  I get different section of assembly code like  .text and .comment and many other .....But i want only hex code of instuction for my processor  Like hex code for these command (ADD, AUPIC,JAL,JALR) with their operands  that i directly  load in my Program Memory( FPGA Block RAM)    
Thanks for your reply 

    riscv64-unknown-elf-gcc hello.c
     riscv64-unknown-elf-objdump -Ds a.out > hello.s

Mohsannaeem, 2019-02-20:
I had build my own toolchain by following these commands
https://github.com/riscv/riscv-gnu-toolchain
I am just trying your option by running these commands
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
make linux 
 and getting this error
compilation terminated.
make[3]: *** [_negdi2.o] Error 1
make[3]: Leaving directory `/home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv32-unknown-linux-gnu/libgcc'
make[2]: *** [all-target-libgcc] Error 2
make[2]: Leaving directory `/home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2'
make: *** [stamps/build-gcc-linux-stage2] Error 2

After this i compile c code but its failed
AND getting these error
mohsan@mohsan-EliteBook:~/test$ riscv64-unknown-elf-gcc -o hello hello.c -march=rv32i -mabi=ilp32
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/crt0.o
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/crtbegin.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/crtbegin.o
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__atexit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__atexit.o)
[... cut ...]
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-errno.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/crtend.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/crtend.o

Mohsannaeem, 2019-02-20:
I had build my own toolchain by following these commands
https://github.com/riscv/riscv-gnu-toolchain
I am just trying your option by running these commands
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
make linux
and getting this error
compilation terminated.
make[3]: *** [_negdi2.o] Error 1
make[3]: Leaving directory /home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv32-unknown-linux-gnu/libgcc' make[2]: *** [all-target-libgcc] Error 2 make[2]: Leaving directory /home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/mohsan/riscv-gnu-toolchain/build-gcc-linux-stage2'
make: *** [stamps/build-gcc-linux-stage2] Error 2

After this i compile c code but its failed
AND getting these error
mohsan@mohsan-EliteBook:~$ cd test
mohsan@mohsan-EliteBook:~/test$ riscv64-unknown-elf-gcc -o hello hello.c -march=rv32i -mabi=ilp32
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/crt0.o
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/crtbegin.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/crtbegin.o
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__atexit.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
[... cut ...]
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-errno.o)
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /lib/gcc/riscv64-unknown-elf/8.2.0/crtend.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /lib/gcc/riscv64-unknown-elf/8.2.0/crtend.o

jim-wilson, 2019-02-20:
There are multiple questions here, this is a little confusing.

1) Linux requires the A extension, so you can't build a working rv32i linux system.

2) Linux and multilib gets complicated very quickly because of shared libraries and shared library search paths.  Basically, you should not try this unless you are an expert.  I don't understand why you are even trying, because you were asking questions about a riscv64-unknown-elf-gcc, not a riscv64-unknown-linux-gnu-gcc.

3) It is better to use --enable-linux at configure time if you want a linux compiler, or --disable-linux (the default) if you want an embedded elf toolchain.  i don't recommend using the "make linux" and "make newlib" makefile targets, as they can lead to problems.

4) You didn't show the actual toolchain build failure above, you just showed the tail end of the log file which does not have any useful info.

5)  You then showed a failure running a embedded elf compiler after building a linux compiler which doesn't make any sense.  If your elf compiler isn't multilib then adding -march/-mabi options will give the linker failure that you saw.

closed 2019-05-16 by jim-wilson

Virtual-Machine, 2021-06-27:
Sorry for the necro, but FWIW, I am adding this here since I was struggling with a similar error recently despite following the instructions for 32bit support. What solved it for me was to pass the flag -m elf32lriscv to riscv64-unknown-elf-ld. Hope this might help others with the same issue.