riscv-collab/riscv-gnu-toolchain #410: Compile Error

flylcl, 2019-02-20:
I got this error message after run compile in eclipse.
Hope someone help me.
Thanks.

[100%] Linking C executable ../../../core.bin
cd /home/flylcl/Project/EN675/Validation/en675_fpag_rc001_190219101223/Debug/Source/core/shared && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/core.bin.dir/link.txt --verbose=1
/opt/riscv1/bin/riscv64-unknown-elf-gcc -DDEBUG -DC_FLAGS -O0 -MMD -MP -Wall -D__NO_INLINE__ -mcmodel=medany -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -g  -static -Wl,--gc-sections -T /home/flylcl/Project/EN675/Validation/en675_fpag_rc001_190219101223/Source/core/shared/exbl.lds -DDEBUG -DLINKER_FLAGS -Wl,--build-id=none -nostartfiles -static CMakeFiles/core.bin.dir/clib.c.obj CMakeFiles/core.bin.dir/syscalls.c.obj CMakeFiles/core.bin.dir/boot.c.obj CMakeFiles/core.bin.dir/exbl.c.obj CMakeFiles/core.bin.dir/msg.c.obj CMakeFiles/core.bin.dir/mentry.S.obj  -o ../../../core.bin  -L/home/flylcl/Project/EN675/Validation/en675_fpag_rc001_190219101223/Debug/lib ../../../lib/libfreertos.a ../../../lib/libcore0.a ../../../lib/libcore1.a ../../../lib/libcore2.a ../../../lib/libcore3.a ../../../lib/libdev.a 
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strtol.o): in function `.L6':
Source/core/shared/CMakeFiles/core.bin.dir/build.make:156: recipe for target 'core.bin' failed
gmake[2]: Leaving directory '/home/flylcl/Project/EN675/Validation/en675_fpag_rc001_190219101223/Debug'
strtol.c:(.text+0x14e): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
CMakeFiles/Makefile2:653: recipe for target 'Source/core/shared/CMakeFiles/core.bin.dir/all' failed
gmake[1]: Leaving directory '/home/flylcl/Project/EN675/Validation/en675_fpag_rc001_190219101223/Debug'
Makefile:105: recipe for target 'all' failed
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-locale.o): in function `_setlocale_r':
locale.c:(.text+0x8): relocation truncated to fit: R_RISCV_HI20 against `.LC0'
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mbtowc_r.o): in function `_mbtowc_r':
mbtowc_r.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-wctomb_r.o): in function `_wctomb_r':
wctomb_r.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
collect2: error: ld returned 1 exit status
gmake[2]: *** [core.bin] Error 1
gmake[1]: *** [Source/core/shared/CMakeFiles/core.bin.dir/all] Error 2
gmake: *** [all] Error 2
"/usr/bin/gmake -j4 all" terminated with exit code 2. Build might be incomplete.

aswaterman, 2019-02-20:
Looks like you have a C library that was compiled with `-mcmodel=medlow` and an application compiled with `-mcmodel=medany`.  If you recompile the C libraty to match the application, these link errors will disappear.

jim-wilson, 2019-02-20:
If you built your own compiler, you can use --with-cmodel=medany to get one that defaults to medany.

flylcl, 2019-02-20:
aswaterman, Thanks your reply,
I found that my C library was compiled with medlow. I'm trying to compile C library with medany.
Btw, I got more error message after change the riscv compile option to medlow same as C libaray. 
Could you explain the reason?

jim-wilson, 2019-02-20:
You didn't report the new errors with medlow, so we can't say anything about them.

flylcl, 2019-02-20:
> You didn't report the new errors with medlow, so we can't say anything about them.

It works fine after build the toolchain with medany option.
Thanks a lot.

closed 2019-02-20 by jim-wilson