riscv-collab/riscv-gnu-toolchain #413: make some changes to use my spec file instead of default spec.

cghalibkhan, 2019-02-22:
Actually, am Building a Riscv-Gnu-Toolchain with my spec file to use it will Building Riscv-Binutils and Riscv-gcc with Target of make all-gcc and make all & installing it, when my Executables Riscv64-unknown-Linux-gnu-gcc  is used to compile a Basic code (Hello World) it getting so many Errors, can anyone please 
 help it out, How Riscv-Gnu-toolchain makes to use my default spec file.

jim-wilson, 2019-02-22:
You didn't mention what your spec file does, or how you are using it, or what errors you are getting.  Generally, we need enough info to reproduce a problem in order to suggest solutions.

The reference to make all-gcc is odd.  This is a makefile rule inside the gcc source tree.  Generally, when using riscv-gnu-toolchain, you should only be running make at the top level, you shouldn't be running make in subdirs.  You can make this work if you know what you are doing, but it isn't clear exactly what you did, so it isn't clear if you did this right.

cghalibkhan, 2019-02-25:
[diff-spec.txt](https://github.com/riscv/riscv-gnu-toolchain/files/2900333/diff-spec.txt)
I'm a newbie. Trying to analyse gcc driver's handling of the specs file.
There are two specs created
1. make
riscv64-unknown-elf-gcc --dumpspecs > default-elf.specs

2.make linux
riscv64-unknown-linux-gnu-gcc --dumpspecs > default-linux-gnu.specs

When I do a diff of default-elf.specs with default-linux-gnu.specs, this is the partial result:
diff ~/default-elf.specs ~/default-linux-gnu.specs 
20c20
< 
---
> %{pthread:-D_REENTRANT}
35c35
< 
---
> %{profile:-p}
44c44
< %G %L %G
---
> %{static|static-pie:--start-group} %G %L    %{static|static-pie:--end-group}%{!static:%{!static-pie:%G}}
50c50
< crtend%O%s
---
> %{fvtable-verify=none:%s;      fvtable-verify=preinit:vtv_end_preinit.o%s;      fvtable-verify=std:vtv_end.o%s}    %{static:crtend.o%s;      shared|static-pie|pie:crtendS.o%s;      :crtend.o%s}    crtn.o%s    
53c53
< -melf%{march=rv32*:32}%{march=rv64*:64}lriscv %{mno-relax:--no-relax} %{shared}
---

..

I would like to know how the specs are created differently and on what parameters.
For example:
%{pthread:-D_REENTRANT} is present in spec of with riscv64-unknown-linux-gnu-gcc while not with riscv64-unknown-elf-gcc. Trying to figure out this. The diff is attached.

Thanks

cghalibkhan, 2019-02-25:
I'm a newbie. Trying to analyse gcc driver's handling of the specs file.
There are two specs created
1. make
riscv64-unknown-elf-gcc --dumpspecs > default-elf.specs

2.make linux
riscv64-unknown-linux-gnu-gcc --dumpspecs > default-linux-gnu.specs

When I do a diff of default-elf.specs with default-linux-gnu.specs, there are differences (attached):

I would like to know how the specs are created differently and on what parameters.
For example:
%{pthread:-D_REENTRANT} is present in spec of with riscv64-unknown-linux-gnu-gcc while not with riscv64-unknown-elf-gcc. Trying to figure out this. 

Thanks

cghalibkhan, 2019-02-25:
I'm a newbie. Trying to analyse gcc driver's handling of the specs file.
There are two specs created

1. make
   riscv64-unknown-elf-gcc --dumpspecs > default-elf.specs

2.make linux
riscv64-unknown-linux-gnu-gcc --dumpspecs > default-linux-gnu.specs

When I do a diff of default-elf.specs with default-linux-gnu.specs, there are differences (attached):

I would like to know how the specs are created differently and on what parameters.
For example:
%{pthread:-D_REENTRANT} is present in spec of with riscv64-unknown-linux-gnu-gcc while not with riscv64-unknown-elf-gcc. Trying to figure out this.

Thanks
[diff-spec.txt](https://github.com/riscv/riscv-gnu-toolchain/files/2900355/diff-spec.txt)

jim-wilson, 2019-02-25:
Specs are target dependent.  If you look in gcc/config/riscv/linux.h you will see that it defines it own LIB_SPEC, CPP_SPEC, and LINK_SPEC.  This is what ends up in the linux specs file.

See also the elf.h file in the same dir for stuff that ends up in the embedded elf specs files.

Most of the rest is standard gcc boilerplate.

There is also a freebsd.h file that defiles specs for FrreBSD.

closed 2019-05-16 by jim-wilson