riscv-collab/riscv-gnu-toolchain #415: Generating executable for Rocket Chip

vincenzociotoli, 2019-03-12:
Hello everyone,

I’m trying to generate an executable helloworld.run to do the emulation with Rocket Chip.
According to page (in section 2 of Debugging with GDB)

https://github.com/freechipsproject/rocket-chip

I have to generate the .run file with these 3 commands:

% riscv-gcc -c common/crt.S -o crt.o
% riscv-gcc -c hello.c -o hello.o
% riscv-gcc -T common/test.ld common/syscalls.c -lc hello.o -o hello.run

The last command doesn’t work and I receive these errors:

/tmp/ccJNVZKj.o: In function _init': syscalls.c:(.text+0x63c): multiple definition of_init’

/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/…/…/…/…/riscv64-unknown-elf/lib/crt0.o:(.text+0x52): first defined here

/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/…/…/…/…/riscv64-unknown-elf/lib/crt0.o: In function .L0 ': (.text+0x8): undefined reference to_edata’

/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/crtbegin.o: In function deregister_tm_clones': crtstuff.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 againstTMC_LIST’

/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/…/…/…/…/riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status

In addition I have tried to add the “-mcmodel=medany” option but is the same.

Where is the problem??

Thank you all!

ilia-diachkov, 2019-03-12:
Hello,

I don't know which crt.S, syscalls.c and test.ld do you use, so cannot reproduce. But it's definitely that the first error is due to default crt0.o passed by gcc to linker. crt0.o has the same symbol as in your syscalls.c. The second one is connected to undefined to_edata from crt0.o. It should be probably defined in the default linker script but you use custom script without the definition. Perhaps the third error also relates to default crtbegin.o and your script.

The solution would be to avoid linking with default crt*.o files. You can pass -nostdlib to gcc on the last step. However you'll need to pass all required libs (e.g. libc, libgcc) by hands.

Also I have noticed that you build crt.o but does not use it in the final linking.

jim-wilson, 2019-03-14:
You are copying files out of riscv-tests and trying to use them to compile.  These files require using the compiler in a non-standard way.  This really should be reported as a bug against the rocket-chip repo, as this isn't a compiler problem. Or you can look at riscv-tests to see how it uses the file.  If you build riscv-tests and then look at the build log, you can see how the files are meant to be used.

Anyways, this should work if you add --nostartfiles as previously mentioned.  It works for me.

gamma05:2036$ build-install/bin/riscv32-unknown-elf-gcc -O crt.S hello.c syscalls.c -Ttest.ld -nostartfiles -mcmodel=medany
gamma05:2037$ file a.out
a.out: ELF 32-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, not stripped
gamma05:2038$ 

I used an rv32 compiler as I had one handy, but it will work the same way with a rv64 compiler.

vincenzociotoli, 2019-03-15:
@ilia-diachkov Thank you for your answer!
I don't know how to pass by hand libraries in the last step. Can you show me,please?
I tried this command : 

riscv64-unknown-elf-gcc -T  /riscv-tools/riscv-tests/benchmarks/common/test.ld  /riscv-tools/riscv-tests/benchmarks/common/syscalls.c -nostartfiles -mcmodel=medany -lc helloworld.o -ohelloworld.run

But i receive these errors: 

helloworld.o: In function `strcpy':
(.text+0x4cd8): multiple definition of `strcpy'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0x1184): first defined here
helloworld.o: In function `printf':
(.text+0x2a8): multiple definition of `printf'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0xdb4): first defined here
helloworld.o: In function `memcpy':
(.text+0x726c): multiple definition of `memcpy'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0xf04): first defined here
helloworld.o: In function `_init':
(.text+0x36): multiple definition of `_init'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0x63c): first defined here
helloworld.o: In function `memset':
(.text+0x1dc): multiple definition of `memset'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0xfbc): first defined here
helloworld.o: In function `strcmp':
(.text+0x4bee): multiple definition of `strcmp'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0x1114): first defined here
helloworld.o: In function `sprintf':
(.text+0x4ae2): multiple definition of `sprintf'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0xe4c): first defined here
helloworld.o: In function `exit':
(.text+0x114): multiple definition of `exit'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0x4fa): first defined here
helloworld.o: In function `strlen':
(.text+0x4cea): multiple definition of `strlen'
/tmp/ccn5Jms2.o:syscalls.c:(.text+0x108a): first defined here
/tmp/ccn5Jms2.o: In function `syscall':
syscalls.c:(.text+0x348): undefined reference to `tohost'
syscalls.c:(.text+0x354): undefined reference to `fromhost'
syscalls.c:(.text+0x360): undefined reference to `fromhost'
/tmp/ccn5Jms2.o: In function `.L0 ':
syscalls.c:(.text+0x486): undefined reference to `tohost'
/tmp/ccn5Jms2.o: In function `printstr':
syscalls.c:(.text+0x562): undefined reference to `_tdata_end'
syscalls.c:(.text+0x566): undefined reference to `_tdata_end'
/tmp/ccn5Jms2.o: In function `thread_entry':
syscalls.c:(.text+0x56a): undefined reference to `_tdata_end'
syscalls.c:(.text+0x56e): undefined reference to `_tdata_begin'
syscalls.c:(.text+0x572): undefined reference to `_tdata_begin'
syscalls.c:(.text+0x576): undefined reference to `_tdata_begin'
/tmp/ccn5Jms2.o: In function `main':
syscalls.c:(.text+0x592): undefined reference to `_tbss_end'
syscalls.c:(.text+0x596): undefined reference to `_tbss_end'
syscalls.c:(.text+0x59a): undefined reference to `_tbss_end'
syscalls.c:(.text+0x59e): undefined reference to `_tdata_end'
syscalls.c:(.text+0x5a2): undefined reference to `_tdata_end'
syscalls.c:(.text+0x5a6): undefined reference to `_tdata_end'
collect2: error: ld returned 1 exit status

vincenzociotoli, 2019-03-15:
@jim-wilson  Hi Jim, Thank you for your answer!

I tried your command with a simple "helloworld" program.
File a.out it's generated properly and I receive these informations:

"a.out: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, not stripped"

My command is :

riscv64-unknown-elf-gcc -O  /riscv-tools/riscv-tests/benchmarks/common/crt.S helloworld.c  /riscv-tools/riscv-tests/benchmarks/common/syscalls.c -T  /riscv-tools/riscv-tests/benchmarks/common/test.ld -nostartfiles -mcmodel=medany

But when I try to generate the hex file with the command :  

elf2hex 32 16384  a.out  > helloworld.hex

I have :  

"elf2hex: ../fesvr/htif_hexwriter.cc:17: virtual void htif_hexwriter_t::read_chunk(addr_t, size_t, void*): Assertion `taddr < width*depth' failed."

So, I increase the width and the depth but doesn't work until I reach the order of milions.
It's so strange for a simple helloworld. 

Can you help me?

ilia-diachkov, 2019-03-15:
@vincenzociotoli as I see, you don't need the libraries for this example since you have all required functions in syscalls.c. So just add crt.S and remove -lc. You already have it in your last comment.

jim-wilson, 2019-03-15:
I don't know anything about the fesvr elf2hex.  All I can point out is that riscv-tools is poorly maintained, and that you probably shouldn't be using anything in there.​ ​  Also, as I mentioned before, your problems aren't toolchain problems, they are rocket-chip problems, you really should be asking the rocket-chip folks for help.

Just doing a web search on the error message, I found this which looks relevant, mentioning that it fails because the load address changed to 0x80000000, and that rocket-chip no longer uses elf2hex, and hence it isn't maintained anymore.
​https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/NPYFXN4XNYk

​if you really need an elf2hex program, any elf2hex program, then SiFive has one at
    ​https://github.com/sifive/elf2hex
but I don't know anything about that one either, as it also isn't part of the toolchain.

vincenzociotoli, 2019-03-18:
@ilia-diachkov Thank you! It works with this command.

vincenzociotoli, 2019-03-18:
@jim-wilson Thanks! I think that you're right. So, to see the assembly code of my program I can type this command:

riscv64-unknown-elf-objdump -d helloworld

closed 2019-05-16 by jim-wilson