riscv-collab/riscv-gnu-toolchain #416: riscv tool chain under Windows platform

topchenhui, 2019-03-14:
How to compile riscv tool chain under Windows platform

ilia-diachkov, 2019-03-14:
You can build it in cygwin environment.

ilg-ul, 2019-03-14:
if you are using an up-to-date windows 10, you can install WSL and on top of it one (or multiple) GNU/Linux distribution(s), like Ubuntu 18 LTS; then using mingw-w64 you can cross-compile any toolchain to exe files.

WSL is not a virtual machine, it is a windows subsystem that behaves much like a linux kernel, on top of it can be installed real GNU/Linux distributions.

on older Windows systems, you can use MSYS2 + mingw-w64.

ajf58, 2021-12-29:
@lazyparser this issue is labelled as help wanted, but has been inactive for some time. Is there more help required? Can this issue be closed?

TommyMurphyTM1234, 2022-09-27:
There are several options for compiling or cross compiling the tools (or any GNU style software) for Windows.
Closing this as it's not really an issue with the RISC-V GCC tools per se.

closed 2022-09-27 by TommyMurphyTM1234