riscv-collab/riscv-gnu-toolchain #418: riscv gnu tool chain Compile error

abhijeet085, 2019-03-21:
16:55:11 **** Incremental Build of configuration Debug for project Hello World ****
make all 
Building target: Hello World.elf
Invoking: GNU RISC-V Cross C Linker
riscv-none-embed-gcc -march=rv64imafc -mabi=lp64f -mcmodel=medlow -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -Xlinker --gc-sections -Wl,-Map,"Hello World.map" -o "Hello World.elf"  ./src/main.o   
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-exit.o): In function `.L0 ':
exit.c:(.text.exit+0x20): undefined reference to `_exit'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-sbrkr.o): In function `.L0 ':
sbrkr.c:(.text._sbrk_r+0xe): undefined reference to `_sbrk'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-writer.o): In function `.L0 ':
writer.c:(.text._write_r+0x14): undefined reference to `_write'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-closer.o): In function `.L0 ':
closer.c:(.text._close_r+0xe): undefined reference to `_close'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-lseekr.o): In function `.L0 ':
lseekr.c:(.text._lseek_r+0x14): undefined reference to `_lseek'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-readr.o): In function `.L0 ':
readr.c:(.text._read_r+0x14): undefined reference to `_read'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-fstatr.o): In function `.L0 ':
fstatr.c:(.text._fstat_r+0x12): undefined reference to `_fstat'
c:/users/aharshadbhai/appdata/roaming/xpacks/@gnu-mcu-eclipse/riscv-none-gcc/8.1.0-2.1/.content/bin/../lib/gcc/riscv-none-embed/8.1.0/../../../../riscv-none-embed/lib\libg.a(lib_a-isattyr.o): In function `.L0 ':
isattyr.c:(.text._isatty_r+0xe): undefined reference to `_isatty'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:44: Hello World.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

abhijeet085, 2019-03-21:
Also how to get gcc prefix as "riscv64-unknown-elf-gcc"?
Right now my gcc prefix is "riscv-none-embed-gcc".

ilg-ul, 2019-03-21:
To keep the linker happy, you can use `--specs=nosys.specs`, but this does not mean you'll get a functional project, you need startup code and linker scripts, not to mention that probably `-mcmodel=medlow` will crash the link.

For the HiFive1 and Arty 31/51 boards, you can generate functional projects within Eclipse with the SiFive Project Templates.

If you want to use `riscv64-unknown-elf-gcc`, download the SiFive toolchain separately and update the project settings to use it. However I do not guarantee that projects generated by the SIFive project template can be compiled with `riscv64-unknown-elf-gcc`.

abhijeet085, 2019-03-21:
@ilg-ul .
 I have used option --specs=nosys.specs and it compiled.
Can you explain function of  --specs=nosys.specs option?
Thanks.

ilg-ul, 2019-03-21:
It defines empty functions for all those names that triggered errors.

abhijeet085, 2019-03-22:
@ilg-ul 
Hi,
If it defines empty functions, then the functionality of code changes?

ilg-ul, 2019-03-22:
The reason why those names trigger errors is because the application should define them, to provide the desired functionality, like redirecting the printf() output to some device (POSIX retargetting).

If the application is not interrested in such functionality, for convenience, it can use the empty functions, which obviously do nothing, but at least the linker will no longer complain.

ilg-ul, 2019-03-22:
Please close this ticket, since it is not related to a toolchain bug. For questions and support, use the forums.

abhijeet085, 2019-03-22:
@ilg-ul 
Thanks

closed 2019-03-22 by abhijeet085

ilg-ul, 2019-03-22:
You're welcome.