riscv-collab/riscv-gnu-toolchain #419: How to build a 32-bit linux toolchain

V1Kin9, 2019-03-25:
I found that the version of glibc is 2.26.However,I noticed that the glibc with version 2.26 does not support the riscv32-*-linux-gnu configure.

In the branch 2.29 of [glibc](https://github.com/riscv/riscv-glibc/tree/riscv-glibc-2.29) seems to supprot the riscv32-*-linux-gnu configure,so I tried to instead the glibc of the riscv-gnu-toolchain,but it doesn't work when I compiled it.

This is the info 
`running configure fragment for sysdeps/unix/sysv/linux
checking installed Linux kernel header files... missing or too old!
configure: error: GNU libc requires kernel header files from
Linux 3.2.0 or later to be installed before configuring.
The kernel header files are found usually in /usr/include/asm and
/usr/include/linux; make sure these directories use files from
Linux 3.2.0 or later.  This check uses <linux/version.h>, so
make sure that file was built correctly when installing the kernel header
files.  To use kernel headers not from /usr/include/linux, use the
configure option --with-headers.
Makefile:192: recipe for target 'stamps/build-glibc-linux-rv32imac-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imac-ilp32] Error 1
`

I want to know if instead the glibc only can build the 32-bit linux gnu toolchain?

V1Kin9, 2019-03-25:
I tried to use the gcc with the version 7.2.0,and the result with another mistake
`/home/viking/Tools/prefix/Linux_glibc2.29/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++
/home/viking/Tools/prefix/Linux_glibc2.29/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
../Rules:193: recipe for target '/home/viking/Tools/riscv-gnu-toolchain_32/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/support/links-dso-program' failed
make[3]: *** [/home/viking/Tools/riscv-gnu-toolchain_32/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32/support/links-dso-program] Error 1
make[3]: Leaving directory '/home/viking/Tools/riscv-gnu-toolchain_32/riscv-gnu-toolchain/riscv-glibc/support'
Makefile:258: recipe for target 'support/others' failed
make[2]: *** [support/others] Error 2
make[2]: Leaving directory '/home/viking/Tools/riscv-gnu-toolchain_32/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/viking/Tools/riscv-gnu-toolchain_32/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32'
Makefile:192: recipe for target 'stamps/build-glibc-linux-rv32imac-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imac-ilp32] Error 2
`

Is it wrong with my configure?

jim-wilson, 2019-03-25:
The 32-bit RISC-V support has not been contributed upstream yet.  So if you are trying to use FSF glibc-2.29 that will not work.

If you use top of tree riscv/riscv-gnu-toolchain you can build a 32-bit linux toolchain, because we are using an old obsolete pre-upstream copy of glibc that has the 32-bit support.

Zong Li of Andes is working on contributing the 32-bit RISC-V glibc support upstream, and has a branch in riscv/riscv-glibc that of a more recent FSF glibc with the 32-bit support added.  This looks like the 32bit branch which looks like maybe glibc-2.28.  He has also posted patch sets on the glibc development mailing list.

Keep in mind that doing anything with recent glibc and 32-bit support will be complicated until the support is upstream.  You probably shouldn't try unless you are interested in helping contribute to this effort, and willing to learn a lot about glibc.

Don't use gcc-7.2.  It is known to be broken.  Anything gcc-7.3 or later should be OK, but you may as well using gcc-8.x now.

But your 7.2.0 build failures looks like you did something wrong.  But since you didn't say exactly what you did, I can't tell you what you did wrong.  If you want good help, you must provide enough info for the person on the other side to reproduce what you did from the beginning.

futaris, 2019-05-07:
https://github.com/SymbiFlow/ideas/issues/35

dagmcr, 2019-09-23:
Hi @jim-wilson,
Is there any update on this?
I tried to compile [rv32-glibc](https://github.com/riscv/riscv-gnu-toolchain/tree/rv32-glibc) branch which usues glibc 2.29 with rv32 support but still prints the same error.

```shell
./configure --prefix=/opt/riscv --with-arch=rv32i
sudo make linux
```
Output error:
```
make: *** [Makefile:252: stamps/build-glibc-linux-rv32i-ilp32] Error 1
```

jim-wilson, 2019-09-23:
You didn't show the actual error, but I assume it is a complaint about missing atomic support.  Linux requires the A (atomic) extension, so rv32i linux is not possible.

Mainline riscv-gnu-toolchain is now using glibc-2.29 with Zong Li's patches, and it should build and run fine, though the ABI will change before it is upstreamed.

dagmcr, 2019-09-24:
Thanks, @jim-wilson for the support.

I've tried with these different configurations and they both worked:

```shell
./configure --prefix=/opt/riscv --with-arch=rv32imac --with-abi=ilp32 --with-cmodel=medlow
sudo make linux

./configure --prefix=/opt/riscv --with-arch=rv32imafdc --with-abi=ilp32 --with-cmodel=medlow
sudo make linux
```

V1Kin9, 2020-06-16:
> I tried to use the gcc with the version 7.2.0,and the result with another mistake [...]

Thanks for your help @jim-wilson 
I had add another rv64 Toolchain to the $PATH.And when I remove that,the new toolchain can compile successfully.

closed 2020-06-16 by V1Kin9