riscv-collab/riscv-gnu-toolchain #422: How to change the "rm" filed in RV32F instructions dynamically by writing C codes?

hyf6661669, 2019-04-04:
My rv32 compiler is built using --with-arch=rv32imfc and --with-abi=ilp32f, and its version is 7.2.0. I have written some simple codes to test RV32F. But I find that the ```rm``` fields of different RV32F instructions generated by riscv32-unknown-elf-gcc are always equal to ```111```, which means the core should use the content in ```frm``` register as rounding mode.

Here is my C codes:
```
#include <math.h>
volatile float a, b, c, d, e, g;

float float_fun_test(float a, float b)
{
	float c = a * b + a;
	return sqrtf(c);
}

int main()
{
	volatile float float_temp;
	volatile long double temp2 = 2.8;
	a = 1;
	b = 2;
	c = a + b;
	d = a + 3.0;
	e = a * b;
	g = c / d;
	float_temp = float_fun_test(d, g);
	temp2 = float_temp * g;
	return 0;	
}
```

Here is some of the asm codes generated by using ```OBJDUMP```:
```
00010074 <main>:
   10074:	000126b7          	lui	a3,0x12
   10078:	4e06a787          	flw	fa5,1248(a3) # 124e0 <__SDATA_BEGIN__>
   1007c:	00012737          	lui	a4,0x12
   10080:	c6f1a827          	fsw	fa5,-912(gp) # 12528 <a>
   10084:	4e472787          	flw	fa5,1252(a4) # 124e4 <__SDATA_BEGIN__+0x4>
   10088:	c6f1a027          	fsw	fa5,-928(gp) # 12518 <b>
   1008c:	c701a787          	flw	fa5,-912(gp) # 12528 <a>
   10090:	c601a707          	flw	fa4,-928(gp) # 12518 <b>
   10094:	00012537          	lui	a0,0x12
   10098:	00e7f7d3          	fadd.s	fa5,fa5,fa4
   1009c:	4e852707          	flw	fa4,1256(a0) # 124e8 <__SDATA_BEGIN__+0x8>
   100a0:	7179                	addi	sp,sp,-48
   100a2:	ee22                	fsw	fs0,28(sp)
   100a4:	c6f1a427          	fsw	fa5,-920(gp) # 12520 <c>
   100a8:	c701a787          	flw	fa5,-912(gp) # 12528 <a>
   100ac:	d606                	sw	ra,44(sp)
   100ae:	00e7f7d3          	fadd.s	fa5,fa5,fa4
   100b2:	c6f1a627          	fsw	fa5,-916(gp) # 12524 <d>
   100b6:	c701a787          	flw	fa5,-912(gp) # 12528 <a>
   100ba:	c601a707          	flw	fa4,-928(gp) # 12518 <b>
   100be:	10e7f7d3          	fmul.s	fa5,fa5,fa4
   100c2:	c6f1aa27          	fsw	fa5,-908(gp) # 1252c <e>
   100c6:	c681a787          	flw	fa5,-920(gp) # 12520 <c>
   100ca:	c6c1a707          	flw	fa4,-916(gp) # 12524 <d>
   100ce:	18e7f7d3          	fdiv.s	fa5,fa5,fa4
   100d2:	c6f1a227          	fsw	fa5,-924(gp) # 1251c <g>
   100d6:	c6c1a507          	flw	fa0,-916(gp) # 12524 <d>
   100da:	c641a787          	flw	fa5,-924(gp) # 1251c <g>
   100de:	50f57543          	fmadd.s	fa0,fa0,fa5,fa0
   100e2:	f00007d3          	fmv.w.x	fa5,zero
   100e6:	00102773          	frflags	a4
   100ea:	a0f517d3          	flt.s	a5,fa0,fa5
   100ee:	00171073          	fsflags	a4
   100f2:	58057453          	fsqrt.s	fs0,fa0
   100f6:	e799                	bnez	a5,10104 <main+0x90>
   100f8:	50b2                	lw	ra,44(sp)
   100fa:	e622                	fsw	fs0,12(sp)
   100fc:	4501                	li	a0,0
   100fe:	6472                	flw	fs0,28(sp)
   10100:	6145                	addi	sp,sp,48
   10102:	8082                	ret
   10104:	2a01                	jal	10214 <sqrtf>
   10106:	bfcd                	j	100f8 <main+0x84>
```

aswaterman, 2019-04-04:
For most floating-point ops, GCC only knows how to use the dynamic rounding
mode. The exception to this rule is truncation to integer (casting a float
to an int), which will use the rtz static rounding mode.

For everything else, you need to use inline assembly.  Good luck!

closed 2019-04-04 by aswaterman

jim-wilson, 2019-04-04:
You can use the functions in fenv.h to get and set the rounding modes.  This is supported in glibc (linux) and newlib (elf).  The linux support is known working, I don't know how well tested the newlib support is.  This is an ISO C standard header file, so code using fenv.h should be portable.  See fegetround and fesetround.