riscv-collab/riscv-gnu-toolchain #423: Probable regression in toolchain while building freedom-e-sdk hello application

AmeyaVS, 2019-04-06:
I am having issues building the freedom-e-sdk using this toolchain(using master branch) built from source.

The toolchain is installed in `/opt/riscv-install` and the environment variables are set appropriately as per the requirements of the freedom-e-sdk setup guide.
```
export RISCV_PATH=/opt/riscv-install
export RISCV_OPENOCD_PATH=/opt/riscv-install
```
When I try to build the hello sample application using the following command:
`make TARGET=sifive-hifive1 CONFIGURATION=debug PROGRAM=demo software`
 
I am observing the following snippet of the error message from the toolchain:
```
riscv64-unknown-elf-gcc -DPACKAGE_NAME=\"freedom-metal\" -DPACKAGE_TARNAME=\"freedom-metal\" -DPACKAGE_VERSION=\"v0.1.2\" -DPACKAGE_STRING=\"freedom-metal\ v0.1.2\" -DPACKAGE_BUGREPORT=\"palmer@sifive.com\" -DPACKAGE_URL=\"\" -DPACKAGE=\"freedom-metal\" -DVERSION=\"v0.1.2\" -I. -I/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal    -lriscv_metal_ -D__METAL_MACHINE_HEADER=\"/home/ameya/apps/src/sifive/freedom-e-sdk/bsp/sifive-hifive1/metal.h\" -march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/home/ameya/apps/src/sifive/freedom-e-sdk/bsp/sifive-hifive1/install/include -O0 -g -MT src/drivers/libriscv__mmachine__sifive-hifive1_a-riscv,cpu.o -MD -MP -MF src/drivers/.deps/libriscv__mmachine__sifive-hifive1_a-riscv,cpu.Tpo -c -o src/drivers/libriscv__mmachine__sifive-hifive1_a-riscv,cpu.o `test -f 'src/drivers/riscv,cpu.c' || echo '/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal/'`src/drivers/riscv,cpu.c
/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal/src/drivers/riscv,cpu.c: Assembler messages:
/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal/src/drivers/riscv,cpu.c:139: Error: unknown CSR `mtvt'
/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal/src/drivers/riscv,cpu.c:161: Error: unknown CSR `mtvt'
/home/ameya/apps/src/sifive/freedom-e-sdk/freedom-metal/src/drivers/riscv,cpu.c:165: Error: unknown CSR `mtvt'
Makefile:1010: recipe for target 'src/drivers/libriscv__mmachine__sifive-hifive1_a-riscv,cpu.o' failed
make[1]: *** [src/drivers/libriscv__mmachine__sifive-hifive1_a-riscv,cpu.o] Error 1
make[1]: Leaving directory '/home/ameya/apps/src/sifive/freedom-e-sdk/bsp/sifive-hifive1/build/debug'
scripts/libmetal.mk:34: recipe for target '/home/ameya/apps/src/sifive/freedom-e-sdk/bsp/sifive-hifive1/install/lib/debug/stamp' failed
make: *** [/home/ameya/apps/src/sifive/freedom-e-sdk/bsp/sifive-hifive1/install/lib/debug/stamp] Error 2
```

I have also attached the complete log from the build process.

Let me know in-case more information is needed.

Regards,
Ameya Vikram Singh

[makeError.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3050149/makeError.txt)

jim-wilson, 2019-04-06:
mtvt is one of the CLIC CSRs.  Since CLIC is not a RISC-V standard, the patches for that are not in the upstream Binutils sources, they are only in the SiFive BInutils sources.  So you need to use a SiFive toolchain in order to use the CLIC support in freedom-e-sdk.

We have toolchains on our web site
    https://www.sifive.com/boards
Our source trees can be found on github.  sifive/freedom-tools is used to build our toolchain releases.

Or I suppose you could hack freedom-e-sdk to use the mtvt register number instead of its name.  That will work with any assembler.

I believe that HiFive1 uses the CLINT not the CLIC.  Interesting that building for HiFive1 requires CLIC support.  You could report that as a bug against freedom-e-sdk but that might be part of the design and may not be fixable.

palmer-dabbelt, 2019-04-08:
Thanks.  This is a freedom-metal bug so I opened one over there.

closed 2019-05-16 by jim-wilson