riscv-collab/riscv-gnu-toolchain #424: Is this repo upstreamed?

marcopeereboom, 2019-04-11:
I was wondering if this repo has been completely upstreamed or if it still has bits that have not made it yet.

jim-wilson, 2019-04-11:
The 32-bit glibc support is not upstream yet.  Zong Li of Andes has patches for this, and we are hoping that he submits them for review for the next glibc release.  This was blocked waiting for the time_t infrastructure changes.  All 32-bit linux targets currently use a 32-bit time_t that will overflow in 2038, and will require an inconvenient ABI break to fix by changing time_t to a 64-bit type.  As RISC-V is new, we were waiting for the 64-bit time_t on 32-bit linux systems support to go in first, so we could avoid an ABI break shortly after upstreaming the port.  That support is in now I believe, so it is just a matter of someone having enough time to do the work of upstreaming the changes.

Everything else is upstream.

Well, everything that matters that is.  Technically, we have a gdb sim port that isn't upstream, but few people care about this, and this doesn't block anything as qemu and spike are better choices for almost everyone.  Embecosm has a cgen based gdb sim port that they may or may not eventually upstream.  Otherwise, we can upstream the hand written one we have in gdb.

marcopeereboom, 2019-04-12:
Fantastic! Thank you!

closed 2019-04-12 by marcopeereboom