riscv-collab/riscv-gnu-toolchain #425: More multilib linux rv32 configs

Dolu1990, 2019-04-15:
Looking at 
https://github.com/riscv/riscv-gnu-toolchain/blob/67c88daab2bf5379a0119cc1de8491ec12894348/configure.ac#L101
I can see that the --enable-multilib + make linux do generate few architectural combination, but no  rv32ia, rv32iam, rv32iac. What is the reason for this ?

Since recently, VexRiscv can run linux with this kind of minimal configs. Can we add those configs ?

jim-wilson, 2019-04-15:
This topic comes up occasionally, but I'm concerned about the multilib explosion here.  If we add new multilibs for one person, then we have to add them for everyone, and we will end up with so many multilibs that it is likely to cause trouble for people.  Every additional multilib we add will need to be built by everyone across the world that does multilib builds.  Builds will take longer, builds and installs will require more disk space, and most importantly, verifying that all of the multilibs work will be far more time consuming.  I think it is a better approach if people add the multilibs they need to their own source trees.  SiFive has our own multilib list we build which matches the hardware we support.  We can also add OS dependent multilibs, e.g. rtems has its own multilib list in the FSF GCC source tree.  But linux is a little tricky here as what the desktop linux folks want is not the same as what the embedded linux folks want.  I don't think that there is any good solution here that makes everyone happy.

Dolu1990, 2019-04-15:
(also had to edit https://github.com/riscv/riscv-gnu-toolchain/blob/afcc8bc655d30cf6af054ac1d3f5f89d0627aa79/configure#L3311)

Dolu1990, 2019-04-15:
What's about adding some additional optional arguements ? 
Like for instance : 

```
./configure --prefix=/opt/riscv --enable-multilib --with-arch-abi="rv32i-ilp32 rv32iac-ilp32 rv32im-ilp32 rv32imac-ilp32"
``` 
 ?

I understand you point about the default setting.

jim-wilson, 2019-04-15:
configure is a generated file.  You edit configure.ac and then regenerate configure via autoconf.

If you want to add multilibs to riscv-gnu-toolchain, then you need to edit configure, the riscv-gcc/gcc/config/riscv/*multilib* files, and the regression/Makefile.  This is inconveniently complicated.  It would be better if one could just edit the gcc multilib list, and the other places get the multilib list by querying gcc.  That would require some rewriting of riscv-gnu-toolchain though, and this may never happen.

It isn't just a arch/abi question.  There is also a search path issue.  The current glibc ABI says that we look for libraries in /lib32/ilp32 for a rv32-ilp32 target, but this doesn't work if we have more than one rv32-ilp32 target.  If we change the search path for glibc, then that is an incompatible ABI change which could cause problems for linux distros.

A configure option to choose multilibs sounds convenient, but turns out to be harder to implement than you might think, because of the complexity of handling all of the arch/abi combinations, plus the arch aliases, and the compatible arch combinations, and the search path handling.

GCC incidentally does have a --with-multilib-list= configure option, but the RISC-V version of this only supports one multilib at the moment.  This is needed by linux distros, because they only get the correct search paths for glibc with --enable-multilib, but since the 32-bit glibc support is not upstream yet, they can't actually build the 32-bit multilibs, so they have to use --with-multilib-list= to select the one rv64gc-lp64d multilib they need and which works with upstream sources.  I looked at trying to support more than one arch/abi via --with-multilib-list=, but it gets so complicated so very quickly that it wasn't worth the trouble.  Plus this just a gcc configure option, not a riscv-gnu-toolchain option, so it doesn't help you build the glibc multilibs.

There are a lot of hard problems here.  In general, linux and multilib is a lot harder than embedded elf and multilib, and trying to make this work may not be practical.  It is simpler to just build a different compiler for each linux arch/abi combination you need.

Dolu1990, 2019-04-15:
Ok, thanks for the explanation.
No worries, i will rebuild it for each of the configs in need.

Thanks for your time :)

closed 2019-04-15 by Dolu1990