riscv-collab/riscv-gnu-toolchain #426: @palmer-dabbelt

thomascp, 2019-04-17:
@palmer-dabbelt 

Hi Palmer,
Here is a problem.

I wrote this test codes:

.global main
        .global _start

        .text
        .balign 4

_start:
        j main

        .org 0x40

main:
        nop

First, I compiled it into .o:

a.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start-0x2>:
   0:   0001                    nop

00000002 <_start>:
   2:   a83d                    j       40 <main>
        ...

00000040 <main>:
  40:   0001                    nop
        ...

Then I linked it to a.out,

"riscv32-unknown-linux-gnu-gcc -nostdlib -Ttext=0x0 a.s"

a.out:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start>:
   0:   a83d                    j       3e <main>
        ...

0000003e <main>:
  3e:   0001                    nop
        ...

This is not what we want, right? We want the function main located at 0x40.
But because compiler first inserts extra nops linker remove it, function main moved to 0x3e.
Is this a bug? How to solve it? Thanks!

BR,
Peng

_Originally posted by @thomascp in https://github.com/riscv/riscv-gnu-toolchain/issues/171#issuecomment-484074422_

jim-wilson, 2019-04-17:
Yes, this looks like a bug.  The easy solution would be to disallow .org.  It looks like xtensa solves the problem by disabling relaxation for everything before the .org which could work, but is maybe not the solution you wanted as that will hurt performance.  It could take some time to find a way to fix this.

Note that you can put start and main in different sections, and use a linker script to assign addresses to the sections, and then we can relax both sections, and emit padding nops in between them.  That is a much easier way to get the result you want from a toolchain point of view.

jim-wilson, 2019-04-18:
I took another look at this, and still don't see any good way to fix it, other than by disallowing .org, or by forcing linker relaxation off when .org is seen which is what the xtensa toolchain does.  Linker relaxation to reduce code size is fundamentally at odds with how .org works.

For your testcase, you can workaround the problem by using the assembler option -mno-relax, which just turns off relaxation for this one file which should be good enough to get the result you want.

In general, as mentioned before, I would suggest using sections and linker scripts instead of .org to get fixed addresses.  Sections and linker scripts is the ELF way of doing this.  .org is the old obsolete AOUT object file format way of solving this problem, and doesn't interact well with modern ELF linker features.

thomascp, 2019-04-23:
@jim-wilson Thanks for your help! I will try to fix it by using sections and linker scripts way you mentioned.

closed 2019-04-23 by thomascp