riscv-collab/riscv-gnu-toolchain #427: Does the riscv-newlib support different rounding modes for conversion from a single-precision floaintg point number to int/uint?

hyf6661669, 2019-04-21:
My riscv-newlib compiler is built with:
```
--with-arch=rv32imfc --with-abi=ilp32f
make
```
Just some very simple C codes:
```
#include <fenv.h>
#include <math.h>

volatile float a;
volatile int d;
int main()
{
	a = 0.5;
	d = rintf(a);
	return 0;	
}

```
The command I use is:
```
/home/hyf/riscv-compiler/rv32imfc/bin/riscv32-unknown-elf-gcc rounding_test.c -o 
rounding_test -lm -fno-math-errno -O2
/home/hyf/riscv-compiler/rv32imfc/bin/riscv32-unknown-elf-objdump -d rounding_test > rounding_test.dump
```
If I don't use ```-lm``` it would return:
```
undefined reference to `roundf'
```

The related generated asm codes are:
```
00010074 <main>:
   10074:	00012737          	lui	a4,0x12
   10078:	8b072787          	flw	fa5,-1872(a4) # 118b0 <__SDATA_BEGIN__>
   1007c:	1141                	addi	sp,sp,-16
   1007e:	c4f1ac27          	fsw	fa5,-936(gp) # 118e0 <a>
   10082:	c581a507          	flw	fa0,-936(gp) # 118e0 <a>
   10086:	c606                	sw	ra,12(sp)
   10088:	20dd                	jal	1016e <roundf>//this is related to f2i conversion
   1008a:	c0051753          	fcvt.w.s	a4,fa0,rtz
   1008e:	40b2                	lw	ra,12(sp)
   10090:	c4e1aa23          	sw	a4,-940(gp) # 118dc <d>
   10094:	4501                	li	a0,0
   10096:	0141                	addi	sp,sp,16
   10098:	8082                	ret
```
If I use the function ```ceilf()``` in C, the asm code at address 0x1008a would be ```jal	1016e <ceilf>```.
If I use the function ```floorf()``` in C, the asm code at address 0x1008a would be ```jal	1016e <floorlf>```.

It seems that the compiler uses the softfloat library for f2i conversion.

According to [https://github.com/riscv/riscv-asm-manual/blob/master/riscv-asm.md](url), the 
instruction should be(just ignore the register name):
```
fcvt.w.s a0, fa0, rmm, for roundf()
fcvt.w.s a0, fa0, rup, for ceilf()
fcvt.w.s a0, fa0, rdn, for floorf()
```

If I'm wrong, could you please tell me how to make the compiler emit expected instructions of differnet rounding modes for f2i conversion?

hyf6661669, 2019-04-21:
In ```riscv-gnu-toolchain/riscv-newlib/newlib/libm/machine/riscv```, there are no implementations about ```ceilf(), roundf(), floorf()```. But in ```riscv-gnu-toolchain/riscv-glibc/sysdeps/riscv/rvf``` there are many related *.c files for RV32F instructions and functions. Does it mean that I must use glibc for compiling?

jim-wilson, 2019-04-21:
There are multiple issues here.

The ISO C standard says that math functions set errno on errors.  So we can't just use the hardware instruction as it doesn't set errno.  You can tell gcc that you don't want math library errno by using -fno-math-errno.  That will give the single hardware instruction you want.

Ideally the newlib math library functions should just use the hardware instruction and then have a few more instruction to set errno, but no one has done that RISC-V porting work yet, so the math library will instead use soft-float code as you noticed which is inefficient.

We don't have any support for generating rounding  modes directly from the compiler yet.  We do have fenv.h support though.  You can include fenv.h and call fesetround().  And then maybe another fesetround call after the operation to put the default rounding mode back.  So this is two or three instructions instead of just the one you want, but can work.

Otherwise, you can use an extended asm to get the instruction you want.  See the gcc docs
    https://gcc.gnu.org/onlinedocs/gcc-8.3.0/gcc/Using-Assembly-Language-with-C.html
Extended asms can be tricky to use if you have never used them before, so expect to spend some time learning how to get the right result.

jim-wilson, 2019-04-21:
glibc is only supported for linux.

hyf6661669, 2019-04-21:
@jim-wilson 
Thanks for your fast reply. I know that the function ```fesetround()``` will change the rounding modes. The problem is that for f2i operations, if I just write:
```
float f;
int a = (int)f;
```
The compiler will generate ```fcvt.w.s a0, fa0, rtz```. It doesn't even want to set the ```rm``` field in the RV32F instruction to ```111```. 

Since I'm writing C codes for embedded MCUs which is not able to run linux, it seems that the only solution is writing asm codes for different rounding modes to do f2i conversion.

hyf6661669, 2019-04-21:
> @jim-wilson
> Thanks for your fast reply. I know that the function `fesetround()` will change the rounding modes. The problem is that for f2i operations, if I just write:
> 
> ```
> float f;
> int a = (int)f;
> ```
> The compiler will generate `fcvt.w.s a0, fa0, rtz`. It doesn't even want to set the `rm` field in the RV32F instruction to `111`.
> 
> Since I'm writing C codes for embedded MCUs which are not able to run linux, it seems that the only solution is writing asm codes for different rounding modes to do f2i conversion.
> Or do you know how to make the compiler set the the `rm` field in the RV32F instruction to `111` when I'm doing f2i conversion?

jim-wilson, 2019-04-21:
I was in a rush early this morning, and didn't have a chance to look at this closely.

The ISO C standard, in section 6.3.1.4 Real floating and integer, says that when a float is converted to int, the fractional part is discarded (i.e. the value is truncated to zero).  So what gcc is emitting for a simple cast is correct as per the ISO C standard.

I see that we don't have support for rint or round, but curiously we do have support for lrint and lround that will do the right thing.  Perhaps because long is always the same as the register size (currently), which makes this easier to support.

```
rohan:2109$ cat tmp.c
#include <math.h>
long sub (float f) { return lrintf (f); }
long sub2 (float f) { return lroundf (f); }
rohan:2110$ riscv32-unknown-elf-gcc -O -S -fno-math-errno tmp.c
rohan:2111$ cat tmp.s
	.file	"tmp.c"
	.option nopic
	.text
	.align	1
	.globl	sub
	.type	sub, @function
sub:
	fcvt.w.s a0,fa0,dyn
	ret
	.size	sub, .-sub
	.align	1
	.globl	sub2
	.type	sub2, @function
sub2:
	fcvt.w.s a0,fa0,rmm
	ret
	.size	sub2, .-sub2
	.ident	"GCC: (GNU) 8.3.0"
rohan:2112$ 
```

So you can get the result you want by using lrint and lround.

The missing bits are fixable here, it just needs someone to volunteer their time to fix the problems.

hyf6661669, 2019-04-22:
@jim-wilson 
Very amazing solution1 I will leave the issue open for some days to see whether others can give some advice.

closed 2019-04-25 by hyf6661669