riscv-collab/riscv-gnu-toolchain #430: native

longxjk, 2019-04-28:


longxjk, 2019-04-28:
I make the gcc use linux-native s，but get this erro

# gcc test.c
test.c:1:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.

jim-wilson, 2019-04-28:
This is a little confusing.  I see a question in email but not in the issue, and you already closed the issue, so it isn't clear if you actually have a question or not.

Anyways, the more info you provide the better the answer we can give.  You didn't mention exactly how you are running the native gcc that you built.  If you are running it in a root filesystem, then that root file system needs to have a copy of glibc installed, and it needs to be the exact same glibc that you used to build gcc.

The linux-native rule in riscv-gnu-toolchain is not very useful.  it was interesting 1.5 years ago, but now that we have linux distros, it is much better to boot a linux distro and then use the gcc it provides, or build a new gcc natively.  If you want to build your own distro from scratch, then look at open embedded (riscv/meta-riscv).

longxjk, 2019-04-28:
Thank you very much for your reply. At the beginning, I didn't think how to describe it accurately, so I closed the problem，I obtained the code for branch native- GCC, using the following configuration
./configure \
--prefix=/opt/native-riscv0428 \
--with-arch=rv64imafdc \
--with-abi=lp64 \
--enable-linux \
And use the command make linux-native to compile successfully.
Copy all native files under /opt/native-riscv0428 to the root directory
And copy all files in the sysroot directory to the root directory as well
Write a test code gcctest.c,
The result is that g++ works, but GCC fails
The error is as follows

~ # gcc gcctest.c
In file included from /usr/include/stdio.h:75:0,
                 from gcctest.c:1:
/usr/include/libio.h:305:3: internal compiler error: in grokdeclarator, at c/c-decl.c:6639
   char _unused2[15 * sizeof (int) - 4 * sizeof (void *) - sizeof (size_t)];
   ^~~~
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.

jim-wilson, 2019-04-28:
Does g++ work on your gcctest.c file?  If so, that suggests that your compiler is basically working, and you may have to debug your gcc to see what is wrong with it.

You mentioned that you copied stuff into the root filesystem, but you didn't say where the root file system came from, or where you copied stuff to in the root file system, or how you are running the code (qemu?), etc.

Otherwise, I can only reiterate that whatever you are trying to do, you are probably doing it wrong.  You should not be using the riscv-gnu-toolchain rules for building a native toolchain.  This is one of the hardest ways to build a native compiler.  It was only useful in the beginning when we didn't have usable linux distros.

longxjk, 2019-04-29:
Yes, g++ is working well, GCC is wrong.How do I debug GCC?
I use the hardware of riscv instruction ，not the qume. My root file system is built by using buildroot. I directly overwrite the files under native and sysroot into my root file system.

jim-wilson, 2019-04-29:
Debugging gcc is complicated and likely to take some time to learn how gcc works.  If you have gdb working, thatr might be the fastest approach, otherwise you will have to use printf.  You have the error message, you can search the gcc sources to find where the error is printed, then add some printf statements to try to figure out why the error was printed, and go backwards from there.

However, I would have to say again that what you are doing is fundamentally wrong.  Using riscv-gnu-toolchain to build a native compiler is fundamentally wrong.  Copying the riscv-gnu-toolchain sysroot overtop of the buildroot sysroot is fundamentally wrong.  As long as you continue to do everything the wrong way, you can going to have no end of problems.

jim-wilson, 2019-04-29:
if you want a native gcc, then it is best to use one of the desktop distros, e.g. fedora, debian, opensuse, etc

If you really want buildroot, then you should probably stick with what buildroot provides.  If buildroot provides a native compiler, then use that native compiler.  If buildroot only provides a cross, then you should probably only use that cross.

if you really want buildroot, and a native compiler, and buildroot doesn't provide one, then you should create a sysroot from your buildroot target file system, build a cross binutils and gcc using the sysroot, and then use the sysroot and cross binutils/gcc to canandian cross build a native binutils and gcc.  that will give you a compiler that is compatible with the buildroot rootfs.

or you might try looking at open embedded/yocto which works similarly to buildroot, and should have support for building a native compiler.

You should not use riscv-gnu-toolchain at any step of this, as that will just create problems.

longxjk, 2019-04-30:
Thank you for your careful reply. I tried to use the root directory as sysroot, but GCC could not specify sysroot in the configuration item. Sysroot was generated when GCC was compiled!How do I specify sysroot for GCC ?

longxjk, 2019-04-30:
I'm also confused, how do I configure the native- GCC compilation rules ?

jim-wilson, 2019-04-30:
You need to use the sysroot for both binutils and gcc.  There is a configure option --with-sysroot= that you can use to point at the directory that holds the target sysroot.  The sysroot should be created from the target root filesystem, and should contain /usr/include, /usr/lib, /lib, and other related dirs as necessary, e.g. you might need /lib64 for a 64-bit system too.

closed 2019-05-16 by jim-wilson